gazebosim/gz-sim #810: Use math::SpeedLimiter inside controllers

chapulina, 2021-05-06:
Controllers like `diff_drive` have a [SpeedLimiter](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo5/src/systems/diff_drive/SpeedLimiter.hh) class inside them. This file is currently repeated inside these plugins:

* `diff_drive`
* `ackermann_steering`

Now that we have the [ignition::math::SpeedLimiter](https://ignitionrobotics.org/api/math/6.8/classignition_1_1math_1_1SpeedLimiter.html) class, we should use that and remove those custom limiters. The [mecanum_drive](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo5/src/systems/mecanum_drive/MecanumDrive.cc) plugin is already using that.

## Desired behavior

* all `SpeedLimiter` files are removed from `ign-gazebo`
* `diff_drive` and `ackermann_steering` use `math::SpeedLimiter`

All tests should continue to pass.

## Alternatives considered

We moved the `SpeedLimiter` to `ign-math` so we could use it in multiple places. An alternative is to keep duplicating the file, but that doesn't scale well. Plus, we made improvements and fixes to the limiter when porting it to math.

## Implementation suggestion

Take a look at how the mecanum drive is using it.

scpeters, 2021-05-17:
For reference, 9bbccebb80aae51dd645ed55c8d6c25687b8f9dc is the commit in #683 that changed the mecanum drive to use the `ignition::math::SpeedLimiter`

LolaSegura, 2021-05-21:
Hi I have been working on this issue and I noticed it has a 'citadel' tag. I was wondering, wouldn't it make sense to push this on the latest version (gazebo6) as I saw it hasn't been changed in either of the previous versions.

scpeters, 2021-05-21:
> Hi I have been working on this issue and I noticed it has a 'citadel' tag. I was wondering, wouldn't it make sense to push this on the latest version (gazebo6) as I saw it hasn't been changed in either of the previous versions.

we have added some "guiding principles" to our ignitionrobotics.org/about page, but they haven't been deployed yet, so I'll excerpt one here:

~~~
<b>Updates</b>:
            New backwards compatible features are added to stable releases
            whenever possible. This way, users can take advantage of new features
            without the need to upgrade to a later release, and can postpone
            upgrade efforts until breaking changes are needed.
~~~

Since this change affects only the implementation of a controller and not the public API or a change in dependency version, it is backwards compatible. So we prefer to target the oldest supported version (citadel, `ign-gazebo3`) and then merge it forward. How does that sound?

It is also possible to target the newest branch and then backport, but we have a slight preference for targeting the older versions first and merging forwards.

LolaSegura, 2021-05-24:
That sounds good. I will be doing that, thank you for the reply.

francocipollone, 2021-05-24:
Thanks @scpeters for the clarification. 
So given that `diff_drive_controller` was added in `citadel` and `ackerman_steering` was added in `dome`, the next steps would be:

 - [ ] `diff_drive_controller`. Remove custom limiter in `citadel`
 - [ ] `ackerman_steering_controller`. Remove custom limiter in `dome`
 - [ ] Port `diff_drive_controller` changes from `citadel` to `dome`
 - [ ] Port `diff_drive_controller` and `ackerman _steering_controller` changes from `dome` to `edifice`
 - [ ] Port `diff_drive_controller` and `ackerman _steering_controller` changes from `edifice` to `fortress`/`main`

scpeters, 2021-05-24:
> the next steps would be:

yes, precisely. Thanks for laying it out so clearly

scpeters, 2021-05-27:
> * `diff_drive_controller`. Remove custom limiter in `citadel`

see https://github.com/ignitionrobotics/ign-gazebo/pull/833

LolaSegura, 2021-05-27:
> * `achermann_steering`. Remove custom limiter in `dome`

see #837

closed 2021-06-22 by chapulina