gazebosim/gz-sim #820: Confirmation dialog on exiting Ignition

mabelzhang, 2021-05-17:
## Desired behavior

On closing Ignition via the X button in the system title bar, have a dialog pop up to confirm exit.
That behavior can get annoying in development and testing, so an additional checkbox menu item for "ask before exit" would be nice.

Currently, the 3 dot "..." menu button is really close to the Close X button on Ubuntu. It's an easy mistake to close all of Ignition when the user meant to open the menu to inspect something in the middle of a simulation.

## Alternatives considered

Hmmmmm alternatives..... tough to think of.
"It's okay, no one wants to keep Ignition open anyway."
"Just let them close it, they can always restart. Who cares."
"Only someone as clumsy as this issue's opener would make a mistake like that." (and I did)

....... None seem acceptable to me, especially the last one.

## Implementation suggestion

This PR might help as reference https://github.com/ignitionrobotics/ign-gazebo/pull/609

## Additional context

See red box in screenshot below.
The screenshot also illustrates that the problem is exacerbated on HiDPI screens, where the mouse cursor is teeny and harder to see or control (I'll open a separate issue for that).

[image]

francocipollone, 2021-05-20:
I am tackling this on my branch: https://github.com/ignitionrobotics/ign-gui/commit/1589fef6f38209728e465929d8adaeac2abb1747

So far:

https://user-images.githubusercontent.com/53065142/119032527-a8929d80-b982-11eb-8d9e-95053a00bf39.mp4

Actually, I wanted to add this feature from `ign-gazebo` but given that I had to connect to the `closing` signal of the `AplicationWindow` it wasn't that straightforward.
It is true that the request was only for the close button (x), not for the `quit` option in the sidebar, I could change that if needed.

If this meets what is required I will create the correspondent PR. :+1:

mabelzhang, 2021-05-20:
Ooh nice! Thank you for tackling this! Please go ahead and make a PR and we'll review it.

mjcarroll, 2021-06-21:
Closed by ignitionrobotics/ign-gui#225 and ignitionrobotics/ign-gazebo#850

closed 2021-06-21 by mjcarroll