gazebosim/gz-sim #821: Better scrolling for plugins menu

mabelzhang, 2021-05-17:
## Desired behavior

This is a UX improvement feature request for qualitatively "better" scrolling of the plugins menu.
I've only tested on a HiDPI laptop with track pad, not sure if the behavior is different on other screens.

Currently, the scrolling is pretty unusable for me.
I get 5 menu items, and if I scroll normally, it jumps in a flash from the top of the menu to the bottom, with everything in the middle rushing by in a blur.
To get to the item I want, I have to scroll extra carefully, and that is if I already know the plugin's name and where in the alphabetically ordered menu to look for it.
See GIF below for illustration.

Two things I can quantify "better":
1. Longer menu, with more items
2. Slower scrolling. This will require testing on multiple systems, to make sure it works well for everyone. In general though, my scrolling configuration is pretty normal -- my default scrolling is not extra slow. I should be able to expect the scrolling speed to be similar to a browser window, for example. Currently, it's a lot faster.

## Alternatives considered

## Implementation suggestion

This PR might help to see where to get started https://github.com/ignitionrobotics/ign-gui/pull/134

## Additional context

This is the *slowest* I can scroll on my laptop track pad, if I scroll very carefully. Some menu items still take several tries to get to.
If I scroll *normally*, it just jumps from the top to the bottom. Nothing in between is readable.
[image]

A difference I notice is that I only get 5 items in the menu. In this other issue about plugin menu scrolling https://github.com/ignitionrobotics/ign-gazebo/issues/368 , the window looks impossibly long, perhaps it was taken on a vertical monitor?
A longer menu would help alleviate some of the bad scrolling.

chapulina, 2021-06-14:
That menu becomes less usable the more plugins we have. I think we should consider refactoring it completely. Some other upgrades I can think of:

* Let the user search for the plugin name, with autocomplete.
* Allow passing configuration to plugins when inserting them from the menu (https://github.com/ignitionrobotics/ign-gui/issues/142).
* The menu only offers GUI plugins at the moment. We could make the menu extendable, so downstream applications can append their own plugin types to the menu (in an organized way). Related to https://github.com/ignitionrobotics/ign-gazebo/issues/190

Also, note that this fix will go into `ign-gui`.

chapulina, 2021-09-01:
> Let the user search for the plugin name

See https://github.com/ignitionrobotics/ign-gui/pull/277

Since the other items have separate issues tracking them, I'll close this issue once the PR above is merged.

closed 2021-09-02 by chapulina