gazebosim/gz-sim #822: Large cursor size on HiDPI screens

mabelzhang, 2021-05-17:
## Desired behavior

UX improvement feature request for larger cursor size on HiDPI screens. Followup (kind of) on https://github.com/ignitionrobotics/ign-gazebo/pull/291

Scale cursor size to be similar to system-wide cursor size.

A common occurrence with HiDPI screens is that application windows are teeny, so in #291, we made the Ignition window bigger, respecting the system scaling factor.

I realize I complained in that PR that the cursor is too big, and now I'm complaining the cursor is too small.
But it contributes to an overall user experience problem, one instance of which is seen in https://github.com/ignitionrobotics/ign-gazebo/issues/820
The cursor is harder to see and control, meaning the user has to be extra-careful while using Ignition.
See GIF below for illustration.

## Alternatives considered

## Implementation suggestion

I don't know if this is possible, and if so, I have no idea where this lives. Will add if I come across something.

Just documenting for awareness if a solution comes up.

## Additional context

Comparison of my system-wide cursor size and the teeny cursor size in the Ignition window:

[image]

chapulina, 2021-06-14:
Interesting, are you running inside a container by any chance?

mabelzhang, 2021-06-15:
Ah, I am. At some point I switched from native development to Docker. Is this a parameter that should be passed into the container?

chapulina, 2021-06-16:
Hummm I think it may be because of the container, and perhaps not Ignition specific. I've noticed that on some of my containers the cursor looks different from the host, but it's usually the same size :thinking: 

You could check another graphical interface like Gazebo classic or RViz inside the container, and also check what the cursor looks like if you install Ignition on the host.