gazebosim/gz-sim #827: a plugin to capture video with the help of the ThermalCameraSensor

addy1997, 2021-05-20:
## Desired behavior
Currently, the Thermal Camera can only be used to capture an image and. How to add the functionality to capture a video with the help of this sensor?

## Alternatives considered
I tried adding the functionality of the video camera recorder to the thermal camera sensor but it failed as I misestimated the functioning of both slightly.

## Additional context
Since I am working on the application to track animals' movements in the forest, I wanted something that can record the changes in the image, i.e., the real-time changes but with the thermal camera. Otherwise, I would have switched to the existing video camera recorder. Could you suggest some tweaks to add the functionality of `video recording` in the `thermal camera sensor`?

Thanks

addy1997, 2021-05-26:
@scpeters, can this feature be incorporated in the `thermal camera`?  if not, is there an existing `sensor` that can help me in my application?

iche033, 2021-05-27:
this will need to be done in a separate plugin. The current video recorder records a video from the GUI camera only. In order to record a video from the thermal camera, you'll need to write a plugin that subscribes to the thermal camera topic, and encode the image frames on callback using `common::VideoEncoder`. You can take a look at the [Scene3D.cc](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo5/src/gui/plugins/scene3d/Scene3D.cc) class to see how video recording is done - just search for `videoEncoder`

addy1997, 2021-05-27:
Thanks, @iche033. I'll close this issue.

closed 2021-05-27 by addy1997