gazebosim/gz-sim #835: Move some plugins to ign-gui

ahcorde, 2021-05-26:
I was checking some of the plugins that are currently in ign-gazebo but they don't have any gazebo dependency. I think we should move this plugins to ign-gui.

For example:

 - video_recorder
 - view_angle
 - grid_config
 - tape_measure (when this PR is in https://github.com/ignitionrobotics/ign-gui/pull/213)

@scpeters @iche033 @mjcarroll thoughts?

mjcarroll, 2021-05-27:
I'm in favor of moving whatever possible into ign-gui. :+1:

ahcorde, 2021-05-27:
what's the process of doing that ? should I need to deprecate something ? or is it just remove from ign-gazebo and move the plugin to ign-gui ?

mjcarroll, 2021-05-27:
My opinion would be that we can move them in the next major release of both `gui` and `gazebo` without deprecation.  Anybody depending on `ign-gazebo` would already be depending on `ign-gui`.

scpeters, 2021-05-27:
I support moving plugins to ign-gui when possible

> * grid_config
> * tape_measure (when this PR is in [ignitionrobotics/ign-gui#213](https://github.com/ignitionrobotics/ign-gui/pull/213))

these ones look straightforward to me

> * video_recorder
> * view_angle

these ones depend on ignition transport services `/gui/view_angle` and `/gui/record_video`:

* https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo5/src/gui/plugins/view_angle/ViewAngle.cc#L64
* https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo5/src/gui/plugins/video_recorder/VideoRecorder.cc#L71

I think the [camera_video_recorder system](https://github.com/ignitionrobotics/ign-gazebo/blob/a41cf80049c07977079d0b215398dc4225b63038/src/systems/camera_video_recorder/CameraVideoRecorder.cc) and the `view_angle` service would have to be migrated to `ignition::gui::Scene3D`

scpeters, 2021-05-27:
I know there are different names used for `Scene3D` in ign-gui and `GzScene3D` in ign-gazebo since the config files just look for shared libraries. So I expect we might need to use different names for the `ign-gui` version?

chapulina, 2021-06-14:
Yes! These plugins should have never been added to `ign-gazebo` to begin with, we did that for expediency at first, but new plugins such as `Screenshot` have been added directly to `ign-gui`.

> So I expect we might need to use different names for the ign-gui version?

Correct, we'd need to use different names if we want them to coexist. But @mjcarroll 's idea is interesting:

> we can move them in the next major release of both gui and gazebo without deprecation.

Technically, Ignition GUI would just pick up the different library, so I think it could work. If we take this route, we should make sure we don't change any of the plugin arguments so users can transition seamlessly.

ahcorde, 2021-06-21:
PRs
  - [ ] Removed from ign-gazebo https://github.com/ignitionrobotics/ign-gazebo/pull/870
  - [ ] Added to ign-gui https://github.com/ignitionrobotics/ign-gui/pull/236

chapulina, 2021-09-08:
All plugins that could be trivially moved have been

closed 2021-09-08 by chapulina