gazebosim/gz-sim #842: How to change material of model in system plugin?

gezp, 2021-05-31:
I want to change color of model dynamically by using plugin. Then I modify component  `Material`  in component  ` Visual`  by using `SetComponentData()`  in `PreUpdate()` , but there is no effect. 

```c++
SDF::Material material ;
ignition::math::Color color(1,0,0,1);
m.SetAmbient(color);
m.SetDiffuse(color);
m.SetSpecular(color);
_ecm.SetComponentData<components::Material>(visualEntity,m);
```
So, Does anyone know how to change material of model in system plugin?  Thank you!

chapulina, 2021-06-15:
That functionality hasn't been implemented yet. The implementation would be similar to how we update lights, see https://github.com/ignitionrobotics/ign-gazebo/pull/482 as a reference.

mhl787156, 2024-12-31:
Hi, Sorry to reopen - @chapulina do you know if there has been any progress on this since 2021? I'm on gazebo fortress (ignition/gazebo6 i think). 

Kind of assumed this was working and made a plugin for students to work with in 2 weeks time...

azeey, 2025-01-06:
This was implemented as part of https://github.com/gazebosim/gz-sim/pull/1123. You can change material attributes using a `VisualCmd` component. For reference, see https://github.com/gazebosim/gz-sim/blob/9d6230a562c5d463306070b07269b135f9ecd134/src/systems/user_commands/UserCommands.cc#L1865-L1912

closed 2025-01-06 by azeey

mhl787156, 2025-03-05:
Hi @azeey I finally got back to working on this - apologies a couple of questions as I'm still not used to the component model

As far as I understand, VisualCommand is updated via an ign topic exposed by the UserCommands component. How exactly do I enable the UserCommands component so that the VisualCommands topic is created? 

1. Should it be enabled in the plugin I am writing (for changing colours of a specific object) (in which case what should I attach it to), or 
2. Should it be enabled at the world.sdf level? (In which case what is the xml to enable it?) 

Observing my empty.sdf we have the following which does enable the UserCommands plugin, but when running `ign topics -l` I do not see a `/visual_config` topic as would be created per what that pull request calls. 

Sorry to reopen and thanks in advance!

mhl787156, 2025-03-05:
So just to add on, I have created a publisher to visual_config within my plugin and can see the echos when running 

```
> ign topic -i -t /world/empty_visual/visual_config -e
id: 38
transparency: 0.5
material {
  ambient {
    r: 0.8
    g: 0.8
    b: 0.8
    a: 1
  }
  diffuse {
    r: 0.3
    g: 0.4
    b: 0.5
    a: 1
  }
  specular {
    r: 0.8
    g: 0.8
    b: 0.8
    a: 1
  }
  emissive {
    r: 0.3
    g: 0.4
    b: 0.5
    a: 1
  }
}
type: VISUAL

id: 39
transparency: 0.5
material {
  ambient {
    r: 0.8
    g: 0.8
    b: 0.8
    a: 1
  }
  diffuse {
    r: 0.8
    g: 0.8
    b: 0.8
    a: 1
  }
  specular {
    r: 0.8
    g: 0.8
    b: 0.8
    a: 1
  }
  emissive {
    r: 0.8
    g: 0.8
    b: 0.8
    a: 1
  }
}
```

I have double checked that the entity ids match what the Visuals I'm trying to change in the gazebo Tree. 

For more information, here is the entity I am creating:
```
			// Create a visual
			sdf::Visual sdfVisual;
			sdfVisual.SetName("led" + std::to_string(i));
			sdfVisual.SetPoseRelativeTo("base_frame");

			// Calculate x, y position
			float x = this->ring_radius * std::cos(angle_delta * i);
			float y = this->ring_radius * std::sin(angle_delta * i);
			sdfVisual.SetRawPose(gz::math::Pose3d(x, y, this->ring_z_offset, 0.0, 0.0, 0.0));

			// Set cylinder geometry
			sdf::Cylinder cyl;
			cyl.SetRadius(0.015);
			cyl.SetLength(0.015);
			sdf::Geometry sdfGeom;
			sdfGeom.SetType(sdf::GeometryType::CYLINDER);
			sdfGeom.SetCylinderShape(cyl);
			sdfVisual.SetGeom(sdfGeom);

			float r = this->led_colours[i * 3 + 0];
			float b = this->led_colours[i * 3 + 1];
			float g = this->led_colours[i * 3 + 2];

			// Set material
			sdf::Material sdfMaterial;
			// Set the diffuse color to match the LED's body color
			sdfMaterial.SetDiffuse(gz::math::Color(r,g,b, 1.0));
			// Set the emissive color to simulate light emission
			sdfMaterial.SetEmissive(gz::math::Color(r,g,b, 1.0));  // Bright red
			// Set the specular color to simulate glossy reflection
			sdfMaterial.SetSpecular(gz::math::Color(0.8, 0.8, 0.8, 1.0));  // Near-white for strong highlights
			// Optionally add transparency for a translucent effect
			sdfVisual.SetTransparency(0.1);  // Slightly translucent
			sdfVisual.SetMaterial(sdfMaterial);

			// Add the visual to the link
			sdfLink.AddVisual(sdfVisual);

			// Assuming `entityCreator` is already being used to create the entity
			auto visualEntity = entityCreator.CreateEntities(&sdfVisual);
			if (visualEntity == gz::sim::kNullEntity) {
				ignerr << "Failed to create visual entity" << std::endl;
				continue;
			} 
			// Ensure it's marked as a Visual in Gazebo
			_ecm.CreateComponent(visualEntity, gz::sim::components::Visual());
			_ecm.CreateComponent(visualEntity, gz::sim::components::Material());
			this->led_id_entity_map[i] = visualEntity;
			entityCreator.SetParent(visualEntity, linkEntity);
```

And here is how I'm calling the topic:

```
	gz::msgs::Visual msg;
		gz::sim::Entity visual_entity = this->led_id_entity_map[i];
		msg.set_id(visual_entity);
		// msg.set_name("drone0::leds_link::led0");
		// Modify the material color (RGBA format)
		gz::msgs::Material *mat = msg.mutable_material();
		gz::msgs::Set(mat->mutable_diffuse(), gz::math::Color(r, g, b, 1.0));  // Red color
		gz::msgs::Set(mat->mutable_emissive(), gz::math::Color(r, g, b, 1.0));  // Red color
		gz::msgs::Set(mat->mutable_specular(), gz::math::Color(0.8, 0.8, 0.8, 1.0));  // Red color
		gz::msgs::Set(mat->mutable_ambient(), gz::math::Color(0.8, 0.8, 0.8, 1.0));  // Red color
		msg.set_transparency(0.5);
		msg.set_type(gz::msgs::Visual::VISUAL);
		this->visual_config_pub.Publish(msg);
		ignerr << "Published Diffusive and Emmisive to " << visual_entity << std::endl;
		// }
```
For greater context here is gazebo visuals. I am trying to create a programmable ring of lights for this crazyflie drone model to match the real light rings. 

[image]

mhl787156, 2025-03-06:
Hi all, I finally figured out the sticking point which is that visual_config is a service! Hence it needs a service call! 

```cpp
gz::msgs::Visual msg;
gz::sim::Entity visual_entity = this->led_id_entity_map[i];
msg.set_id(visual_entity);

// Modify the material color (RGBA format)
gz::msgs::Material *mat = msg.mutable_material();
gz::msgs::Set(mat->mutable_diffuse(), gz::math::Color(r, g, b, 1.0));  // Red color
gz::msgs::Set(mat->mutable_emissive(), gz::math::Color(r, g, b, 1.0));  // Red color

std::function<void(const ignition::msgs::Boolean &, const bool)> cb =
	[](const ignition::msgs::Boolean &/*_rep*/, const bool _result)
{
	if (!_result)
	ignerr << "Error setting material color configuration"
			<< " on visual" << std::endl;
};

this->node.Request(this->visual_config_service_name, msg, cb);

ignerr << "Sent Service Call Diffusive and Emmisive to " << visual_entity << std::endl;
this->visual_config_pub.Publish(msg);
```

With that fixed, the whole stack works amazing. Hope this thread helps others! :) 

[image]

iche033, 2025-03-14:
ah didn't get a chance to respond earlier, but glad you're able to figure it out. Thanks for posting back the code snippet!

boat1470, 2025-12-17:
this is my model.sdf
```
xml
<?xml version='1.0'?>
<sdf version='1.6'>
  <model name='traffic_light'>
    <link name='traffic_light'>
      <pose>0 0 0 0 0 0</pose>
      <collision name='collision'>
        <geometry>
          <box>
            <size>0.12 0.025 0.25</size>
          </box>
        </geometry>   
      </collision>
      <visual name='visual_traffic_light'>
        <geometry>
          <box>
            <size>0.12 0.025 0.25</size>
          </box>
        </geometry>
        <material>
          <pbr>
            <metal>
              <albedo_map>model://turtlebot3_autorace_2020/traffic_light/materials/textures/traffic_light.png</albedo_map>
            </metal>
          </pbr>
          <ambient>0.3 0.3 0.3 1</ambient>
          <diffuse>0.7 0.7 0.7 1</diffuse>
          <specular>0.01 0.01 0.01 1</specular>
        </material>
      </visual>
    </link>
    <plugin name="turtlebot3_gazebo::TrafficLight" filename="libtraffic_light_plugin">
    </plugin>   
    <static>1</static>
  </model>
</sdf>
```
this is my traffic_light_plugin.cpp
```
#include "turtlebot3_gazebo/traffic_light_plugin.hpp"

#include <gz/plugin/Register.hh>
#include <gz/msgs/Utility.hh>
#include <gz/common/Console.hh>
#include <gz/transport/Node.hh>

#include <gz/sim/components/Name.hh>
#include <gz/sim/components/Material.hh>
#include <gz/sim/components/Visual.hh>
#include <gz/sim/Model.hh>
#include <gz/sim/Util.hh>
#include <gz/msgs/material.pb.h>
#include <gz/sim/World.hh>

using namespace turtlebot3_gazebo;

TrafficLight::TrafficLight()
: status(0),
  traffic_cycle(5.0) 
{

  this->textures = {
    "model://turtlebot3_autorace_2020/traffic_light/materials/textures/traffic_light_red.png",
    "model://turtlebot3_autorace_2020/traffic_light/materials/textures/traffic_light_yellow.png",
    "model://turtlebot3_autorace_2020/traffic_light/materials/textures/traffic_light_green.png"
  };
}

[... 125 lines cut ...]
    this->SetTexture(this->status);

    gzmsg << "[TrafficLight] Switched to texture: "
          << this->textures[this->status] << std::endl;
  }
}

// ต้อง Register plugin ด้วย macro นี้
GZ_ADD_PLUGIN(
  turtlebot3_gazebo::TrafficLight,
  gz::sim::System,
  gz::sim::ISystemConfigure,
  gz::sim::ISystemPreUpdate)

GZ_ADD_PLUGIN_ALIAS(
  turtlebot3_gazebo::TrafficLight,
  "turtlebot3_gazebo::TrafficLight")
```

[image]

now i can change color  but i want change this
<albedo_map>model://turtlebot3_autorace_2020/traffic_light/materials/textures/traffic_light.png</albedo_map>   intead color   

can you suggest me how to do