gazebosim/gz-sim #872: DiffDrive odometry values don't match robot pose values in Ignition Gazebo

jasmeet0915, 2021-06-21:
## Environment
* OS Version: Ubuntu 20.04 
* Source or binary build: Binary (Ignition Edifice)
    
## Description
* Expected behavior: 
Odometry values being published by the DiffDrive plugin to `/model/{name_of_model}/odometry` should match the pose value for the bot being displayed in ignition gazebo
* Actual behavior:
The values differ by a significant amount

The issue was discovered while working on a diff drive robot with ROS2 foxy, rviz2 and Ignition Edifice. The odometry values from ignition were bridged with ros2 using the ros_ign_bridge which was built from source for edifice.
The odometry values being published by the DiffDrive plugin (and being bridged to ROS2) are different from the pose values for the bot being displayed in Ignition Gazebo. 

Below is a GIF demonstrating the issue using the [diff_drive.sdf](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo5/examples/worlds/diff_drive.sdf) with vehicle_green removed.
## Steps to reproduce

1. Run the diff_drive.sdf using: `ign gazebo diff_drive.sdf`
2. Create a bridge for Twist message between /model/vehicle_blue/cmd_vel topic in ignition and /cmd_vel topic in ros2 to control the bot with ros2 teleop_twist_keyboard.
`ros2 run ros_ign_bridge parameter_server /model/vehicle_blue/cmd_vel@geometry_msgs/msg/Twist[ignition.msgs.Twist --ros-args --remap /model/vehicle_blue/cmd_vel:=/cmd_vel`
3. View the odometry values being published by diff_drive plugin using:
`ign topic -t /model/vehicle_blue/odometry -e`

## Output

[image]

chapulina, 2021-06-21:
The odometry published by the diff drive system is not ground truth, so some difference is expected. Odometry is calculated using [ignition::math::DiffDriveOdometry](https://ignitionrobotics.org/api/math/6.8/classignition_1_1math_1_1DiffDriveOdometry.html#details) and simulates how odometry would be calculated from the model's kinematics and joint state, as if there was no access to ground truth. If there's slip, for example, odometry will diverge from ground truth.

If you're looking for ground truth, you can try the [OdometryPublisher](https://ignitionrobotics.org/api/gazebo/5.0/classignition_1_1gazebo_1_1systems_1_1OdometryPublisher.html#details) system instead. Let us know if it helps!

jasmeet0915, 2021-06-22:
@chapulina Thanks for the reply! I guess the OdometryPublisher system plugin is not available in the binary version of edifice so I will give it a try after building ignition from source.
Also I noticed that the odometry values published by the DiffDrive plugin does not take into account the initial pose of the robot and always shows bot with x=0 and y=0 initially.

For eg here it can be seen that the bot is initialised at (2, 2) but odometry values show the bot (0, 0):
[image]

Also, here the bot was initialised at (2, 2) and moved close to 6m in the x axis. The position in ignition is seen close to 8m as it should be but the odometry values show close to 6m as they are calculated from (0, 0)
[image]

So I guess the DiffDrive plugin publishes the odometry values with wheel encoder as the odometry source as opposed to the diff_drive plugin in classic gazebo where the `<odometrySource>` defaults to world?

If this is the case then is it possible to add this information in the [diff_drive.sdf ](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo5/examples/worlds/diff_drive.sdf)example world as it could be a bit confusing for people migrating from classic gazebo.? Would love to make a PR for the same if it's fine!

chapulina, 2021-06-22:
> the OdometryPublisher system plugin is not available in the binary version of edifice

Stay tuned, it will eventually get there :wink: 

> So I guess the DiffDrive plugin publishes the odometry values with wheel encoder as the odometry source as opposed to the diff_drive plugin in classic gazebo where the <odometrySource> defaults to world?

That sounds right. It would be nice to give users an option to set the odometry source.

> is it possible to add this information in the diff_drive.sdf example world as it could be a bit confusing for people migrating from classic gazebo.?

+1

> Would love to make a PR for the same if it's fine!

Looking forward to it!

jasmeet0915, 2021-06-24:
>Stay tuned, it will eventually get there 😉

Sure!

>That sounds right. It would be nice to give users an option to set the odometry source.

Yes! 
I guess it would be nice to introduce an `<odometrySource>` parameter similar to the [diff_drive plugin in classic gazebo](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/231a7219b36b8a6cdd100b59f66a3df2955df787/gazebo_plugins/src/gazebo_ros_diff_drive.cpp#L104).

In the classic plugin, the odometry calculated by integrating the velocities is used when `odometrySource` is set to [`ENCODER`](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/231a7219b36b8a6cdd100b59f66a3df2955df787/gazebo_plugins/src/gazebo_ros_diff_drive.cpp#L412) (like the [ignition::math::DiffDriveOdometry](https://github.com/ignitionrobotics/ign-math/blob/9519ce74dae8b8c68303ef81e85fda27d9f4f494/src/DiffDriveOdometry.cc#L214) does) and it returns the world pose of the bot when `odometrySource` is set to [`WORLD`](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/231a7219b36b8a6cdd100b59f66a3df2955df787/gazebo_plugins/src/gazebo_ros_diff_drive.cpp#L418) which is what the [OdometryPublisher](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo5/src/systems/odometry_publisher/OdometryPublisher.cc) system plugin is doing I suppose.

Would love to work on developement of this feature too!