gazebosim/gz-sim #880: Differential drive for four-wheeled robots doesn't behave properly

ctampier, 2021-06-24:
## Environment
* OS Version: Ubuntu 20.04 (dockerized)
* Source or binary build? binary, ignition-fortress
  
## Description
* Expected behavior: robot should roughly move in circles
* Actual behavior: robot moves mostly straight, occasionally turns in 90 degrees 

## Steps to reproduce

1. run the world found in https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo5/examples/worlds/skid_steer_mecanum.sdf
2. run the command: ign topic -t "/model/vehicle_blue/cmd_vel" -m ignition.msgs.Twist -p "linear: {x: 0.5}, angular: {z: 0.5}"

## Output
[image]

chapulina, 2021-06-24:
I believe this is an issue with the DART version on Ubuntu Focal. On Ubuntu Bionic we use our fork of DART, which has better friction support. Can you check what `libdart` version you have installed?

Related issues: https://github.com/ignitionrobotics/ign-physics/issues/49. https://github.com/ignitionrobotics/ign-physics/issues/258

ctampier, 2021-06-25:
Thanks for the reply Louise. The version of DART in my container appears to be 6.9.2-2build4.
Unfortunately I don't know much about physics engines. Any hint on how it would be possible to overcome this issue?

chapulina, 2021-06-25:
> Any hint on how it would be possible to overcome this issue?

Some workarounds for now:

* Using Ubuntu Bionic
* Compiling our [DART fork](https://github.com/ignition-forks/dart) and all of Ignition from source on Focal

scpeters, 2022-04-18:
> > Any hint on how it would be possible to overcome this issue?
> 
> Some workarounds for now:
> 
> * Using Ubuntu Bionic
> * Compiling our [DART fork](https://github.com/ignition-forks/dart) and all of Ignition from source on Focal

one other workaround is using the [dartsim PPA](https://launchpad.net/~dartsim/+archive/ubuntu/ppa)