gazebosim/gz-sim #883: JointStatePublisher does not publish efforts

vatanaksoytezer, 2021-06-25:
## Environment
* OS Version: Ubuntu 20.04 
* Source or binary build?
Edifice Binary / ROS2 Foxy

Tested using both ODE and DART physics in the world file.

## Description
* Expected behavior: Joint States Publisher publishing efforts.
* Actual behavior:  Joint States Publisher (ignition::gazebo::systems::JointStatePublisher) does not publish efforts. Position and velocities look fine.

## Steps to reproduce
1. Launch any robot in ignition with ignition::gazebo::systems::JointStatePublisher plugin.
2. Start ROS2 parameter bridge
3. Echo /joint_states

Appreciate any insights / thoughts.

chapulina, 2021-06-25:
I believe we need to be able to get the torques from `ign-physics` first, see https://github.com/ignitionrobotics/ign-physics/issues/124.

vatanaksoytezer, 2021-06-25:
@chapulina Thanks for the fast reply! The issue seems a little inactive for some time. Is there anyway we can help this happen?

chapulina, 2021-06-25:
I believe the same reply applies here :sweat_smile:  https://github.com/ignitionrobotics/ign-sensors/issues/25#issuecomment-861709293

If you're willing to get this rolling before we get to it, you could try prototyping [Gazebo classic's approach](https://github.com/osrf/gazebo/blob/7a43a8d23b61cda57ab587b7f7032d9887ba9931/gazebo/physics/dart/DARTJoint.cc#L642-L688) into Ignition Physics. It's not that many lines of code. We're just busy with other priorities right now.

vatanaksoytezer, 2021-06-25:
That's reasonable thanks!

svepe, 2023-10-20:
I believe this is still an issue. I have a joint with a `JointController` in velocity mode and this is what I get from the JointStatePublisher:

```
joint {
  name: "joint"
  id: 4
  parent: "first_link"
  child: "second_link"
  pose {
    position {
    }
    orientation {
      w: 1
    }
  }
  axis1 {
    xyz {
      y: 1
    }
    limit_lower: -3.14
    limit_upper: 3.14
    position: 1.216194800124363e-08
    velocity: 5.1744987370740463e-10
  }
}
```
I'd expect to see effort here as well or am I wrong?

scpeters, 2023-10-24:
> I believe this is still an issue. I have a joint with a `JointController` in velocity mode and this is what I get from the JointStatePublisher:
> 
> ```
> joint {
>   name: "joint"
>   id: 4
>   parent: "first_link"
>   child: "second_link"
>   pose {
>     position {
>     }
>     orientation {
>       w: 1
>     }
>   }
>   axis1 {
>     xyz {
>       y: 1
>     }
>     limit_lower: -3.14
>     limit_upper: 3.14
>     position: 1.216194800124363e-08
>     velocity: 5.1744987370740463e-10
>   }
> }
> ```
> 
> I'd expect to see effort here as well or am I wrong? Changing the mode of the controller to force does not change things.

I see [code in JointStatePublisher](https://github.com/gazebosim/gz-sim/blob/gz-sim7_7.6.0/src/systems/joint_state_publisher/JointStatePublisher.cc#L287-L308) that copies data from any existing `JointForce` components into the `msgs::Joint` before publishing. Nothing about efforts will appear when echoing the topic if the values are precisely zero since this version of protobuf only shows non-zero values.

I think the underlying issue is that I don't see any system in gz-sim that writes data to a `JointForce` component.

svepe, 2023-10-26:
> I think the underlying issue is that I don't see any system in gz-sim that writes data to a `JointForce` component.

Not sure if it is relevant, but putting a JointForceTorque sensor does show the expected forces/torques on a joint, but that's different from the effort the joint is exerting itself.

JinbaoXiong, 2023-12-23:
Hallo, I have met the same problem. Is here any solution?

Fixit-Davide, 2024-03-28:
I have read the discussion in https://github.com/gazebosim/gz-physics/pull/143 and https://github.com/gazebosim/gz-sim/pull/952.
Any updates on this topic?
I have actually modified the file `Physics.cc` in order to update the `components::JointForce` but still I am unable to obtain the data regarding the efforts. Is this related to the physics engine that is used (i.e. Dart, Bullet, Bullet-featherstone)?