gazebosim/gz-sim #884: Cmake not Found when first compiling

nikitabeebe, 2021-06-25:
Downloading /ign-gazebo for the first time on a Windows computer after cloning /ign-cmake. However upon opening ign-gazebo in VS 2019, I get an error that it cannot find cmake. Where can I modify the directory path gazebo uses for referencing cmake, or is there another build/compile process I need for cmake?

[image]

chapulina, 2021-06-25:
Please use the issue template next time :wink: 

Did you see our Windows install instructions?

https://ignitionrobotics.org/docs/edifice/install_windows_src

nikitabeebe, 2021-06-25:
@chapulina I did, but I was unable to proceed with step 5 installing core dependencies. I believe it has to do with my network firewall, but every time I run it (even through administrator), I receive an error that the connection failed. 

I can access the site directly though, as both https://repo.anaconda.com/pkgs/main/win-64 and anaconda.com are not blocked. Is there a work around to manually install those repo packages and include them somehow into my conda environment?

[image]

traversaro, 2021-06-26:
> I can access the site directly though, as both https://repo.anaconda.com/pkgs/main/win-64 and anaconda.com are not blocked. Is there a work around to manually install those repo packages and include them somehow into my conda environment?

As a general comment, it seems that your install command is not correct, as the one provided in the docs is:
~~~
conda install cmake git vcstool curl pkg-config colcon-common-extensions eigen freeimage gts glib dlfcn-win32 ffmpeg ruby tinyxml2 tinyxml protobuf urdfdom zeromq cppzmq ogre jsoncpp libzip qt --channel conda-forge
~~~
In particular, the final `--channel conda-forge` option is quite important, as otherwise you would install packages from the main anaconda repo that have severe usage restriction (see )

> I can access the site directly though, as both https://repo.anaconda.com/pkgs/main/win-64 and anaconda.com are not blocked. Is there a work around to manually install those repo packages and include them somehow into my conda environment?

If you are able to access them via the browser (in particular the file https://repo.anaconda.com/pkgs/main/win-64/repodata.json) but not from the conda command, my guess (but I could be wrong) is that you have some kind of company proxy configured in the browser, but not for conda. For some documentation on how to configure conda behind a proxy, see https://conda.io/projects/conda/en/latest/user-guide/configuration/use-condarc.html#config-proxy .

nikitabeebe, 2021-06-27:
Thank you for the comment. I should have clarified, I tried the entire line of command and had the same error. But after following those proxy server settings (I am using this on a company computer, so I assume your guess about company proxy being automatically configured is likely the culprit), I get the same error. 

I've edited the .condarc file that shows: (kept the exact proxy address out since it is a company computer)
`channels:
  - conda-forge
  - defaults

proxy_servers:
    http: http://proxyvip.....com:3128

ssl_verify: true`

But I still receive an error about accessing the site (that I can still access by entering into the browser)
`(base) C:\ProgramData\Anaconda3\Scripts>conda install cmake git vcstool curl pkg-config colcon-common-extensions eigen freeimage gts glib dlfcn-win32 ffmpeg ruby tinyxml2 tinyxml protobuf urdfdom zeromq cppzmq ogre jsoncpp libzip qt --channel conda-forge
Collecting package metadata (current_repodata.json): failed

CondaHTTPError: HTTP 000 CONNECTION FAILED for url <https://conda.anaconda.org/conda-forge/win-64/current_repodata.json>
Elapsed: -

An HTTP error occurred when trying to retrieve this URL.
HTTP errors are often intermittent, and a simple retry will get you on your way.
'https://conda.anaconda.org/conda-forge/win-64'`

Is this an error still with the proxy settings?

scpeters, 2026-09-08:
closing since this is an old issue; please reopen if the issue persists

closed 2026-09-08 by scpeters