gazebosim/gz-sim #885: New release for Edifice

Tobias-Fischer, 2021-06-25:
Hi there,
We are in the process of packaging Edifice in conda-forge (see https://github.com/conda-forge/libignition-gazebo-feedstock/pull/5). At the moment Windows fails, which I think is because https://github.com/ignitionrobotics/ign-gazebo/pull/758 is not contained in the 5.0.0 release. Any chance to release a new version so we don't need to apply patches?

/cc @traversaro

chapulina, 2021-06-25:
Sure thing, I'll work on a release as soon as https://github.com/ignitionrobotics/ign-gazebo/pull/877 is in :+1:

chapulina, 2021-06-30:
The GitHub release is out, debs in the oven: https://build.osrfoundation.org/job/ign-gazebo5-debbuilder/

closed 2021-06-30 by chapulina

Tobias-Fischer, 2021-06-30:
Many thanks @chapulina!

PR #896 (2021-06-30): 🎈 5.1.0
# 🎈 Release

Preparation for 5.1.0 release.

Comparison to 5.0.0: https://github.com/ignitionrobotics/ign-gazebo/compare/ignition-gazebo5_5.0.0...ign-gazebo5

I did some manual testing with the pre-release from #892 and everything seems to be in place :tada: 

Closes #885

## Checklist
- [x] Asked team if this is a good time for a release
- [x] There are no changes to be ported from the previous major version - there's always something, a big pass was done in #877
- [x] No PRs targeted at this major version are close to getting in - this release is already large enough that I don't thi [... cut]