gazebosim/gz-sim #886: command line tool not available ( Ignition Fortress)

ibrahiminfinite, 2021-06-25:
## Environment
* OS Version: Ubuntu 20.04 
* Source build , latest commit #875 
 
## Description
* Expected behavior: `ign` command line should work after sourcing workspace
* Actual behavior: 
             1. `ign` command not found error, 
             2.  `. ~/workspace/install/setup.bash`  does not append to path

## Steps to reproduce

1. wget https://raw.githubusercontent.com/ignition-tooling/gazebodistro/master/collection-fortress.yaml
2. vcs import < collection-fortress.yaml
3. colcon build --cmake-args -DCMAKE_BUILD_TYPE=Debug --merge-install
4. .  ~/workspace/install/setup.bash
5. ign gazebo -g

Additional info :  
I have Ignition citadel , also source build
The citadel command line tool works if I source the citadel workspace

chapulina, 2021-06-25:
Can you see the `install/bin/ign` executable in your workspace?

After you compile all libraries, can you check `ignition-tools`'s build logs? Normally you'd have a line like:

```
[0.167s] -- Installing: /home/chapulina/dev_bionic/ws_fortress/install/bin/ign
```

ibrahiminfinite, 2021-06-26:
Hi @chapulina 
1. The `install/bin` folder does not exist
2. `ignition-tools` log does not contain a line like that

This was the end output of build process
```
Summary: 16 packages finished [15min 29s]
  3 packages had stderr output: ignition-physics5 ignition-rendering6 sdformat12
```

chapulina, 2021-06-26:
I just ran into this as well in a fresh workspace, here's a fix: https://github.com/ignitionrobotics/ign-tools/pull/56

ibrahiminfinite, 2021-06-26:
Update :

I just did a clean build of the citadel (including download of the sources) to check if it is still working.
There is a `install\bin` directory but there is no `ign`  file which used to be there before.

ibrahiminfinite, 2021-06-26:
Hi @chapulina 

> I just ran into this as well in a fresh workspace, here's a fix: [ignitionrobotics/ign-tools#56](https://github.com/ignitionrobotics/ign-tools/pull/56)

This fixed it for both citadel and fortress
Command lIne works now

Thankyou

closed 2021-06-28 by mjcarroll