gazebosim/gz-sim #891: Gazebo crashes with Edifice binaries

adlarkin, 2021-06-28:
## Environment
* OS Version: Ubuntu 18.04 (also have tried Ubuntu 20.04, and get the same result)
* Source or binary build?
  Binary - Edifice
  - ign-tools 1.2.0
  - ign-cmake 2.8.0
  - ign-common 4.1.0
  - ign-fuel-tools 6.0.0
  - ign-gazebo 5.0.0
  - ign-gui 5.1.0
  - ign-launch 4.0.0
  - ign-math 6.8.0
  - ign-msgs 7.1.0
  - ign-physics 4.1.0
  - ign-plugin 1.2.0
  - ign-rendering 5.1.0
  - ign-sensors 5.0.0
  - ign-transport 10.0.0
  - ign-utils 1.0.0
  - sdformat 11.2.0

## Description
* Expected behavior: Gazebo should run
* Actual behavior: Gazebo crashes

I believe that there is an issue with the Edifice binaries. I have tried to run gazebo with the Edifice binaries on both Ubuntu bionic and focal (using docker containers), but it always crashes. I don't believe that it's an issue with my docker setup because I am able to run the Citadel, Dome and Fortress binaries inside docker without any issues. I also was able to run Edifice in docker if I built it from source. I also don't think it's an issue with my system because I tried using the Edifice binaries on my laptop as well, and it still crashes there :upside_down_face: 

## Steps to reproduce

1. Download the edifice binaries: https://ignitionrobotics.org/docs/edifice/install_ubuntu
2. Run gazebo: `ign gazebo shapes.sdf`

## Output

Here's the backtrace I am seeing from the crash.
<details>
<summary>backtrace</summary>

```
/usr/lib/ruby/ignition/cmdgazebo5.rb:431: [BUG] Segmentation fault at 0x0000000000000101
ruby 2.5.1p57 (2018-03-29 revision 63029) [x86_64-linux-gnu]

-- Control frame information -----------------------------------------------
c:0007 p:---- s:0076 e:000075 CFUNC  :call
c:0006 p:0017 s:0054 e:000053 METHOD /usr/lib/ruby/ignition/cmdgazebo5.rb:431
c:0005 p:0160 s:0048 e:000047 BLOCK  /usr/lib/ruby/ignition/cmdgazebo5.rb:460 [FINISH]
c:0004 p:---- s:0045 e:000044 CFUNC  :fork
c:0003 p:0675 s:0041 e:000040 METHOD /usr/lib/ruby/ignition/cmdgazebo5.rb:456
c:0002 p:0695 s:0020 E:000730 EVAL   /usr/bin/ign:275 [FINISH]
c:0001 p:0000 s:0003 E:002650 (none) [FINISH]

-- Ruby level backtrace information ----------------------------------------
/usr/bin/ign:275:in `<main>'
/usr/lib/ruby/ignition/cmdgazebo5.rb:456:in `execute'
/usr/lib/ruby/ignition/cmdgazebo5.rb:456:in `fork'
/usr/lib/ruby/ignition/cmdgazebo5.rb:460:in `block in execute'
/usr/lib/ruby/ignition/cmdgazebo5.rb:431:in `runServer'
/usr/lib/ruby/ignition/cmdgazebo5.rb:431:in `call'

-- Machine register context ------------------------------------------------
 RIP: 0x00007f8dce992e02 RBP: 0x00007fff4a38dc40 RSP: 0x00007fff4a38d8e0
 RAX: 0x00000000ffffffec RBX: 0x0000000000000031 RCX: 0x0000000000000062
 RDX: 0x0000000000000000 RDI: 0x00007f8dcf1a4ce0 RSI: 0x00007f8dcd988ed1
  R8: 0x0000000000001000  R9: 0x00007f8dd13e5690 R10: 0x0000000000000065
 R11: 0x00007f8dce9931f0 R12: 0x000056414971b0b0 R13: 0x00007f8dcf1a4ce0
 R14: 0x000056414971bd10 R15: 0x000056414888e0a8 EFL: 0x0000000000010213

-- C level backtrace information -------------------------------------------
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7f8dd1825a15) [0x7f8dd1825a15]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7f8dd1825c4c) [0x7f8dd1825c4c]
[... 674 lines cut ...]
7f8dd1b08000-7f8dd1b09000 rw-p 002a0000 103:03 62134437                  /usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5.1
7f8dd1b09000-7f8dd1b19000 rw-p 00000000 00:00 0 
7f8dd1b19000-7f8dd1b42000 r-xp 00000000 103:03 61607388                  /lib/x86_64-linux-gnu/ld-2.27.so
7f8dd1bf9000-7f8dd1c21000 r--s 00000000 103:03 62136514                  /usr/lib/x86_64-linux-gnu/libignition-gazebo5-ign.so.5.0.0
7f8dd1c21000-7f8dd1d29000 rw-p 00000000 00:00 0 
7f8dd1d29000-7f8dd1d33000 r--s 00000000 103:03 62654925                  /usr/lib/x86_64-linux-gnu/ruby/2.5.0/fiddle.so
7f8dd1d33000-7f8dd1d3b000 r--s 00000000 103:03 61735244                  /usr/lib/x86_64-linux-gnu/libffi.so.6.0.4
7f8dd1d3b000-7f8dd1d3d000 r--s 00000000 103:03 62134551                  /usr/bin/ruby2.5
7f8dd1d3d000-7f8dd1d3e000 ---p 00000000 00:00 0 
7f8dd1d3e000-7f8dd1d42000 rw-p 00000000 00:00 0 
7f8dd1d42000-7f8dd1d43000 r--p 00029000 103:03 61607388                  /lib/x86_64-linux-gnu/ld-2.27.so
7f8dd1d43000-7f8dd1d44000 rw-p 0002a000 103:03 61607388                  /lib/x86_64-linux-gnu/ld-2.27.so
7f8dd1d44000-7f8dd1d45000 rw-p 00000000 00:00 0 
7fff49b93000-7fff4a392000 rw-p 00000000 00:00 0                          [stack]
7fff4a3d7000-7fff4a3db000 r--p 00000000 00:00 0                          [vvar]
7fff4a3db000-7fff4a3dd000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 --xp 00000000 00:00 0                  [vsyscall]

[NOTE]
You may have encountered a bug in the Ruby interpreter or extension libraries.
Bug reports are welcome.
For details: http://www.ruby-lang.org/bugreport.html
```

</details>

adlarkin, 2021-06-28:
I should also mention that the crash happens when using both stable and nightly binaries for Edifice (the backtrace I shared above is for the stable binaries).

chapulina, 2021-06-28:
The backtrace points to when reading a pose from SDF. In fact, when I load 

```
  <sdf version="1.6">
   <world name="empty">
  
      <model name="ground_plane">
        <!--pose>0 0 10 0 0 0</pose-->
        <static>true</static>
        <link name="link">
        </link>
      </model>
  
    </world>
  </sdf>

```

it works, but uncommenting the pose it crashes.

It smells like an ABI incompatibility issue between `libsdformat` and another downstream library.

chapulina, 2021-06-28:
I don't see anything that's potentially ABI-breaking on the latest SDFormat release into Edifice, that came out 5 days ago: https://github.com/osrf/sdformat/pull/599

chapulina, 2021-06-29:
The crash is fixed with the latest SDF 11.2.1 release :tada:

closed 2021-06-29 by chapulina