gazebosim/gz-sim #894: 👩‍🌾 Wheel friction tests started failing on Edifice

chapulina, 2021-06-29:
## Environment
* OS Version: Ubuntu and macOS
* Source or binary build? Source, `ign-gazebo5` branch, starting since at least 7876ca3d1c1bbcf3399b1f7bd089539ae04ed781

## Description
* Expected behavior: tests pass :)
* Actual behavior: tests fail :(

The tests only fail for `ign-gazebo5`:

* `ign-gazebo5`: [image]
* `ign-gazebo4`: [image]
* `ign-gazebo3`: [image]

## Steps to reproduce

Run CI for the `ign-gazebo5` branch, for example:

* [GitHub actions](https://github.com/ignitionrobotics/ign-gazebo/runs/2918222829?check_suite_focus=true)
* [Jenkins Ubuntu](https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo5-bionic-amd64/18/
* [Jenkins macOS](https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo5-homebrew-amd64/18/)

I haven't been able to reproduce the failures locally.

## Output

The tests are failing because the vehicle is not being driven correctly:

```
/var/lib/jenkins/workspace/ignition_gazebo-ci-ign-gazebo5-bionic-amd64/ign-gazebo/test/integration/ackermann_steering_system.cc:168
Expected: (poses[0].Pos().X()) < (poses[3999].Pos().X()), actual: -6.245e-15 vs -0.000480315
```

```
/var/lib/jenkins/workspace/ignition_gazebo-ci-ign-gazebo5-bionic-amd64/ign-gazebo/test/integration/breadcrumbs.cc:186
Expected: (poseVehicle->Data().Pos().X()) > (0.4), actual: 4.91237e-05 vs 0.4
```

```
/var/lib/jenkins/workspace/ignition_gazebo-ci-ign-gazebo5-bionic-amd64/ign-gazebo/test/integration/diff_drive_system.cc:172
Expected: (poses[0].Pos().Y()) < (poses[3999].Pos().Y()), actual: 3.99537e-12 vs -3.05904e-06
```

```
/var/lib/jenkins/workspace/ignition_gazebo-ci-ign-gazebo5-bionic-amd64/ign-gazebo/test/integration/physics_system.cc:538
Value of: isIncreasingYPos
  Actual: false
Expected: true
(7.62527, 1), (7.62527, 0.1), (7.62527, 0.01), 
```

```
/var/lib/jenkins/workspace/ignition_gazebo-ci-ign-gazebo5-bionic-amd64/ign-gazebo/test/integration/performer_detector.cc:158
Expected equality of these values:
  4u
    Which is: 4
  this->poseMsgs.size()
    Which is: 0
```

```
/var/lib/jenkins/workspace/ignition_gazebo-ci-ign-gazebo5-bionic-amd64/ign-gazebo/test/integration/wheel_slip.cc:581
The difference between noSlipLinearSpeed - worldVelTrisphere0->Data()[0] and 0.0 is 4.9019999999999797, which exceeds 5e-3, where
noSlipLinearSpeed - worldVelTrisphere0->Data()[0] evaluates to 4.9019999999999797,
0.0 evaluates to 0, and
5e-3 evaluates to 0.0050000000000000001.
```

---

https://github.com/osrf/buildfarmer/issues/207

chapulina, 2021-06-29:
See @scpeters 's comment here: https://github.com/ignitionrobotics/ign-gazebo/pull/892#issuecomment-870170106

This may have been caused by libSDFormat 11.2.0's ABI break.

scpeters, 2021-06-29:
libsdformat 11.2.1 has been released:

* [image] https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo5-homebrew-amd64/19/

chapulina, 2021-06-29:
Great! here's Linux: [image]

chapulina, 2021-06-29:
Yes! It was the ABI break, the builds are fixed :tada:

closed 2021-06-29 by chapulina