gazebosim/gz-sim #909: Heightmap models crash Ignition

MarqRazz, 2021-07-09:
## Environment
* OS Version: Ubuntu 20.04 
* Source or binary build?
  Edifice Binary / ROS2 Foxy

## Description
* Expected behavior: Load heightmap sdf 
* Actual behavior: error is printed and Ignition crashes

## Steps to reproduce

1. Add [this SDF](https://app.ignitionrobotics.org/chapulina/fuel/models/Heightmap%20Bowl) model to your world file
```
<include>
      <uri>
      https://fuel.ignitionrobotics.org/1.0/chapulina/models/Heightmap Bowl
      </uri>
</include>
```
2. Launch Ignition with the above world file

I also have a sdf created using [this tutorial](https://vimeo.com/58409707) that I have used in Gazebo classic that throws the same error.

## Output
```
[ign gazebo-1] [Err] [SDFFeatures.cc:781] The geometry element of collision [collision] couldn't be created
[ign gazebo-1] terminate called after throwing an instance of 'std::bad_optional_access'
[ign gazebo-1]   what():  bad optional access
```

chapulina, 2021-07-09:
Heightmap physics support hasn't been merged yet, but it's close. See #661.

It shouldn't crash though, so there's definitely something wrong. Proper support is very close to getting in, so I think we should just wait for that to get in.

vatanaksoytezer, 2021-08-02:
Heightmaps are merged and seems to work fine in my setup with latest edifice branches @chapulina @MarqRazz I think we can close this.

closed 2021-08-02 by chapulina