gazebosim/gz-sim #910: Scene background and ambient light values are ignored by Ogre2

azeey, 2021-07-09:
## Environment
* OS Version: Ubuntu 18.04
* Source or binary build? source, `sdf9` branch

## Description
* Expected behavior: The background and ambient light values in `<scene>` should be used by Ogre2 to light the scene
* Actual behavior: The values in `<scene>` are ignored. Instead, `Ogre2` uses `<background_color>` and `<ambient_light>` values inside `<plugin filename="GzScene3D" >`. On the contrary, `Ogre` seems to only use the values in `<scene>`

## Steps to reproduce

1. Run the attached SDFormat file
2. Change `<engine>` to either `ogre` or `ogre2`
3. Update values in `<background>` and `<ambient>` in `<scene>` if `engine` is Ogre2 and notice no changes
4. 3. Update values in `<background_color>` and `<ambient_light>` in `<plugin filename="GzScene3D" >` if `engine` is Ogre and notice no changes.

## Output

### Ogre2: 
[image]

### Ogre:
[image]

iche033, 2021-07-09:
I know at least for background color, we are using the param in `GzScene3D` to set the background color for the GUI and the one in `<scene>` for sensors so that they can have different background colors.

chapulina, 2021-07-19:
We need to document this better, but as @iche033 mentioned, the goal is that the parameters within the plugin override the `<scene>` tag.

However, the current code is setting first the color from the plugin and then overriding that with the one from `<scene>` here:

https://github.com/ignitionrobotics/ign-gazebo/blob/d7741f6c3ec307e465275033434ceea286546aa7/src/rendering/RenderUtil.cc#L644

The difference in behaviour we're seeing between `ogre` and `ogre2` is that the former is actually doing the right thing, while the latter isn't. It looks like `Ogre2Camera`'s background isn't updated when it's set through the API. I think this piece of code isn't doing what it should:

https://github.com/ignitionrobotics/ign-rendering/blob/44a19485aa32f48fd116ad6b5e938996eb5fbd90/ogre2/src/Ogre2RenderTarget.cc#L570-L583

athenaz2, 2024-06-26:
Issue of the sim using `<plugin filename="GzScene3D">`'s `<background_color>` and `<ambient_light>` (as opposed to `<scene>`'s `<background>` and `<ambient>`) to light the scene cannot be reproduced, so no longer a concern. The [`GzScene3D` GUI plugin was deprecated for `MinimalScene`](https://github.com/gazebosim/gz-sim/blob/gz-sim8/Migration.md) during transition from Gazebo 5 to 6. Regardless of `Ogre` version and whether or not SDF file has `GzScene3D`, sim’s background and ambient colors use values from `<scene>`’s `<background>` and `<ambient>`. If use `MinimalScene` plugin (default uses `Ogre2`, but even if original `Ogre`), sim’s background and ambient colors still use values from `<scene>`’s `<background>` and `<ambient>`.

Original `test_scene.sdf` file (that uses `GzScene3D`), shown with `Ogre` vs. `Ogre2`.

[image]
[image]

Same SDF file with `<plugin filename="GzScene3D">` removed, which defaults sim to use GUI plugins from `gui.config`, including `MinimalScene` with `Ogre2`.

[image]

iche033, 2024-06-26:
As mentioned in https://github.com/gazebosim/gz-sim/issues/910#issuecomment-877422638, can we make it so that the `<background_color>` field in gui MinimalPlugin overrides the `<scene>` tag for the GUI? The goal is to be able to use a different background color in GUI compared to camera sensors.

One example world that you can use for testing is: [sensors_demo.sdf](https://github.com/gazebosim/gz-sim/blob/gz-sim8/examples/worlds/sensors_demo.sdf)

azeey, 2024-07-01:
The original issue was about the `<scene>` tag not having an effect on the GUI. As @athenaz2 confirmed, the issue is now different. Instead, the `<scene>` tag currently takes precedence and there is no way to assign different backgrounds for the GUI and sensors. I'll close this issue and @athenaz2 will create a separate issue for that.

closed 2024-07-01 by azeey