gazebosim/gz-sim #920: No camera/render on ubuntu (WSLg + vGPU)

FirefoxMetzger, 2021-07-17:
I am trying to set up Ignition on a Ubuntu VM (WSL2) that has GPU hardware acceleration for both rendering and computation (via the new WSLg). It works almost fine; I get a window that is rendered, responsive and GPU accelerated; however, I don't seem to be able to render the main camera's view (which instead remains black). The same setup using `llvmpipe` as a (CPU-based) renderer works without issue.

I remember having encountered a similar situation when I tried to install ignition on bare Windows, but I can't find the issue just now (perhaps someone else knows and can link it :) ).

There is another issue that the entire VM crashes whenever I close the GUI while a simulation is running (does not happen with other GUI windows), but that one is secondary for now.

## Environment
* OS Version: Ubuntu 20.04.2 LTS (Kernel 5.10.43.3-microsoft-standard-WSL2)
* Source or binary build? Tested on: Fortress (binary) Edifice (binary + source)
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system.
    - Rendering plugin: ogre2
      - [ ] GUI rendering error: None reported, though there are warning messages
    - Generally, mention all circumstances that might affect rendering capabilities:
      - [ ] running on real hardware
      - [x] running in virtual machine
      - [x] GPU is concurrently used for other tasks
        - [x] desktop acceleration
        - [x] video decoding (i.e. a playing Youtube video)
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
      - [ ] other
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
        <details>
        <summary>command output </summary>

        ```
        firefoxmetzger@DESKTOP-2SO47AE:~/workspace$ LANG=C lspci -nn | grep VGA  # might require installing pciutils
        firefoxmetzger@DESKTOP-2SO47AE:~/workspace$ echo "$DISPLAY"
        :0
        firefoxmetzger@DESKTOP-2SO47AE:~/workspace$ LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
        direct rendering: Yes
            Preferred profile: core (0x1)
            Max core profile version: 3.3
            Max compat profile version: 3.1
            Max GLES1 profile version: 1.1
            Max GLES[23] profile version: 3.0
        OpenGL vendor string: Microsoft Corporation
        OpenGL renderer string: D3D12 (NVIDIA GeForce RTX 3070)
        OpenGL core profile version string: 3.3 (Core Profile) Mesa 21.1.5 - kisak-mesa PPA
        OpenGL core profile shading language version string: 3.30
        OpenGL core profile context flags: (none)
        OpenGL core profile profile mask: core profile
        OpenGL version string: 3.1 Mesa 21.1.5 - kisak-mesa PPA
        OpenGL shading language version string: 1.40
        OpenGL context flags: (none)
        OpenGL ES profile version string: OpenGL ES 3.0 Mesa 21.1.5 - kisak-mesa PPA
        OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
        firefoxmetzger@DESKTOP-2SO47AE:~/workspace$ ps aux | grep Xorg
        firefox+ 10271  0.0  0.0   8164   728 pts/0    S+   12:14   0:00 grep --color=auto Xorg
        firefoxmetzger@DESKTOP-2SO47AE:~/workspace$ sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
        [sudo] password for firefoxmetzger:
        
        X.Org X Server 1.20.9
        X Protocol Version 11, Revision 0
        Build Operating System: Linux 4.15.0-140-generic x86_64 Ubuntu
        Current Operating System: Linux DESKTOP-2SO47AE 5.10.43.3-microsoft-standard-WSL2 #1 SMP Wed Jun 16 23:47:55 UTC 2021 x86_64
        Kernel command line: initrd=\initrd.img panic=-1 nr_cpus=16 swiotlb=force pty.legacy_count=0
        Build Date: 08 April 2021  12:29:22PM
        xorg-server 2:1.20.9-2ubuntu1.2~20.04.2 (For technical support please see http://www.ubuntu.com/support)
        Current version of pixman: 0.38.4
                Before reporting problems, check http://wiki.x.org
                to make sure that you have the latest version.
        ```
        
        </details>

    - [x] Please, attach the ogre.log or ogre2.log file from  `~/.ignition/rendering`: [ogre2.log](https://github.com/ignitionrobotics/ign-gazebo/files/6834563/ogre2.log)

    
<details>
[... 452 lines cut ...]
0:312(28): warning: `nNormal' used uninitialized
0:315(14): warning: `material' used uninitialized
0:315(30): warning: `material' used uninitialized
0:334(18): warning: `F0' used uninitialized
0:334(60): warning: `F0' used uninitialized
0:337(37): warning: `material' used uninitialized
0:341(21): warning: `material' used uninitialized
0:342(45): warning: `material' used uninitialized
0:343(50): warning: `material' used uninitialized
0:351(69): warning: `diffuseCol' used uninitialized
```

</details>

## Description
* Expected behavior: the examples `shapes.sdf` and `lights.sdf` load correctly and the GUI camera renders the scene.
* Actual behavior: The camera view remains gray.

[image]

## Steps to reproduce
1. Setup WSL2 + https://github.com/microsoft/wslg
2. Install ignition using preferred method following the instructions (I tried source and binary versions)
3. `ign shapes.sdf -v3`
4. Wonder what is going wrong.

traversaro, 2021-10-12:
There is a similar problem when trying to run ignition gazebo 6.0.0 (installed via apt) on WSL2 with X server (no WSLg). I tried with both `X410` and `Xming`, but the output is the same: nothing is rendered. Classic Gazebo works fine on both WSL1/2 with X server (there were some issue back in time, but they were solved, see: https://github.com/osrf/gazebo/issues/2351 ) and also WSLg (Classic Gazebo was actually one of the applications used by Microsoft to showcase https://devblogs.microsoft.com/commandline/the-initial-preview-of-gui-app-support-is-now-available-for-the-windows-subsystem-for-linux-2/).

I wonder if it may be worth to open an issue also in  https://github.com/microsoft/wslg for this issue.

FirefoxMetzger, 2021-10-13:
@traversaro Good point; I have not tested Citadel. Mainly because gym-ignition targets the latest ign-gazebo, so it wouldn't do much for my use-case. I will give it a try next week (currently at a conference) and report back on how it goes.

FirefoxMetzger, 2021-10-16:
@traversaro I just tried Citadel with WSLg. It doesn't render `shapes.sdf` either (on my machine) and looks the same as the issue I described in the original post. I do, however, get an error message that I don't get with the other ignition versions:

```
[GUI] [Err] [Application.cc:324] Failed to load plugin [Lights] : couldn't find shared library.
```

So it could simply be that the light sources are not working and, because of that, I only ever see a gray screen.

Here is the full log for reference in case that is interesting:

<details>
<summary>full log</summary>

```
$ ign gazebo -v4 shapes.sdf
[Msg] Ignition Gazebo GUI    v3.9.0
[Msg] Ignition Gazebo Server v3.9.0
[Msg] Loading SDF world file[/usr/share/ignition/ignition-gazebo3/worlds/shapes.sdf].
[Dbg] [Application.cc:87] Initializing application.
[GUI] [Dbg] [Application.cc:427] Create main window
[Dbg] [EntityComponentManager.cc:641] Using components of type [2251689575469537287] / [ign_gazebo_components.World].
[Dbg] [EntityComponentManager.cc:641] Using components of type [13994732549916512682] / [ign_gazebo_components.Name].
[Dbg] [EntityComponentManager.cc:641] Using components of type [12592746352568925681] / [ign_gazebo_components.Gravity].
[Dbg] [EntityComponentManager.cc:641] Using components of type [2188341333082264598] / [ign_gazebo_components.Physics].
[Dbg] [EntityComponentManager.cc:641] Using components of type [13224937992534617849] / [ign_gazebo_components.MagneticField].
[Dbg] [EntityComponentManager.cc:641] Using components of type [3630648173860223239] / [ign_gazebo_components.PhysicsEnginePlugin].
[Dbg] [EntityComponentManager.cc:641] Using components of type [17459188283658606303] / [ign_gazebo_components.RenderEngineServerPlugin].
[Dbg] [EntityComponentManager.cc:641] Using components of type [8705992680619689917] / [ign_gazebo_components.RenderEngineGuiPlugin].
[Dbg] [EntityComponentManager.cc:641] Using components of type [8753193699724811771] / [ign_gazebo_components.Wind].
[Dbg] [EntityComponentManager.cc:641] Using components of type [12173050716021724529] / [ign_gazebo_components.WorldLinearVelocity].
[Dbg] [EntityComponentManager.cc:641] Using components of type [15943768124495574352] / [ign_gazebo_components.WorldLinearVelocitySeed].
[Dbg] [EntityComponentManager.cc:641] Using components of type [3297509811873971798] / [ign_gazebo_components.ParentEntity].
[Dbg] [EntityComponentManager.cc:641] Using components of type [17100615127981600159] / [ign_gazebo_components.Scene].
[Dbg] [EntityComponentManager.cc:641] Using components of type [17605309075052480649] / [ign_gazebo_components.Atmosphere].
[Dbg] [EntityComponentManager.cc:641] Using components of type [8064491505919932473] / [ign_gazebo_components.Level].
[Dbg] [EntityComponentManager.cc:641] Using components of type [2668898242563798256] / [ign_gazebo_components.DefaultLevel].
[Dbg] [EntityComponentManager.cc:641] Using components of type [11371360182141354106] / [ign_gazebo_components.LevelEntityNames].
[Dbg] [EntityComponentManager.cc:641] Using components of type [4981278897826323946] / [ign_gazebo_components.WorldSdf].
[Dbg] [EntityComponentManager.cc:641] Using components of type [6687176221774458630] / [ign_gazebo_components.Model].
[Dbg] [EntityComponentManager.cc:641] Using components of type [6612894081701502240] / [ign_gazebo_components.Pose].
[Dbg] [EntityComponentManager.cc:641] Using components of type [8546580419506082455] / [ign_gazebo_components.Static].
[Dbg] [EntityComponentManager.cc:641] Using components of type [9712747055438129860] / [ign_gazebo_components.WindMode].
[Dbg] [EntityComponentManager.cc:641] Using components of type [5661073481138181711] / [ign_gazebo_components.SelfCollide].
[Dbg] [EntityComponentManager.cc:641] Using components of type [11683062252779233161] / [ign_gazebo_components.SourceFilePath].
[Dbg] [EntityComponentManager.cc:641] Using components of type [5081358965268446661] / [ign_gazebo_components.Link].
[Dbg] [EntityComponentManager.cc:641] Using components of type [8112400427272910195] / [ign_gazebo_components.Inertial].
[Dbg] [EntityComponentManager.cc:641] Using components of type [16454635107327670381] / [ign_gazebo_components.Visual].
[Dbg] [EntityComponentManager.cc:641] Using components of type [13011964647677164955] / [ign_gazebo_components.CastShadows].
[Dbg] [EntityComponentManager.cc:641] Using components of type [13440282432131634483] / [ign_gazebo_components.Transparency].
[Dbg] [EntityComponentManager.cc:641] Using components of type [17121648710877364109] / [ign_gazebo_components.Geometry].
[Dbg] [EntityComponentManager.cc:641] Using components of type [9853217982010720764] / [ign_gazebo_components.Material].
[Dbg] [EntityComponentManager.cc:641] Using components of type [17938588655714334139] / [ign_gazebo_components.Collision].
[Dbg] [EntityComponentManager.cc:641] Using components of type [9225962031573086509] / [ign_gazebo_components.CollisionElement].
[Dbg] [EntityComponentManager.cc:641] Using components of type [10522242218202596205] / [ign_gazebo_components.CanonicalLink].
[Dbg] [EntityComponentManager.cc:641] Using components of type [11536476718181283925] / [ign_gazebo_components.ModelSdf].
[Dbg] [EntityComponentManager.cc:641] Using components of type [3866641186784191835] / [ign_gazebo_components.Light].
[Msg] Loaded level [3]
[Msg] No systems loaded from SDF, loading defaults
[... cut ...]
[GUI] [Dbg] [RenderUtil.cc:1201] Create scene [scene]
[GUI] [Dbg] [Gui.cc:292] Shutting down ign-gazebo-gui
[GUI] [Dbg] [Application.cc:130] Terminating application.
[GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[GUI] [Dbg] [Scene3D.cc:1694] Destroy scene [scene]
[Dbg] [SignalHandler.cc:141] Received signal[2].
[Dbg] [ServerPrivate.cc:108] Server received signal[2]
[Dbg] [ign.cc:362] Shutting down ign-gazebo-server
[Dbg] [SimulationRunner.cc:526] Exiting postupdate worker thread (0)
```

</details>

FirefoxMetzger, 2021-10-16:
@traversaro I agree, there could be a high chance that it is a Wayland vs X problem. 

After my current project finishes, I will hard-reset my Linux machine at work. I'll use that as an opportunity to get a clean bare-bones Ubuntu 20.04 + Wayland. Let's see what happens then.

blakermchale, 2021-11-09:
I've been trying to use ign-gazebo on WSLg with WIndows 11 also and tried with the headless command `ign gazebo shapes.sdf --headless-rendering`. It still doesn't render the scene.

chapulina, 2021-11-09:
@blakermchale , what version of Ignition and Ogre 2 do you have installed? Are you able to get Ogre logs when running with headless rendering?

blakermchale, 2021-11-10:
When I run `ign gazebo --version` I get: `Ignition Gazebo, version 6.1.0`  
My ogre version is Ogre 2.2

Here are my ogre logs from `~/.ignition/rendering/ogre2.log`:
```
19:00:54: Creating resource group General
19:00:54: Creating resource group Internal
19:00:54: Creating resource group Autodetect
19:00:54: SceneManagerFactory for type 'DefaultSceneManager' registered.
19:00:54: Registering ResourceManager for type Material
19:00:54: Registering ResourceManager for type Mesh
19:00:54: Registering ResourceManager for type Mesh2
19:00:54: Registering ResourceManager for type OldSkeleton
19:00:54: MovableObjectFactory for type 'ParticleSystem' registered.
19:00:54: ArchiveFactory for archive type FileSystem registered.
19:00:54: ArchiveFactory for archive type Zip registered.
19:00:54: ArchiveFactory for archive type EmbeddedZip registered.
19:00:54: DDS codec registering
19:00:54: FreeImage version: 3.18.0
19:00:54: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
19:00:54: Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,psb,cut,xbm,xpm,gif,hdr,g3,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
19:00:54: OITD codec registering
19:00:54: Registering ResourceManager for type HighLevelGpuProgram
19:00:54: MovableObjectFactory for type 'Decal' registered.
19:00:54: MovableObjectFactory for type 'InternalCubemapProbe' registered.
19:00:54: MovableObjectFactory for type 'Entity' registered.
19:00:54: MovableObjectFactory for type 'Item' registered.
19:00:54: MovableObjectFactory for type 'Light' registered.
19:00:54: MovableObjectFactory for type 'Rectangle2Dv2' registered.
19:00:54: MovableObjectFactory for type 'BillboardSet' registered.
19:00:54: MovableObjectFactory for type 'ManualObject2' registered.
19:00:54: MovableObjectFactory for type 'BillboardChain' registered.
19:00:54: MovableObjectFactory for type 'RibbonTrail' registered.
19:00:54: MovableObjectFactory for type 'WireAabb' registered.
19:00:54: *-*-* OGRE Initialising
19:00:54: *-*-* Version 2.2.6 (Cerberus)
19:00:54: OverlayElementFactory for type Panel registered.
19:00:54: OverlayElementFactory for type BorderPanel registered.
19:00:54: OverlayElementFactory for type TextArea registered.
19:00:54: Registering ResourceManager for type Font
19:00:54: Loading library /usr/lib/x86_64-linux-gnu/OGRE-2.2/OGRE/RenderSystem_GL3Plus.so
19:00:54: Installing plugin: GL 3+ RenderSystem
19:00:54: OpenGL 3+ Rendering Subsystem created.
19:00:54: Found Num EGL Devices: 1
19:00:54: EGL Device: EGL_MESA_device_software #0
19:00:54: Trying to init device: EGL_MESA_device_software #0...
19:00:54: Created GL 3.3 context for device EGL_MESA_device_software #0
19:00:54: Destroying device: EGL_MESA_device_software #0...
19:00:54: Plugin successfully installed
19:00:54: Loading library /usr/lib/x86_64-linux-gnu/OGRE-2.2/OGRE/Plugin_ParticleFX.so
19:00:54: Installing plugin: ParticleFX
19:00:54: Particle Emitter Type 'Point' registered
19:00:54: Particle Emitter Type 'Box' registered
19:00:54: Particle Emitter Type 'Ellipsoid' registered
19:00:54: Particle Emitter Type 'Cylinder' registered
19:00:54: Particle Emitter Type 'Ring' registered
19:00:54: Particle Emitter Type 'HollowEllipsoid' registered
19:00:54: Particle Affector Type 'LinearForce' registered
19:00:54: Particle Affector Type 'ColourFader' registered
19:00:54: Particle Affector Type 'ColourFader2' registered
19:00:54: Particle Affector Type 'ColourImage' registered
19:00:54: Particle Affector Type 'ColourInterpolator' registered
19:00:54: Particle Affector Type 'Scaler' registered
19:00:54: Particle Affector Type 'Rotator' registered
19:00:54: Particle Affector Type 'DirectionRandomiser' registered
19:00:54: Particle Affector Type 'DeflectorPlane' registered
19:00:54: Plugin successfully installed
19:00:54: CPU Identifier & Features
19:00:54: -------------------------
19:00:54:  *   CPU ID: AuthenticAMD: AMD Ryzen 7 3700X 8-Core Processor
[... 360 lines cut ...]
19:00:54: Parsing script TerraShadowGenerator.compositor
19:00:54: Parsing script GpuNormalMapper.compositor
19:00:54: Parsing script EsmGaussianBlurLogFilter.material.json
19:00:54: Parsing script Mipmaps.material.json
19:00:54: Parsing script TerraShadowGenerator.material.json
19:00:54: Finished parsing scripts for resource group General
19:00:54: Creating resources for group General
19:00:54: All done
19:00:54: Parsing scripts for resource group Internal
19:00:54: Finished parsing scripts for resource group Internal
19:00:54: Creating resources for group Internal
19:00:54: All done
19:00:54: Added resource location '/usr/share/ignition/ignition-rendering6/media/materials/textures/' of type 'FileSystem' to resource group 'General'
19:00:54: Can't assign material scene::Material(65503) because this Material does not exist. Have you forgotten to define it in a .material script?
19:00:54: Can't assign material scene::Material(65497) because this Material does not exist. Have you forgotten to define it in a .material script?
[... cut ...]
19:00:54: Can't assign material scene::Material(65478) because this Material does not exist. Have you forgotten to define it in a .material script?
19:00:54: Can't assign material scene::Material(65485) because this Material does not exist. Have you forgotten to define it in a .material script?
19:00:54: Can't assign material scene::Material(65466) because this Material does not exist. Have you forgotten to define it in a .material script?
19:00:54: Vertex Shader: 100000001VertexShader_vs
Fragment Shader: 100000001PixelShader_ps
 GLSL validation result : 
active samplers with a different type refer to the same texture image unit
```

noelex, 2022-06-08:
Is there any progress on this? I'm struggling with the same issue here.

Currently adding `--render-engine ogre` as a temporary workaround.

rubenanapu, 2022-06-08:
If you compile from source, disabling anti-aliasing, it works.
More details on the links below:
* https://answers.gazebosim.org/question/28144/ignition-fortress-shows-a-black-screen-where-the-scene-should-be-with-opengl-45-on-docker-on-aws/
* https://github.com/gazebosim/gz-sim/issues/1116#issuecomment-1142388038

mohseninima, 2022-06-26:
Has there been any progress on this? I am stuck using Gazebo 11 since that works fine. Considering all future versions will be based on Ignition-Gazebo, I don't want to have to compile from source each time and disable anti-aliasing to get this working.

traversaro, 2022-07-01:
> If you compile from source, disabling anti-aliasing, it works. More details on the links below:
> 
>     * https://answers.gazebosim.org/question/28144/ignition-fortress-shows-a-black-screen-where-the-scene-should-be-with-opengl-45-on-docker-on-aws/
> 
>     * [When OpenGL ogre2 rendering is happening via mesa's llvmpipe 21 or 22 with Ignition Fortress a black area is displayed instead of the 3D rendering #1116 (comment)](https://github.com/gazebosim/gz-sim/issues/1116#issuecomment-1142388038)

I am not sure that is the case. That workaround works for a problem that occurs with Mesa software rendering, i.e. `llvmpipe`, this issue is about the problem related to running `ign gazebo` under the `d3d12` driver, i.e. the one that maps OpenGL calls to Direct3D 12 calls. 

According to my tests, when using `d3d12` with a recent mesa version (22.0.6, the one available in Ubuntu 22.04), `ign gazebo` crashes with:
~~~
straversaro@IITICUBLAP257:~$ ign gazebo shapes.sdf
terminate called after throwing an instance of 'Ogre::UnimplementedException'
  what():  OGRE EXCEPTION(9:UnimplementedException):  in GL3PlusTextureGpu::copyTo at /build/ogre-next-UFfg83/ogre-next-2.2.5+dfsg3/RenderSystems/GL3Plus/src/OgreGL3PlusTextureGpu.cpp (line 677)
Stack trace (most recent call last) in thread 3740:
#10   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in
#9    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f5543e699ff, in
#8    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f5543dd7b42, in
#7    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f553e22399d, in
#6    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f553e221f90, in qTerminate()
#5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f553fe642f6, in std::terminate()
#4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f553fe6428b, in
#3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f553fe58bfd, in
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f5543d6b7f2, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f5543d85475, in raise
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f5543dd9a7c, in pthread_kill
Aborted (Signal sent by tkill() 3665 1000)
~~~

Basically, the problem is that `d3d12` does not support OpenGL 4.3 or at least the extension `GL_ARB_copy_image`. I do not think you can workaround that by changing antialiasing settings.

darksylinc, 2022-07-30:
`GL_ARB_copy_image` is a very old extension (introduced in 2012).

Someone may want to contact the maintainer of the d3d12 driver to implement `GL_ARB_copy_image`. It shouldn't be hard for the d3d12 driver (it's literally a GPU-side memcpy).

The maintainer is probably referring to the difficulty other GL 4.3 features that have nothing to do with `GL_ARB_copy_image` that are gatekeeping this feature.

traversaro, 2022-08-26:
> `GL_ARB_copy_image` is a very old extension (introduced in 2012).
>
> Someone may want to contact the maintainer of the d3d12 driver to implement `GL_ARB_copy_image`. It shouldn't be hard for the d3d12 [...]

I checked on https://mesamatrix.net/, and as of today apparently `GL_ARB_copy_image`  and `GL_ARB_texture_view` are the only two extensions missing of OpenGL 4.3 on d3d12:
[image]

traversaro, 2022-10-15:
Since https://github.com/gazebosim/gz-sim/issues/1116 was fixed and landed in a release, a workaround for this issue is to set the env variable `LIBGL_ALWAYS_SOFTWARE=1` . This would use the CPU rendering instead of the GPU one (resulting in more CPU use), but at least should start fine.

traversaro, 2023-03-15:
Interestingly @bperseghetti noted that at least on mesa 22/23 `d3d12` does not list compatibility with `GL_ARB_copy_image` extension, but it list compatibility with `GL_NV_copy_image`:
~~~
traversaro@IITICUBLAP257:~$ glxinfo | grep copy_image
    GL_NV_conditional_render, GL_NV_copy_image, GL_NV_depth_clamp,
    GL_NV_copy_depth_to_color, GL_NV_copy_image, GL_NV_depth_clamp,
~~~ 

So, a possible thing to check is to use the function defined by the `GL_NV_copy_image` extension if `GL_ARB_copy_image` is not supported. Indeed, by checking the ogre-next source code it seems that some code to try to support a `GL_NV_copy_image` is already there, but commented, see https://github.com/OGRECave/ogre-next/blob/v2.3.1/RenderSystems/GL3Plus/src/OgreGL3PlusTextureGpu.cpp#L653-L660 . 

The code was already committed commented back when that code was introduced, see https://github.com/OGRECave/ogre-next/commit/22dc290dcae1692a8577ced093c46fb7e4a66ffa . However, it is probably relatively easy to fix the commented code by copying the arguments passed to `glCopyImageSubData` in the latest version of the code, and check if it works.

bperseghetti, 2023-03-15:
> Interestingly @bperseghetti noted that at least on mesa 22/23 `d3d12` does not list compatibility with `GL_ARB_copy_image` extension, but it list compatibility with `GL_NV_copy_image`:
> 
> ```
> traversaro@IITICUBLAP257:~$ glxinfo | grep copy_image
>     GL_NV_conditional_render, GL_NV_copy_image, GL_NV_depth_clamp,
>     GL_NV_copy_depth_to_color, GL_NV_copy_image, GL_NV_depth_clamp,
> ```
> 
> So, a possible thing to check is to use the function defined by the `GL_NV_copy_image` extension if `GL_ARB_copy_image` is not supported. Indeed, by checking the ogre-next source code it seems that some code to try to support a `GL_NV_copy_image` is already there, but commented, see https://github.com/OGRECave/ogre-next/blob/v2.3.1/RenderSystems/GL3Plus/src/OgreGL3PlusTextureGpu.cpp#L653-L660 .
> 
> The code was already committed commented back when that code was introduced, see [OGRECave/ogre-next@22dc290](https://github.com/OGRECave/ogre-next/commit/22dc290dcae1692a8577ced093c46fb7e4a66ffa) . However, it is probably relatively easy to fix the commented code by copying the arguments passed to `glCopyImageSubData` in the latest version of the code, and check if it works.
> 
> Another interesting question (that perhaps it is more appropriate for the mesa issue tracker) is: if `GL_NV_copy_image` is implemented for `d3d12`, probably it should be possible to easily support `GL_ARB_copy_image` as well?

Found it out from @PetervdPerk-NXP who is using our simulation on WSL.

AllianceDarkSylinc, 2023-03-15:
Hi!

Enabling that path in OgreNext should be easy and straightforward. It seems the commented out code is now out of date, so it probably should have the same function signature as its `GL_ARB_copy_image` counterpart.

It would need testing though.

> Another interesting question (that perhaps it is more appropriate for the mesa issue tracker) is: if GL_NV_copy_image is implemented for d3d12

A quick eye glance over the specs suggest `GL_ARB_copy_image` handles compressed format reinterpretation\* while `GL_NV_copy_image` does not. This is a very advanced use case and shouldn't be a problem for Gazebo, I'm just pointing out that _might_ be the reason `GL_ARB_copy_image` support isn't exposed.

\* Compressed format reinterpretation is when e.g. `BC1_UNORM` is copied to a `RGBA16_UINT` texture for raw binary access. This is mostly relevant for Compute Shader encoders/decoders.

darksylinc, 2023-03-15:
Hi!

Enabling that path in OgreNext should be easy and straightforward. It seems the commented out code is now out of date, so it probably should have the same function signature as its `GL_ARB_copy_image` counterpart.

It would need testing though.

> Another interesting question (that perhaps it is more appropriate for the mesa issue tracker) is: if GL_NV_copy_image is implemented for d3d12

A quick eye glance over the specs suggest `GL_ARB_copy_image` handles compressed format reinterpretation\* while `GL_NV_copy_image` does not. This is a very advanced use case and shouldn't be a problem for Gazebo, I'm just pointing out that _might_ be the reason `GL_ARB_copy_image` support isn't exposed while NV is.

\* Compressed format reinterpretation is when e.g. `BC1_UNORM` is copied to a `RGBA16_UINT` texture for raw binary access. This is mostly relevant for Compute Shader encoders/decoders.

traversaro, 2023-03-15:
Thanks @darksylinc for the answer!

traversaro, 2023-03-17:
Even after patching ogre-next to use `GL_NV_copy_image` (see https://github.com/conda-forge/ogre-next-feedstock/blob/4fee59f304adfd6d1e3074fb11a028fe80bcfbde/recipe/wslg_support.patch, it took some time to understand how gl3w worked for extensions), I tried `gz sim` and now when used with d3d12 it fails with another error (I am no sure how related or not):
~~~
terminate called after throwing an instance of 'Ogre::RenderingAPIException'
  what():  OGRE EXCEPTION(3:RenderingAPIException): Vertex Program 100000000VertexShader_vs failed to compile. See compile log above for details. in GLSLShader::compile at /home/conda/feedstock_root/build_artifacts/ogre-next_1679090885365/work/RenderSystems/GL3Plus/src/GLSL/OgreGLSLShader.cpp (line 361)
Stack trace (most recent call last) in thread 2006:
~~~

In `ogre2.log` the error is similar:
~~~
23:40:44: GLSL compile log: 100000000VertexShader_vs
0:390(29): error: syntax error, unexpected NEW_IDENTIFIER
23:40:44: OGRE EXCEPTION(3:RenderingAPIException): Vertex Program 100000000VertexShader_vs failed to compile. See compile log above for details. in GLSLShader::compile at /home/conda/feedstock_root/build_artifacts/ogre-next_1679090885365/work/RenderSystems/GL3Plus/src/GLSL/OgreGLSLShader.cpp (line 361
~~~

darksylinc, 2023-03-17:
Hi!

It's complaining about an unrelated shader compilation error.

To debug this problem you'll have to go to: https://github.com/gazebosim/gz-rendering/blob/c2e72ee51a7e4dba5156faa96c972c63ca5ab437/ogre2/src/Ogre2RenderEngine.cc#L849

and change that line into:

```cpp
hlmsUnlit->setDebugOutputPath(true, true, "/tmp/path/to/dump");
```

Then do the same for `hlmsPbs` and `hlmsTerra`.

In the folder you'll find the generated shaders and upload them here for further analysis.

**PS:** Slightly unrelated question: Does Vulkan work on WSLg?

traversaro, 2023-03-18:
> Hi!
> 
> It's complaining about an unrelated shader compilation error (unrelated to `GL_NV_copy_image` I mean).
> 
> To debug this problem you'll have to go to: https://github.com/gazebosim/gz-rendering/blob/c2e72ee51a7e4dba5156faa96c972c63ca5ab437/ogre2/src/Ogre2RenderEngine.cc#L849
> 
> and change that line into:
> 
> ```c++
> hlmsUnlit->setDebugOutputPath(true, true, "/tmp/path/to/dump");
> ```
> 
> Then do the same for `hlmsPbs` and `hlmsTerra`.
> 
> In the folder you'll find the generated shaders and upload them here for further analysis.

Thanks! Thanks a lot for the hints. 

> PS: Slightly unrelated question: Does Vulkan work on WSLg?

As for opengl,  there is the software implementation (`lavapipe`/`lvp`, equivalent of `llvmpipe`) and the the one that actually implements vulkan on top of DirectX12 (`dozen`/`dzn`, equivalent of `d3d12`). However, `dozen` is still a work in progress, and I never tested it with ogre-next.

kryptoniancode, 2023-05-03:
In docker wsl2,
```
No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
[ign gazebo-1]
[ign gazebo-1] libEGL warning: NEEDS EXTENSION: falling back to kms_swrast
[INFO] [ros2-8]: process has finished cleanly [pid 209]
[INFO] [ros2-9]: process started with pid [256]
[ign gazebo-1] terminate called after throwing an instance of 'Ogre::UnimplementedException'
[ign gazebo-1]   what():  OGRE EXCEPTION(9:UnimplementedException):  in GL3PlusTextureGpu::copyTo at /build/ogre-next-UFfg83/ogre-next-2.2.5+dfsg3/RenderSystems/GL3Plus/src/OgreGL3PlusTextureGpu.cpp (line 677)
[ign gazebo-1] Stack trace (most recent call last) in thread 243:
[ign gazebo-1] #10   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in
[ign gazebo-1] #9    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f2c465ce9ff, in
[ign gazebo-1] #8    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f2c4653cb42, in
[ign gazebo-1] #7    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f2c40ba299d, in
[ign gazebo-1] #6    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f2c40ba0f90, in qTerminate()
[ign gazebo-1] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f2c427fc2b6, in std::terminate()
[ign gazebo-1] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f2c427fc24b, in
[ign gazebo-1] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f2c427f0bbd, in
[ign gazebo-1] #2    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f2c464d07f2, in abort
[ign gazebo-1] #1    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f2c464ea475, in raise
[ign gazebo-1] #0    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f2c4653ea7c, in pthread_kill
[ign gazebo-1] Aborted (Signal sent by tkill() 149 0)
```

After using this it works.
```
docker .... LD_LIBRARY_PATH="/usr/lib/x86_64-linux-gnu/libstdc++.so.6"
```

But don't know how to use it in wsl2 not in docker.

traversaro, 2023-05-06:
> After using this it works.

Are you sure that it is using the GPU in this case? I do not understand how the `d3d12` mesa driver could start working by just setting `LD_LIBRARY_PATH` to that location.

traversaro, 2023-05-15:
> Nothing worked for me. I am using wsl with my windows 11 host.

Just to clarify, at least setting `export LIBGL_ALWAYS_SOFTWARE=true` works, even if slow as no GPU is used? Otherwise there could be some strange problem in your setup.

bailaC, 2023-05-15:
Well, when I set the env variable, I get one black screen and below error. 

`root@baila:~# export LIBGL_ALWAYS_SOFTWARE=1
root@baila:~# gz sim shapes.sdf
QStandardPaths: runtime directory '/run/user/0/' is not owned by UID 0, but a directory permissions 0700 owned by UID 1000 GID 1000
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: MESA-LOADER: failed to open vgem: /usr/lib/dri/vgem_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
`

When it is not set, the gui comes and close with the below crash message,

> root@baila:~# export LIBGL_ALWAYS_SOFTWARE=0
> root@baila:~# gz sim shapes.sdf
> QStandardPaths: runtime directory '/run/user/0/' is not owned by UID 0, but a directory permissions 0700 owned by UID 1000 GID 1000
> libEGL warning: MESA-LOADER: failed to open vgem: /usr/lib/dri/vgem_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
> 
> libEGL warning: NEEDS EXTENSION: falling back to kms_swrast
> terminate called after throwing an instance of 'Ogre::UnimplementedException'
>   what():  OGRE EXCEPTION(9:UnimplementedException):  in GL3PlusTextureGpu::copyTo at ./RenderSystems/GL3Plus/src/OgreGL3PlusTextureGpu.cpp (line 685)
> Stack trace (most recent call last):
> #31   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fe27cc6b75a, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
> #30   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fe27ccc60b7, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
> #29   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fe27a8423e2, in g_main_context_iteration
> #28   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fe27a8996c7, in
> #27   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fe27a844d3a, in g_main_context_dispatch
> #26   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fe27ccc6a66, in
> #25   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fe27cc6ff26, in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
> #24   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fe27cc6ce39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
> #23   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fe27c3c4712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
> #22   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fe27cc9a41d, in QObject::event(QEvent*)
> #21   Object "/usr/lib/x86_64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0x7fe22611d513, in gz::gui::plugins::RenderWindowItem::Ready()
> #20   Object "/usr/lib/x86_64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0x7fe22611d120, in gz::gui::plugins::RenderThread::Initialize[abi:cxx11]()
> #19   Object "/usr/lib/x86_64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0x7fe226129c7d, in gz::gui::plugins::RenderThreadRhiOpenGL::Initialize[abi:cxx11]()
> #18   Object "/usr/lib/x86_64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0x7fe226120c87, in gz::gui::plugins::GzRenderer::Initialize[abi:cxx11]()
> #17   Object "/lib/x86_64-linux-gnu/libgz-rendering7.so.7", at 0x7fe224e9a768, in gz::rendering::v7::BaseRenderEngine::CreateScene(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
> #16   Object "/lib/x86_64-linux-gnu/libgz-rendering7.so.7", at 0x7fe224ea2cfe, in gz::rendering::v7::BaseRenderEngine::CreateScene(unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
> #15   Object "/lib/x86_64-linux-gnu/libgz-rendering7.so.7", at 0x7fe224eb423a, in gz::rendering::v7::BaseScene::CreateMaterials()
> #14   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre2.so", at 0x7fe1d20acd1a, in gz::rendering::v7::Ogre2Material::SetTextureMapImpl(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Ogre::PbsTextureTypes)
> #13   Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fe1d1d1a224, in Ogre::TextureGpuManager::_waitFor(Ogre::TextureGpu*, bool)
> #12   Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fe1d1d19e4a, in Ogre::TextureGpuManager::_update(bool)
> #11   Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fe1d1be755c, in Ogre::ObjCmdBuffer::execute()
> #10   Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fe1d1be58cd, in Ogre::ObjCmdBuffer::NotifyDataIsReady::execute()
> #9    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fe1d1d09d9b, in Ogre::TextureFilter::GenerateHwMipmaps::_executeSerial(Ogre::TextureGpu*)
> #8    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/OGRE/RenderSystem_GL3Plus.so", at 0x7fe1d1687ac3, in
> #7    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/OGRE/RenderSystem_GL3Plus.so", at 0x7fe1d16843da, in
> #6    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fe27e6c4517, in __cxa_throw
> #5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fe27e6c42b6, in std::terminate()
> #4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fe27e6c424b, in
> #3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fe27e6b8bbd, in
> #2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fe28238b7f2, in abort
> #1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fe2823a5475, in raise
> #0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fe2823f9a7c, in pthread_kill
> Aborted (Signal sent by tkill() 1753 0)

bailaC, 2023-05-15:
@kryptoniancode , so I need to use docker inside wsl to work with gazebo?

bailaC, 2023-05-15:
@kryptoniancode , thanks for your help. After executing you suggested cmds and the LIBGL_ALWAYS_SOFTWARE set, the GUI is opened and I can interact with it well. One doubt, how to identify whether it is using CPU or GPU? I want to use my GPU as I have plenty of it.

[image]

kryptoniancode, 2023-05-15:
> so I need to use docker inside wsl to work with gazebo?

In some case you have to use following in wsl2. (command needs to check)
```
gz sim shapes.sdf --render-engine ogre
```
>  how to identify whether it is using CPU or GPU?

Use mesa-utils

kryptoniancode, 2023-05-15:
@traversaro , is there any fix for this?

I have tried many things to run gazebo with some demos I have, but still get errors similar to below.
I have tried following options and command in terminal.

```
1. export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6
2. --render-engine ogre
3. sudo chmod 777 /dev/dri/renderD128
4. LIBGL_ALWAYS_SOFTWARE=1
```

But still get this errors.
```
[ign gazebo --render-engine ogre-1] libEGL warning: MESA-LOADER: failed to open i965: /usr/lib/dri/i965_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
[ign gazebo --render-engine ogre-1]
[ign gazebo --render-engine ogre-1] libEGL warning: MESA-LOADER: failed to open i965: /usr/lib/dri/i965_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
[ign gazebo --render-engine ogre-1]
[ign gazebo --render-engine ogre-1] terminate called after throwing an instance of 'Ogre::UnimplementedException'
[ign gazebo --render-engine ogre-1]   what():  OGRE EXCEPTION(9:UnimplementedException):  in GL3PlusTextureGpu::copyTo at /build/ogre-next-UFfg83/ogre-next-2.2.5+dfsg3/RenderSystems/GL3Plus/src/OgreGL3PlusTextureGpu.cpp (line 677)
[ign gazebo --render-engine ogre-1] Stack trace (most recent call last) in thread 200442:
[ign gazebo --render-engine ogre-1] #10   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in
[ign gazebo --render-engine ogre-1] #9    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f1ec9c879ff, in
[ign gazebo --render-engine ogre-1] #8    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f1ec9bf5b42, in
[ign gazebo --render-engine ogre-1] #7    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f1ec472499d, in
[ign gazebo --render-engine ogre-1] #6    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f1ec4722f90, in qTerminate()
[ign gazebo --render-engine ogre-1] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f1eca1d72b6, in std::terminate()
[ign gazebo --render-engine ogre-1] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f1eca1d724b, in
[ign gazebo --render-engine ogre-1] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f1eca1cbbbd, in
[ign gazebo --render-engine ogre-1] #2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f1ec9b897f2, in abort
[ign gazebo --render-engine ogre-1] #1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f1ec9ba3475, in raise
[ign gazebo --render-engine ogre-1] #0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f1ec9bf7a7c, in pthread_kill
[ign gazebo --render-engine ogre-1] Aborted (Signal sent by tkill() 200302 1001)
[ign gazebo --render-engine ogre-1] [INFO] [1684184455.617174698] [rclcpp]: signal_handler(signum=2)
[INFO] [ign gazebo --render-engine ogre-1]: process has finished cleanly [pid 200248]
```

traversaro, 2023-05-15:
In theory only one of `export LIBGL_ALWAYS_SOFTWARE=1` or `--render-engine ogre` is needed.

traversaro, 2023-05-20:
> Hi!
>
> It's complaining about an unrelated shader compilation error (unrelated to `GL_NV_copy_image` I mean).
>
> To debug this problem you'll have to go to: [...]

I finally tried to do this (sorry but setting up a local environment was not trivial), and after this modification:
~~~diff
diff --git a/ogre2/src/Ogre2RenderEngine.cc b/ogre2/src/Ogre2RenderEngine.cc
index 049781b4..d88dabdb 100644
--- a/ogre2/src/Ogre2RenderEngine.cc
+++ b/ogre2/src/Ogre2RenderEngine.cc
@@ -862,7 +862,7 @@ void Ogre2RenderEngine::RegisterHlms()
     Ogre::Root::getSingleton().getHlmsManager()->registerHlms(hlmsUnlit);

     // disable writting debug output to disk
-    hlmsUnlit->setDebugOutputPath(false, false);
+    hlmsUnlit->setDebugOutputPath(true, true, "/home/traversaro/gz-ws/srctst/ogrelog");
     hlmsUnlit->setListener(hlmsUnlit);

     this->dataPtr->gzHlmsUnlit = hlmsUnlit;
@@ -907,7 +907,7 @@ void Ogre2RenderEngine::RegisterHlms()
     Ogre::Root::getSingleton().getHlmsManager()->registerHlms(hlmsPbs);

     // disable writting debug output to disk
-    hlmsPbs->setDebugOutputPath(false, false);
+    hlmsPbs->setDebugOutputPath(true, true, "/home/traversaro/gz-ws/srctst/ogrelog");
     hlmsPbs->setListener(hlmsPbs);

     dataPtr->gzHlmsPbs = hlmsPbs;
@@ -942,7 +942,7 @@ void Ogre2RenderEngine::RegisterHlms()
     Ogre::Root::getSingleton().getHlmsManager()->registerHlms(hlmsTerra);

     // disable writting debug output to disk
-    hlmsTerra->setDebugOutputPath(false, false);
+    hlmsTerra->setDebugOutputPath(true, true, "/home/traversaro/gz-ws/srctst/ogrelog");
     hlmsTerra->setListener(hlmsTerra);

     this->dataPtr->terraWorkspaceListener.reset(
~~~

After the execution I do not find anything in `/home/traversaro/gz-ws/srctst/ogrelog`, I am not sure if I am doing anything wrong. 

Anyhow, by concentrating on the error in `ogre2.log` :
~~~
17:16:12: GLSL compile log: 100000000VertexShader_vs
0:317(29): error: syntax error, unexpected NEW_IDENTIFIER
17:16:12: OGRE EXCEPTION(3:RenderingAPIException): Vertex Program 100000000VertexShader_vs failed to compile. See compile log above for details. in GLSLShader::compile at /home/traversaro/gz-ws/srctst/ogre-next/RenderSystems/GL3Plus/src/GLSL/OgreGLSLShader.cpp (line 361)
~~~

I think I understood the error occurs on line https://github.com/gazebosim/gz-rendering/blob/gz-rendering7_7.4.0/ogre2/src/media/Hlms/Pbs/GLSL/VertexShader_vs.glsl#L50, even if I am not able to understand more at the moment unfortunatly.

AllianceDarkSylinc, 2023-05-20:
> After the execution I do not find anything in /home/traversaro/gz-ws/srctst/ogrelog, I am not sure if I am doing anything wrong.

OgreNext will not create the folder, so the dump will fail if the path doesn't exist.

Run `mkdir -p /home/traversaro/gz-ws/srctst/ogrelog` and try again.

traversaro, 2023-05-20:
> After the execution I do not find anything in `/home/traversaro/gz-ws/srctst/ogrelog`, I am not sure if I am doing anything wrong.

I think I got the problem. For some reason ogre was trying to save the files to  `/home/traversaro/gz-ws/srctst/ogrelog.`, by inserting the path as ` `/home/traversaro/gz-ws/srctst/ogrelog/` everything works and I can save the 100000000VertexShader_vs shader:

<details>

<summary> Click to see the saved `100000000VertexShader_vs.glsl` </summary> 

~~~glsl
#if 0
	***	[Hash 0x01cc1123]	0
	***	[Hash 0x0790ba12]	0
	***	[Hash 0x08ffb38a]	0
	***	[Hash 0x123606a9]	0
	***	[Hash 0x1b3509dd]	1
	***	[Hash 0x201f84c0]	0
	***	[Hash 0x25dc73c6]	635204550
	***	[Hash 0x2dd3a2cd]	0
	***	[Hash 0x354d66c0]	0
	***	[Hash 0x359e8062]	0
	***	[Hash 0x378f5735]	1
	***	[Hash 0x38942648]	1
	***	[Hash 0x3b038020]	0
	***	[Hash 0x3fcb60f1]	1070293233
	***	[Hash 0x4508a85c]	1
	***	[Hash 0x516d43dd]	1
	***	[Hash 0x57643473]	0
	***	[Hash 0x5ad27ea7]	0
	***	[Hash 0x5b0b2416]	0
	***	[Hash 0x5cb4719a]	0
	***	[Hash 0x61e63948]	1
	***	[Hash 0x675280f4]	0
	***	[Hash 0x6962e498]	0
	***	[Hash 0x6bf5b20f]	1
	***	[Hash 0x6dc6cf58]	1841745752
	***	[Hash 0x74824502]	1
	***	[Hash 0x8421366d]	240
	***	[Hash 0x86319b9f]	1
	***	[Hash 0x875516cf]	0
	***	[Hash 0x8a7512d7]	0
	***	[Hash 0x93f2327b]	635204550
	***	[Hash 0x962aeb1a]	0
	***	[Hash 0x9abd84b5]	-1698855755
	***	[Hash 0xa1b3cd70]	1
	***	[Hash 0xa461daa9]	0
	***	[Hash 0xa6ac776b]	0
	***	[Hash 0xafaf1bb3]	410
	***	[Hash 0xb22f037a]	0
	***	[Hash 0xb967bb7b]	1
	***	[Hash 0xe4120c6e]	4
	***	[Hash 0xe79b3190]	1
	***	[Hash 0xe7bd0fde]	0
	***	[Hash 0xebcb8569]	-338983575
	***	[Hash 0xec133132]	-334286542
	***	[Hash 0xf742ff75]	1
	DONE DUMPING PROPERTIES
	DONE DUMPING PIECES
#endif

			#version 330 core
	

    #extension GL_ARB_shading_language_420pack: require
    #define layout_constbuffer(x) layout( std140, x )

	#define bufferFetch texelFetch
	#define structuredBufferFetch texelFetch

	#extension GL_AMD_shader_trinary_minmax: require

[... 324 lines cut ...]
			
				outVs_Position.z = outVs_Position.z + shadowConstantBias;
					
			

	/// hlms_uv_count will be 0 on shadow caster passes w/out alpha test
	

	

	

	

	

	
  
  
  

}
~~~

</details>

darksylinc, 2023-05-20:
(reposting because it was from the wrong account)

> After the execution I do not find anything in /home/traversaro/gz-ws/srctst/ogrelog, I am not sure if I am doing anything wrong.

OgreNext will not create the folder, so the dump will fail if the path doesn't exist.

Run `mkdir -p /home/traversaro/gz-ws/srctst/ogrelog` and try again.

traversaro, 2023-05-20:
> OgreNext will not create the folder, so the dump will fail if the path doesn't exist.

The folder was there, but for some reason ogre-next 2.3.1 was addng a `.` to the folder name, adding `\` at the end of the path solved the problem.

darksylinc, 2023-05-20:
Thanks!

I looked into the error.

There should be a file in gz's data folder called `CrossPlatformSettings_piece_all.glsl`

It starts with this:

```
@piece( SetCrossPlatformSettings )
@property( syntax == glslvk )
	#version 450 core
@else
	@property( GL3+ >= 430 )
		#version 430 core
	@else
		#version 330 core
	@end
@end
```

Change it to this:

```
@piece( SetCrossPlatformSettings )
@property( syntax == glslvk )
	#version 450 core
@else
	@property( GL3+ >= 430 )
		#version 430 core
	@else
		#version 330 core

		@property( hlms_readonly_is_tex )
			#extension GL_ARB_shader_storage_buffer_object: require
		@end
	@end
@end
```

And let me know if that fixes it. If so, I'll add a fix to OgreNext.

traversaro, 2023-05-20:
This fixes it (at least on mesa 23.1), thanks a lot!

traversaro, 2023-05-21:
Ok, I opened two PRs to fix this issue:
* https://github.com/OGRECave/ogre-next/pull/388
* https://github.com/gazebosim/gz-rendering/pull/851

Related conda-forge backports:
* https://github.com/conda-forge/ogre-next-feedstock/pull/16
* https://github.com/conda-forge/gz-rendering-feedstock/pull/10

traversaro, 2023-05-21:
Even if the basic functionality with d3d12/WSLg is fixed by the two aforementioned PR, the gz-rendering test suite has still some failures, that anyhow I reported in https://github.com/gazebosim/gz-rendering/issues/852 as this issue is already quite noisy, and I think we can consider it closed once the fixes are merged and released in apt packages.

drustanyjt, 2023-05-27:
> Even if the basic functionality with d3d12/WSLg is fixed by the two aforementioned PR, the gz-rendering test suite has still some failures, that anyhow I reported in [gazebosim/gz-rendering#852](https://github.com/gazebosim/gz-rendering/issues/852) as this issue is already quite noisy, and I think we can consider it closed once the fixes are merged and released in apt packages.

Hi @traversaro I'm facing a similar issue in WSL2 Ubuntu 22.04 with WSLg (though using `llvmpipe` still works fine). This was on Gazebo Garden (installed from binaries).

I wanted to try using your patched OGRECave/ogre-next#388 to fix this issue, and run `gz sim`, but I'm not sure how to do so. Would rebuilding gazebo-garden from the master branch be enough? Would I need to build ogre-next as well?

alexdrc00, 2023-06-08:
Hello @traversaro I am facing the same issue. Same as @drustanyjt I want to try using your approach, could you provide some guidance on how you managed to solve it?

traversaro, 2023-06-08:
Sorry  @drustanyjt, I missed your comment. Thanks @alexdrc00 for the ping.

> Would rebuilding gazebo-garden from the master branch be enough? Would I need to build ogre-next as well?

For what regards gz-rendering, indeed just building from master is ok as https://github.com/gazebosim/gz-rendering/pull/851 was merged. For what regard ogre-next, the fix was merged in https://github.com/OGRECave/ogre-next/tree/v2-3, so I guess you should build from that branch (even if more my tests I actually just cherry-picked the change on top of the ogre-next v2.3.1 packaged in conda-forge).

traversaro, 2023-06-08:
For what regards long term, unfortunatly ogre-next is packaged via official Ubuntu repositories, so excluding complicated procedures like [SRU](https://wiki.ubuntu.com/StableReleaseUpdates) or setting up a PPA just with patched ogre-next, probably the most important thing is to make sure that in Ubuntu 24.04 a new version of ogre-next with this fix is released.

rosiakpiotr, 2023-06-10:
Is there some kind of read that will walk me through applying this patch? I don't have much experience with ubuntu packages, and I am worried that if I uninstall the `apt` package, other packages will be broken due to missing dependency before I compile patched version, am I right?

traversaro, 2023-06-11:
> Is there some kind of read that will walk me through applying this patch?

Unfortunatly I do not have one, as I have developed the patches working with conda-forge packages.

> Is there some kind of read that will walk me through applying this patch? I don't have much experience with ubuntu packages, and I am worried that if I uninstall the `apt` package, other packages will be broken due to missing dependency before I compile patched version, am I right?

That is correct, if you uninstall an apt package in general you need to recompile all the downstram packages by source.

traversaro, 2023-07-31:
> For what regards long term, unfortunatly ogre-next is packaged via official Ubuntu repositories, so excluding complicated procedures like [SRU](https://wiki.ubuntu.com/StableReleaseUpdates) or [...]

Apparently ogre-next is actually packaged in the gazebo apt repository, so we can just backport the fix https://github.com/OGRECave/ogre-next/pull/388 there and that should fix the issue.

traversaro, 2023-08-01:
> > For what regards long term, unfortunatly ogre-next is packaged via official Ubuntu repositories, so excluding complicated procedures like [SRU](https://wiki.ubuntu.com/StableReleaseUpdates) or setting up a PPA just with patched ogre-next, probably the most important thing is to make sure that in Ubuntu 24.04 a new version of ogre-next with this fix is released.
> 
> Apparently ogre-next is actually packaged in the gazebo apt repository, so we can just backport the fix [OGRECave/ogre-next#388](https://github.com/OGRECave/ogre-next/pull/388) there and that should fix the issue.

I tried to do this in https://github.com/gazebo-forks/ogre-2.3-release/pull/4 . Once that is merged and released, we just need a release of gz-rendering7 and in theory the problem should be solved, at least for users wth a recent enough mesa (I tested 2.2.5 and 2.3.1).

PetervdPerk-NXP, 2023-08-01:
Awesome, for people running WSL I would recommend using kisak-mesa ppa.
https://launchpad.net/~kisak/+archive/ubuntu/kisak-mesa

traversaro, 2023-08-09:
Now that the fix has been backported to the ogre-next deb package, the latest thing that is missing to consider this issue closed is to do a new release of gz-rendering 7 .

iche033, 2023-08-18:
gz-rendering 7.4.1 has just been released. It should include the fixes in https://github.com/gazebosim/gz-rendering/pull/851.

traversaro, 2023-08-19:
With some users from my lab (@Giacomobisio and @lucapa17) we tested the latest packages to see if this issue was fixed, but we still have problems (see https://github.com/robotology/gz-yarp-plugins/issues/40 for details). In particular, it seems that ogre-next `2.3.1-8osrf~jammy` was installed, while the last build in https://build.osrfoundation.org/job/ogre-2.3-debbuilder/ (with the fix) is `2.3.1-4osrf~jammy` . In theory `2.3.1-8osrf~jammy` should come later `2.3.1-4osrf~jammy`, but anyhow perhaps something went wrong. @j-rivero do you have any idea how ogre-next `2.3.1-8osrf~jammy` got uploaded in the repo? Thanks!

traversaro, 2023-08-19:
> In theory `2.3.1-8osrf~jammy` should come later `2.3.1-4osrf~jammy`, but anyhow perhaps something went wrong

Indeed, in http://packages.osrfoundation.org/gazebo/ubuntu-stable/pool/main/o/ogre-next-2.3/ogre-next-2.3_2.3.1-8osrf~jammy.debian.tar.xz there is no [`fix-gpu-in-wsl.patch`](https://github.com/gazebo-forks/ogre-2.3-release/blob/master/debian/patches/fix-gpu-in-wsl.patch), that is the one necessary to fix this issue.

traversaro, 2023-09-06:
> @j-rivero do you have any idea how ogre-next `2.3.1-8osrf~jammy` got uploaded in the repo? Thanks!

Friendly ping @j-rivero @azeey, not sure if you saw the first comment!

azeey, 2023-09-11:
It looks like we have different branches for `jammy` and `focal` in https://github.com/gazebo-forks/ogre-2.3-release, but the latest [build](https://build.osrfoundation.org/job/ogre-2.3-debbuilder/54/) used the `master` branch. /cc @mjcarroll

j-rivero, 2023-09-12:
> It looks like we have different branches for jammy and focal in https://github.com/gazebo-forks/ogre-2.3-release, but the latest [build](https://build.osrfoundation.org/job/ogre-2.3-debbuilder/54/) used the master branch. /cc @mjcarroll

I'm looking into this, should not be difficult to fix

j-rivero, 2023-09-12:
> I'm looking into this, should not be difficult to fix

I've released `2.3.1-9osrf` into the prerelease repository for both focal/jammy. Should finishes with the problems of versioning, sorry. Give me an ok and I'll push them to stable.

traversaro, 2023-09-13:
> > I'm looking into this, should not be difficult to fix
> 
> I've released `2.3.1-9osrf` into the prerelease repository for both focal/jammy. Should finishes with the problems of versioning, sorry. Give me an ok and I'll push them to stable.

I just tested on a fresh Windows machine with WSLg + GPU on Ubuntu 22.04 . With stable release it is crashing, while with prerelease it is starting fine.

azeey, 2023-09-13:
https://build.osrfoundation.org/view/ign-harmonic/job/gz_rendering-ci-gz-rendering8-jammy-amd64/7/ builds gz-rendering with ogre `2.3.1-9osrf~jammy` and all tests are passing. I've also ran gz-sim locally with this and things seem to work fine, so I think we're ready to push to stable.

traversaro, 2023-10-09:
Apparently the stable repo still has 2.3.1-8osrf~jammy, see http://packages.osrfoundation.org/gazebo/ubuntu-stable/pool/main/o/ogre-next-2.3/ .

[26 more replies]

closed 2023-11-02 by j-rivero