gazebosim/gz-sim #934: Odometry 3D publisher plugin

blakermchale, 2021-07-26:
## Desired behavior
I've been implementing a multicopter in ignition gazebo and I wanted to get a simple plugin that publishes the pose and velocities of the drone. I saw `odometry_publisher` existed and looked at #547. Ideally, this system would estimate the velocities in a similar way and not use the planar assumption. I want to use this plugin to obtain the ground truth of my vehicle.

## Alternatives considered
This could either be done by extending the `odometry_publisher` to handle 3D space or creating a new odometry 3d publisher. As is, the current name is misleading since its logic is derived from `gazebo_ros_planar_move` and only allows for 2D odometry.

## Implementation suggestion
Same implementation as mentioned above.

chapulina, 2021-07-27:
> This could either be done by extending the odometry_publisher to handle 3D space or creating a new odometry 3d publisher. 

If adding 3D estimation will leverage most of the existing code, I think it makes sense to extend the existing plugin and add an option to provide 2D or 3D. But if it will be mainly new code living alongside the existing one, maybe it's best to create a new plugin.

blakermchale, 2021-07-27:
Do you have any recommendations for how to setup the test for this? I was thinking of using the multicopter velocity control system but realized there would be no way to validate the angular rate for roll and pitch.

Also I created a branch [here](https://github.com/blakermchale/ign-gazebo/commit/5cfd02ec78c582dbf9737d7b649582b812140384) and got it mostly working except for the tests

chapulina, 2021-07-27:
> I created a branch here and got it mostly working except for the tests

Awesome, that was fast!

> how to setup the test for this? 

Did you take a look at [INTEGRATION_odometry_publisher](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo5/test/integration/odometry_publisher.cc)? It's manually setting velocities and checking the result. I think you could create a new test world that has `<dimensions>3</dimensions>` and add a new test case to `odometry_publisher.cc` that is similar to the 2D ones.

blakermchale, 2021-07-27:
Yep! I took a look at that and [INTEGRATION_multicopter](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo5/test/integration/multicopter.cc) since I want to capture the 3D aspect and can't use the current 2D method of applying a velocity directly to a model with wheels. 

Can you think of any way for testing the angular roll and pitch? Those are the only values I can't test right now

chapulina, 2021-07-28:
> can't use the current 2D method of applying a velocity directly to a model with wheels.

What about this part that sends linear and angular velocity commands? They accept 3D values:

https://github.com/ignitionrobotics/ign-gazebo/blob/511287d786b87094208af65e51e1c2960c55003a/test/integration/odometry_publisher.cc#L123-L159

blakermchale, 2021-07-29:
That part works, I got up a test with commanding velocities using it. I'm using the multicopter and command angular yaw and xyz linear velocities.

PR #1180 (2021-11-09): Extend odom publisher to allow 3D

# 🎉 New feature

Closes #934 and replaces #941.

## Summary
This adds in a new tag `<dimensions>` to the odometry publisher plugin that accepts either 2 or 3 (defaults to 2). When the value is 3 it assumes a 3D assumption instead of planar. This is useful for people who want to know the z location and angular rates. In my use case, I was using the multicopter plugin with the X3 model.

## Test it
This feature can be tested by running the `INTEGRATION_odometry_publisher` test or adding the tag to the plugin in an SDF file. I tested this manually and echoed the [... cut]
closed 2021-11-13 by blakermchale