gazebosim/gz-sim #956: Unknown CMake command "PROTOBUF_GENERATE_CPP".

mboisson, 2021-08-03:
I'm trying to build `ign-gazebo3_3.8.0`, and I encounter: 
``` 
26409 CMake Error at src/msgs/CMakeLists.txt:2 (PROTOBUF_GENERATE_CPP):
26410   Unknown CMake command "PROTOBUF_GENERATE_CPP".
26411
``` 

According to this page
https://stackoverflow.com/questions/52533396/cmake-cant-find-protobuf-protobuf-generate-cpp
there would be something missing in the `CMakeLists.txt`, but even patching it did not resolve the issue. 

I have `protobuf/3.12.3` and `cmake/3.16.5` 

Any idea ?

Below is the full log: 
```
== 2021-08-03 17:07:25,653 build_log.py:169 ERROR EasyBuild crashed with an error (at easybuild/site-packages/easybuild-framework/easybuild/base/exceptions.py:124 in __init__): cmd " cmake -DCMAKE_INSTALL_PREFIX=/home/mboisson/.local/easybuild/software/2020/avx2/Compiler/gcc9/ignition/citadel -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_COMPILER='gcc' -DCMAKE_C_FLAGS='-O2 -ftree-vectorize -march=core-avx2 -fno-math-errno' -DCMAKE_CXX_COMPILER='g++' -DCMAKE_CXX_FLAGS='-O2 -ftree-vectorize -march=core-avx2 -fno-math-errno' -DCMAKE_Fortran_COMPILER='gfortran' -DCMAKE_Fortran_FLAGS='-O2 -ftree-vectorize -march=core-avx2 -fno-math-errno' -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_FIND_USE_PACKAGE_REGISTRY=FALSE -DBoost_NO_BOOST_CMAKE=ON -DBOOST_ROOT=/cvmfs/soft.computecanada.ca/easybuild/software/2020/avx2/Compiler/gcc9/boost/1.72.0 -DBoost_NO_SYSTEM_PATHS=ON  -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON  -DENABLE_XHOST=OFF  -DCMAKE_SKIP_INSTALL_RPATH=ON  /tmp/mboisson/avx2/ignition/citadel/GCC-9.3.0/ign-gazebo-ignition-gazebo3_3.8.0" exited with exit code 1 and output:
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /cvmfs/soft.computecanada.ca/easybuild/software/2020/Core/gcccore/9.3.0/bin/gcc
-- Check for working C compiler: /cvmfs/soft.computecanada.ca/easybuild/software/2020/Core/gcccore/9.3.0/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /cvmfs/soft.computecanada.ca/easybuild/software/2020/Core/gcccore/9.3.0/bin/g++
-- Check for working CXX compiler: /cvmfs/soft.computecanada.ca/easybuild/software/2020/Core/gcccore/9.3.0/bin/g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- ignition-gazebo3 version 3.8.0
-- Operating system is Linux
-- Looking for ignition-math6 -- found version 6.8.0
-- Searching for dependencies of ignition-math6
-- Looking for sdformat9 - found

-- Looking for ignition-plugin1 -- found version 1.2.0
-- Searching for dependencies of ignition-plugin1
-- Searching for <ignition-plugin1> component [loader]
-- Looking for ignition-plugin1-loader -- found version 1.2.0
-- Searching for dependencies of ignition-plugin1-loader
-- Searching for <ignition-plugin1> component [register]
-- Looking for ignition-plugin1-register -- found version 1.2.0
-- Searching for dependencies of ignition-plugin1-register
-- Looking for ignition-plugin1 - found

-- Looking for ignition-transport8 -- found version 8.2.0
-- Searching for dependencies of ignition-transport8
-- Found ZLIB: /cvmfs/soft.computecanada.ca/gentoo/2020/usr/lib/libz.so (found version "1.2.11")
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.3.2
-- Found ZeroMQ: TRUE (Required is at least version "4")
-- Checking for module 'uuid'
--   Found uuid, version 2.33.2
-- Found UUID: TRUE
-- Looking for ignition-msgs5 -- found version 5.7.0
-- Searching for dependencies of ignition-msgs5
-- Looking for ignition-math6 -- found version 6.8.0
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 7.1.0
-- Searching for <ignition-transport8> component [log]
-- Looking for ignition-transport8-log -- found version 8.2.0
-- Searching for dependencies of ignition-transport8-log
-- Looking for ignition-transport8 - found

-- Looking for ignition-msgs5 -- found version 5.7.0
-- Looking for ignition-msgs5 - found
[... 207 lines cut ...]
-- Configuring library: ignition-gazebo3-log-video-recorder-system
-- Configuring library: ignition-gazebo3-logicalaudiosensorplugin-system
-- Adding 1 UNIT tests
-- Configuring library: ignition-gazebo3-logical-camera-system
-- Configuring library: ignition-gazebo3-magnetometer-system
-- Configuring library: ignition-gazebo3-multicopter-motor-model-system
-- Configuring library: ignition-gazebo3-multicopter-control-system
-- Configuring library: ignition-gazebo3-performer-detector-system
-- Configuring library: ignition-gazebo3-physics-system
-- Adding 1 UNIT tests
-- Configuring library: ignition-gazebo3-pose-publisher-system
-- Configuring library: ignition-gazebo3-scene-broadcaster-system
-- Configuring library: ignition-gazebo3-sensors-system
-- Configuring library: ignition-gazebo3-thermal-system
-- Configuring library: ignition-gazebo3-touchplugin-system
-- Configuring library: ignition-gazebo3-triggered-publisher-system
-- Configuring library: ignition-gazebo3-user-commands-system
-- Configuring library: ignition-gazebo3-velocity-control-system
-- Configuring library: ignition-gazebo3-wheel-slip-system
-- Configuring library: ignition-gazebo3-wind-effects-system
CMake Error at src/msgs/CMakeLists.txt:2 (PROTOBUF_GENERATE_CPP):
  Unknown CMake command "PROTOBUF_GENERATE_CPP".

-- Configuring incomplete, errors occurred!
```

chapulina, 2021-08-03:
> I have protobuf/3.12.3 and cmake/3.16.5

I think your Protobuf version is too new. What OS are you in?

Would you mind using the issue template next time and filling the relevant info? Thanks! :v: 

---

Cross-referencing this related PR: https://github.com/ignitionrobotics/ign-gazebo/pull/715

traversaro, 2021-08-03:
Probably you need to define to `ON`  the [`protobuf_MODULE_COMPATIBLE`](https://github.com/protocolbuffers/protobuf/blob/3f5fc4df1de8e12b2235c3006593e22d6993c3f5/cmake/protobuf-options.cmake#L6) CMake option in ign-gazebo. The motivation is a bit intricate, but you can find some details in the PR linked by @chapulina : https://github.com/ignitionrobotics/ign-gazebo/pull/715 .

mboisson, 2021-08-03:
My apologies. I did not use the template because none of it is relevant, but still, here it is : 
OS Version: Gentoo on top of CentOS, using Lmod and EasyBuild to provide many pieces of the software. This is for a supercomputer. 
Source version: (version provided above) 
It is not a rendering error, but rather a compilation one. 

What version of protobuf is needed ?

The `PROTOBUF_GENERATE_CPP`  function does exist in our installation of protobuf. It seems that it is not about the function not existing, but rather that it is not being found: 
```
rep -ri PROTOBUF_GENERATE_CPP  $EBROOTPROTOBUF
/cvmfs/soft.computecanada.ca/easybuild/software/2020/avx2/Core/protobuf/3.12.3/lib64/cmake/protobuf/protobuf-module.cmake:function(PROTOBUF_GENERATE_CPP SRCS HDRS)
/cvmfs/soft.computecanada.ca/easybuild/software/2020/avx2/Core/protobuf/3.12.3/lib64/cmake/protobuf/protobuf-module.cmake:  cmake_parse_arguments(protobuf_generate_cpp "" "EXPORT_MACRO" "" ${ARGN})
/cvmfs/soft.computecanada.ca/easybuild/software/2020/avx2/Core/protobuf/3.12.3/lib64/cmake/protobuf/protobuf-module.cmake:  set(_proto_files "${protobuf_generate_cpp_UNPARSED_ARGUMENTS}")
/cvmfs/soft.computecanada.ca/easybuild/software/2020/avx2/Core/protobuf/3.12.3/lib64/cmake/protobuf/protobuf-module.cmake:    message(SEND_ERROR "Error: PROTOBUF_GENERATE_CPP() called without any proto files")
/cvmfs/soft.computecanada.ca/easybuild/software/2020/avx2/Core/protobuf/3.12.3/lib64/cmake/protobuf/protobuf-module.cmake:  if(PROTOBUF_GENERATE_CPP_APPEND_PATH)
/cvmfs/soft.computecanada.ca/easybuild/software/2020/avx2/Core/protobuf/3.12.3/lib64/cmake/protobuf/protobuf-module.cmake:  protobuf_generate(${_append_arg} LANGUAGE cpp EXPORT_MACRO ${protobuf_generate_cpp_EXPORT_MACRO} OUT_VAR _outvar ${_import_arg} PROTOS ${_proto_files})
/cvmfs/soft.computecanada.ca/easybuild/software/2020/avx2/Core/protobuf/3.12.3/lib64/cmake/protobuf/protobuf-module.cmake:  if(PROTOBUF_GENERATE_CPP_APPEND_PATH)
/cvmfs/soft.computecanada.ca/easybuild/software/2020/avx2/Core/protobuf/3.12.3/lib64/cmake/protobuf/protobuf-module.cmake:# By default have PROTOBUF_GENERATE_CPP macro pass -I to protoc
/cvmfs/soft.computecanada.ca/easybuild/software/2020/avx2/Core/protobuf/3.12.3/lib64/cmake/protobuf/protobuf-module.cmake:if(NOT DEFINED PROTOBUF_GENERATE_CPP_APPEND_PATH)
/cvmfs/soft.computecanada.ca/easybuild/software/2020/avx2/Core/protobuf/3.12.3/lib64/cmake/protobuf/protobuf-module.cmake:  set(PROTOBUF_GENERATE_CPP_APPEND_PATH TRUE)
``` 

I already tried patching the relevant (I think) `CMakeLists.txt` files with this patch: 
```
diff -ru ign-gazebo-ignition-gazebo3_3.8.0.orig/CMakeLists.txt ign-gazebo-ignition-gazebo3_3.8.0/CMakeLists.txt
--- ign-gazebo-ignition-gazebo3_3.8.0.orig/CMakeLists.txt       2021-08-03 16:04:54.039525404 +0000
+++ ign-gazebo-ignition-gazebo3_3.8.0/CMakeLists.txt    2021-08-03 16:05:36.884825979 +0000
@@ -131,6 +131,7 @@
 #--------------------------------------
 # Find protobuf
 set(REQ_PROTOBUF_VER 3)
+set(protobuf_MODULE_COMPATIBLE ON CACHE BOOL "")
 ign_find_package(IgnProtobuf
                  VERSION ${REQ_PROTOBUF_VER}
                  REQUIRED
diff -ru ign-gazebo-ignition-gazebo3_3.8.0.orig/src/msgs/CMakeLists.txt ign-gazebo-ignition-gazebo3_3.8.0/src/msgs/CMakeLists.txt
--- ign-gazebo-ignition-gazebo3_3.8.0.orig/src/msgs/CMakeLists.txt      2021-08-03 16:04:54.052525495 +0000
+++ ign-gazebo-ignition-gazebo3_3.8.0/src/msgs/CMakeLists.txt   2021-08-03 16:50:21.020668379 +0000
@@ -1,3 +1,4 @@
+set(protobuf_MODULE_COMPATIBLE ON CACHE BOOL "")
 PROTOBUF_GENERATE_CPP(PROTO_PRIVATE_SRC PROTO_PRIVATE_HEADERS
   peer_info.proto
   peer_control.proto
```

traversaro, 2021-08-03:
> I already tried patching the relevant (I think) `CMakeLists.txt` files with this patch:

I am afraid that patch will not work if `protobuf_MODULE_COMPATIBLE` CMake cache variable is already defined by a previous (transitive) call to `find_package(Protobuf CONFIG)` . I suggest to move the patch earlier in the CMakeLists.txt, or just pass it as a CMake command line option.

traversaro, 2021-08-03:
In a sense, probably we just need to move the line https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo5/CMakeLists.txt#L140 before any call to `find_package` (or at least any call that transitive search for Protobuf).

mboisson, 2021-08-03:
Adding to the `cmake` command line options worked, so I guess I can workaround it without the patch. 

Now I hit another compilation bug, which seems related to  `ignition-physics2`: 
```
26511 CMake Error in src/systems/physics/CMakeLists.txt:
26512   Imported target "ignition-physics2::ignition-physics2" includes
26513   non-existent path
26514
26515     "/cvmfs/soft.computecanada.ca/easybuild/software/2020/Core/eigen/3.3.7//cvmfs/soft.computecanada.ca/easybuild/software/2020/Core/eigen/3.3.7/include"
26516
26517   in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:
26518
26519   * The path was deleted, renamed, or moved to another location.
26520
26521   * An install or uninstall procedure did not complete successfully.
26522
26523   * The installation package was faulty and references files it does not
26524   provide.
``` 

I guess I should open a separate issue for this ?

The path should obviously be
`/cvmfs/soft.computecanada.ca/easybuild/software/2020/Core/eigen/3.3.7/include`
rather than
`/cvmfs/soft.computecanada.ca/easybuild/software/2020/Core/eigen/3.3.7//cvmfs/soft.computecanada.ca/easybuild/software/2020/Core/eigen/3.3.7/include` 

Not sure (yet) how `ignition-physics2` gets that separate path....

mboisson, 2021-08-03:
Any pointer about how those `cmake` files are generated ? It obviously detect eigen and writes it in this `cmake` file, but writes it incorrectly. 

```
$ grep -i eigen /home/mboisson/.local/easybuild/software/2020/avx2/Compiler/gcc9/ignition/citadel/lib64/cmake/ignition-physics2/ignition-physics2-targets.cmake
  INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include/ignition/physics2;/cvmfs/soft.computecanada.ca/easybuild/software/2020/Core/eigen/3.3.7//cvmfs/soft.computecanada.ca/easybuild/software/2020/Core/eigen/3.3.7/include"
  INTERFACE_SYSTEM_INCLUDE_DIRECTORIES "/cvmfs/soft.computecanada.ca/easybuild/software/2020/Core/eigen/3.3.7//cvmfs/soft.computecanada.ca/easybuild/software/2020/Core/eigen/3.3.7/include"
```

traversaro, 2021-08-03:
I guess the patch in https://github.com/ignitionrobotics/ign-physics/pull/282 should fix the problem (note that the PR is against ign-physics4, but I guess that probably a similar fix can be applied to ign-physics 2 .

traversaro, 2021-08-03:
See https://cmake.org/cmake/help/latest/manual/cmake-packages.7.html#creating-relocatable-packages for more details on CMake relocatable packages (that I guess it is what is required by easybuild, even if I am not an expert).

mboisson, 2021-08-03:
We do not (usually) install in place, indeed. So I guess they need to be relocatable. Thanks for the pointer. I will try the patch (or a similar patch).

traversaro, 2021-08-03:
Not sure what is your use case for ign-gazebo, but in that case probably you may be interested in https://github.com/ignitionrobotics/ign-gazebo/issues/626 .

mboisson, 2021-08-03:
🤔 , now the configuration passes, but the build fails... 

Maybe building gazebo is not such a good idea, but if it does not build, doesn't that mean that other components were not built correctly ?

```
26509 ignition/msgs/entity.proto: File not found.
26510 performer_affinity.proto:22:1: Import "ignition/msgs/entity.proto" was not found or had errors.
26511 performer_affinity.proto:29:3: "ignition.msgs.Entity" is not defined.
26512 ignition/msgs/header.proto: File not found.
26513 peer_control.proto:22:1: Import "ignition/msgs/header.proto" was not found or had errors.
26514 peer_control.proto:30:3: "ignition.msgs.Header" is not defined.
26515 make[2]: *** [src/msgs/CMakeFiles/ignition-gazebo_private_msgs.dir/build.make:84: src/msgs/performer_affinity.pb.h] Error 1
26516 make[2]: *** Waiting for unfinished jobs....
26517 make[2]: *** [src/msgs/CMakeFiles/ignition-gazebo_private_msgs.dir/build.make:76: src/msgs/peer_control.pb.h] Error 1
26518 ignition/msgs/header.proto: File not found.
26519 peer_info.proto:22:1: Import "ignition/msgs/header.proto" was not found or had errors.
26520 peer_info.proto:28:3: "ignition.msgs.Header" is not defined.
26521 make[2]: *** [src/msgs/CMakeFiles/ignition-gazebo_private_msgs.dir/build.make:68: src/msgs/peer_info.pb.h] Error 1
26522 ignition/msgs/world_stats.proto: File not found.
26523 ignition/msgs/entity.proto: File not found.
26524 performer_affinity.proto:22:1: Import "ignition/msgs/entity.proto" was not found or had errors.
26525 performer_affinity.proto:29:3: "ignition.msgs.Entity" is not defined.
26526 simulation_step.proto:22:1: Import "ignition/msgs/world_stats.proto" was not found or had errors.
26527 simulation_step.proto:23:1: Import "performer_affinity.proto" was not found or had errors.
26528 simulation_step.proto:32:3: "ignition.msgs.WorldStatistics" is not defined.
26529 simulation_step.proto:36:12: "PerformerAffinity" is not defined.
```

traversaro, 2021-08-03:
> Our user actually asked for ORSF Gazebo 11, and it depends on a lot of the `ignition-citadel` components, so I figured I would install all of the components, and then build ORSF Gazebo 11 using `ignitition-citadel` as a dependency. Is that a bad idea ? Should I skip some components ?

It is not a bad idea, but probably it is more work then necessary, as for Gazebo 11 you will not need ignition-gazebo or ignition-physics, that are exactly the one that are giving you problems. 

However, I guess we are definitely OT at this point here, so perhaps it could make sense to open an issue in osrf/gazebo repo instead?

mboisson, 2021-08-03:
Ok. I will do this.

closed 2021-08-03 by mboisson

chapulina, 2021-08-05:
> build ORSF Gazebo 11 using ignitition-citadel as a dependency.

If it helps, instead of building the entire [collection-citadel](https://github.com/ignition-tooling/gazebodistro/blob/master/collection-citadel.yaml) from source as instructed on our docs, you can just get the Gazebo 11 dependencies from [gazebo11.yaml](https://github.com/ignition-tooling/gazebodistro/blob/master/gazebo11.yaml).