gazebosim/gz-sim #963: GUI fails to render world with simple models and hello_world plugin.

vinaymahamuni, 2021-08-10:
## Environment
* OS Version: Ubuntu 18.04
* Source or binary build? = binary[ignition-citadel]
* Rendering plugin: ogre2. 
 
-   ogre2.log file from  `~/.ignition/rendering` 
    
<details>
<summary> ogre2.log file</summary>
 
```
14:35:43: Creating resource group General
14:35:43: Creating resource group Internal
14:35:43: Creating resource group Autodetect
14:35:43: SceneManagerFactory for type 'DefaultSceneManager' registered.
14:35:43: Registering ResourceManager for type Material
14:35:43: Registering ResourceManager for type Mesh
14:35:43: Registering ResourceManager for type Mesh2
14:35:43: Registering ResourceManager for type OldSkeleton
14:35:43: MovableObjectFactory for type 'ParticleSystem' registered.
14:35:43: ArchiveFactory for archive type FileSystem registered.
14:35:43: ArchiveFactory for archive type Zip registered.
14:35:43: ArchiveFactory for archive type EmbeddedZip registered.
14:35:43: DDS codec registering
14:35:43: FreeImage version: 3.17.0
14:35:43: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
14:35:43: Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,cut,xbm,xpm,gif,hdr,g3,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
14:35:43: OITD codec registering
14:35:43: Registering ResourceManager for type HighLevelGpuProgram
14:35:43: MovableObjectFactory for type 'Entity' registered.
14:35:43: MovableObjectFactory for type 'Item' registered.
14:35:43: MovableObjectFactory for type 'Light' registered.
14:35:43: MovableObjectFactory for type 'BillboardSet' registered.
14:35:43: MovableObjectFactory for type 'ManualObject2' registered.
14:35:43: MovableObjectFactory for type 'BillboardChain' registered.
14:35:43: MovableObjectFactory for type 'RibbonTrail' registered.
14:35:43: MovableObjectFactory for type 'WireAabb' registered.
14:35:43: *-*-* OGRE Initialising
14:35:43: *-*-* Version 2.1.0unstable ('B')
14:35:43: OverlayElementFactory for type Panel registered.
14:35:43: OverlayElementFactory for type BorderPanel registered.
14:35:43: OverlayElementFactory for type TextArea registered.
14:35:43: Registering ResourceManager for type Font
14:35:43: Loading library /usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so
14:35:43: Installing plugin: GL 3+ RenderSystem
14:35:43: OpenGL 3+ Rendering Subsystem created.
14:35:43: Plugin successfully installed
14:35:43: Loading library /usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/Plugin_ParticleFX.so
14:35:43: Installing plugin: ParticleFX
14:35:43: Particle Emitter Type 'Point' registered
14:35:43: Particle Emitter Type 'Box' registered
14:35:43: Particle Emitter Type 'Ellipsoid' registered
14:35:43: Particle Emitter Type 'Cylinder' registered
14:35:43: Particle Emitter Type 'Ring' registered
14:35:43: Particle Emitter Type 'HollowEllipsoid' registered
14:35:43: Particle Affector Type 'LinearForce' registered
14:35:43: Particle Affector Type 'ColourFader' registered
14:35:43: Particle Affector Type 'ColourFader2' registered
14:35:43: Particle Affector Type 'ColourImage' registered
14:35:43: Particle Affector Type 'ColourInterpolator' registered
14:35:43: Particle Affector Type 'Scaler' registered
14:35:43: Particle Affector Type 'Rotator' registered
14:35:43: Particle Affector Type 'DirectionRandomiser' registered
14:35:43: Particle Affector Type 'DeflectorPlane' registered
14:35:43: Plugin successfully installed
14:35:43: CPU Identifier & Features
14:35:43: -------------------------
14:35:43:  *   CPU ID: GenuineIntel: Intel(R) Core(TM) i7-7820HQ CPU @ 2.90GHz
14:35:43:  *   Logical cores: 8
14:35:43:  *      SSE: yes
[... 768 lines cut ...]
[Msg] Hello, world! Simulation is paused.
[Msg] Hello, world! Simulation is paused.
[Msg] Hello, world! Simulation is paused.
[Msg] Hello, world! Simulation is paused.
[Msg] Hello, world! Simulation is paused.
[Msg] Hello, world! Simulation is paused.
[Msg] Hello, world! Simulation is paused.
[Msg] Hello, world! Simulation is paused.
[Msg] Hello, world! Simulation is paused.
[Msg] Hello, world! Simulation is paused.
[Msg] Hello, world! Simulation is paused.
[Msg] Hello, world! Simulation is paused.
[Msg] Hello, world! Simulation is paused.
[Msg] Hello, world! Simulation is paused.
[Msg] Hello, world! Simulation is paused.
[Msg] Hello, world! Simulation is paused.
[Msg] Hello, world! Simulation is paused.
[Msg] Hello, world! Simulation is paused.
[GUI] [Dbg] [Scene3D.cc:2662] RenderEngineGuiPlugin component not found, render engine won't be set from the ECM
[Msg] Hello, world! Simulation is paused.
[Msg] Hello, world! Simulation is paused.
```

</details>
[image]

adlarkin, 2021-08-12:
I'm noticing the same behavior in https://github.com/ignitionrobotics/ign-gazebo/commit/c2834e0c99dff57a810ad833e09a3d27b09c3b4b#r54802891 (that commit branched off of Edifice). I'm also able to re-produce the behavior when following the steps in this issue. I have found that in both the commit I linked and in the example in this issue, commenting out the plugin from the SDF file results in models being rendered as expected.

vinaymahamuni, 2021-08-16:
Thanks. Adding few systems to the world helped 
```
<plugin filename="ignition-gazebo-physics-system" name="ignition::gazebo::systems::Physics">
</plugin> 

<plugin filename="ignition-gazebo-sensors-system" name="ignition::gazebo::systems::Sensors">   
   <render_engine>ogre2</render_engine>
</plugin>

<plugin filename="ignition-gazebo-user-commands-system" name="ignition::gazebo::systems::UserCommands">
</plugin>

<plugin filename="ignition-gazebo-scene-broadcaster-system" name="ignition::gazebo::systems::SceneBroadcaster">
</plugin>
```

adlarkin, 2021-08-16:
I'm closing this issue since it has been resolved/discussed in https://github.com/ignitionrobotics/ign-gazebo/commit/c2834e0c99dff57a810ad833e09a3d27b09c3b4b#r54822602.

closed 2021-08-16 by adlarkin

emapall, 2021-08-20:
Thanks a lot!
I had the same problem and the fix worked for me as well. 
Appreciated this post a lot.