gazebosim/gz-sim #966: VelocityCmd components applied to models don't have any effect

luca-della-vedova, 2021-08-11:
## Environment

Ubuntu 20.04, Ignition Edifice source build (latest ign-gazebo5 branch).

## Description

We relied on using `LinearVelocityCmd` and `AngularVelocityCmd` to apply velocities at the model level in our [robot simulation plugins](https://github.com/open-rmf/rmf_simulation/blob/main/rmf_robot_sim_ignition_plugins/src/slotcar.cpp#L155-L167). However, I noticed that recently the commands don't have any effect.
An example SDF that we use this plugin with is [here](https://github.com/open-rmf/rmf_demos/blob/main/rmf_demos_assets/models/TinyRobot/model.sdf).
I ran `git bisect` and found out the exact commit that introduced this behavior, #678. I didn't look into the commit diff too much but my guess is that, since we are applying the velocity to a model entity and not a link entity it is skipped by that optimization PR.
I wonder if there was a recommended way to adapt our plugin or if this is something that should be fixed in the core code. I tried to fetch the model's canonical link through the `ModelCanonicalLink` component and applying the velocities to it but didn't have much success.

## Steps to reproduce

It is arguably a bit complicated, it would require setting up the whole `rmf_demos` workspace, but let me know if you want detailed instructions.

adlarkin, 2021-08-13:
Thanks for pointing out the issue. I have written a test that does fail to move a model when applying a `LinearVelocityCmd` component: https://github.com/ignitionrobotics/ign-gazebo/commit/8afb93776820cb83a7f9743aa2bcbd4627efb38a (this branches off of `ign-gazebo5`)

However, I am unable to see a behavior change before/after #678. I have tried running this test before and after #678, but it always fails no matter which commit/release of `ign-gazebo5` I use (the model never seems to move). Would you mind taking a look at the test I wrote and try to run it locally to see if you can get it to pass/fail based on #678? Also, if you see anything wrong with the test I wrote, feel free to comment on the commit that I linked.

iche033, 2021-08-13:
oh and is the demo using dart or tpe?

luca-della-vedova, 2021-08-13:
> Thanks for pointing out the issue. I have written a test that does fail to move a model when applying a `LinearVelocityCmd` component: [8afb937](https://github.com/ignitionrobotics/ign-gazebo/commit/8afb93776820cb83a7f9743aa2bcbd4627efb38a) (this branches off of `ign-gazebo5`)
> 
> However, I am unable to see a behavior change before/after #678. I have tried running this test before and after #678, but it always fails no matter which commit/release of `ign-gazebo5` I use (the model never seems to move). Would you mind taking a look at the test I wrote and try to run it locally to see if you can get it to pass/fail based on #678? Also, if you see anything wrong with the test I wrote, feel free to comment on the commit that I linked.

It seems the test world `lin_vel_cmd.sdf` was not committed, can you add it to the repo so I can try on my end?

> oh and is the demo using dart or tpe?
It happened on both dart and TPE

adlarkin, 2021-08-13:
> It seems the test world lin_vel_cmd.sdf was not committed, can you add it to the repo so I can try on my end?

Sorry about that! I've added the world in https://github.com/ignitionrobotics/ign-gazebo/commit/f2cba0bfab480a5ad77805901aa23f253ba39f6a

adlarkin, 2021-08-13:
@luca-della-vedova I am able to run the following example that applies a command velocity to a model, which moves the model: https://github.com/ignitionrobotics/ign-gazebo/compare/adlarkin/linVelCmdTest. This example is based on the latest commit of `ign-gazebo5`, which means #678 is used.

@iche033 pointed out that some plugins aren't loaded if a user specifies their own plugins in a SDF file (https://github.com/ignitionrobotics/ign-gazebo/commit/c2834e0c99dff57a810ad833e09a3d27b09c3b4b#r54822602). In the SDF file you provided in the issue description, it looks like you define a custom plugin, which means that you'll probably need to add in other plugins like physics in order to get desired behavior. Can you try modifying your SDF file based on the recommendations in https://github.com/ignitionrobotics/ign-gazebo/commit/c2834e0c99dff57a810ad833e09a3d27b09c3b4b#r54822602 and let me know if the issue still persists for you?

luca-della-vedova, 2021-08-13:
For reference [this](https://gist.github.com/luca-della-vedova/14f87ff16c0f82bd28e302ab4ddfc226) is the world, we actually redefine all the systems / GUI elements to be safe. I'll try your examples now and see how they behave

luca-della-vedova, 2021-08-13:
OK it has been quite the debugging journey!
I apologize for my previous information because it was partially incorrect. It turns out writing to `VelocityCmd` components works, as you showed, reading is what doesn't, and is the behavior that changed.
Specifically, in our application we are simulating vehicles with a limited acceleration hence we apply speed in the format of:
`v_new = v_old + dt * a`, we rely, or used to, on [reading](https://github.com/open-rmf/rmf_simulation/blob/main/rmf_robot_sim_ignition_plugins/src/slotcar.cpp#L155-L161) the `VelocityCmd` components to get the velocity command that was sent in the previous iteration and this seemed to work until the PR I mentioned above (although to be honest I'm really not sure what changes in the PR might have affected this).
I think it is not quite correct to ready `VelocityCmd` components since I would expect them to be reset at every iteration step, however I tried to do something more proper which is read `LinearVelocity` components but they also seem to be always read as 0 regardless of the commit, so at least reading the `Cmd` components had something working before.
I pushed an example that is closer to what we actually do in `luca/linVelTest`, the relevant code is [here](https://github.com/ignitionrobotics/ign-gazebo/blob/luca/linVelTest/src/systems/lin_vel_cmd/LinVelCmdTest.cc#L37-L40) and I changed the world to download a model from fuel and the model has the plugin tag inside.
This code has a velocity ramp before #678 while it always sends a velocity of 0.01 after. The previous commit (using `LinearVelocity` instead of `LinearVelocityCmd` to read the current velocity) sends 0.01 regardles.

Apologies for the wrong information but I hope this example can help, maybe we are doing something wrong and actually relied on some buggy behavior before?

adlarkin, 2021-08-13:
> we rely, or used to, on reading the VelocityCmd components to get the velocity command that was sent in the previous iteration
>
> I think it is not quite correct to read VelocityCmd components since I would expect them to be reset at every iteration step

You're assumption is correct: looking at the physics system (`ign-gazebo5.1.0`), linear velocity commands are [applied to models](https://github.com/ignitionrobotics/ign-gazebo/blob/ignition-gazebo5_5.1.0/src/systems/physics/Physics.cc#L1714-L1715) ([and links](https://github.com/ignitionrobotics/ign-gazebo/blob/ignition-gazebo5_5.1.0/src/systems/physics/Physics.cc#L1817-L1818)) when it's time to update physics with the next step. After the update is completed, the linear velocity commands are ["zeroed out"](https://github.com/ignitionrobotics/ign-gazebo/blob/ignition-gazebo5_5.1.0/src/systems/physics/Physics.cc#L2505-L2510). So, linear velocity commands should not be read from after a physics update.

> I'm really not sure what changes in the PR might have affected this

At first glance, I'm not sure either :joy: #678 did not touch any of the code that applies/resets linear velocities via command components. So, I'll have to take a closer look at how #678 changed behavior.

> Apologies for the wrong information but I hope this example can help, maybe we are doing something wrong and actually relied on some buggy behavior before?

No apology needed! It sounds like we've had buggy behavior that has gone unnoticed until now :upside_down_face: Based on our discussion, I think there are a few things that need to be addressed:
1. Understand how #678 changed (or did not change) behavior with `LinearVelocityCmd`
2. Make sure that the `LinearVelocity` component gets updated properly (you mentioned that `LinearVelocity` never seems to update even after applying a `LinearVelocityCmd` - this appears to be a bug)

I will go ahead and work on addressing the two things I mentioned above (if there's anything else you'd like me to investigate, let me know). Once everything has been addressed, the correct behavior on your end should be to set `LinearVelocityCmd` components on models/links that you want to move before the next physics update, and then use the `LinearVelocity` component after the physics update to read the most recent velocity of a model/link.

adlarkin, 2021-08-24:
@luca-della-vedova I have proposed a fix for links in #992 that would allow for users to apply a `LinearVelocityCmd` to a link, and then read the link's linear velocity at any future point through the link's `LinearVelocity` or `WorldLinearVelocity` component. As we have already discussed in this thread, before #992, if a `LinearVelocityCmd` is applied to a link, the `LinearVelocity` or `WorldLinearVelocity` components are not created, so there's no way to retrieve the link's linear velocity in future simulation steps if needed.

I made this change in Citadel because I found that this behavior is broken for all released versions. So, what we can do is merge #992 forward to other released versions if it seems like a good solution. I have not made the fix for when `LinearVelocityCmd`s are applied to models yet because that's a little more complicated to do in Citadel/Dome (see the PR description of #992), but I believe the same fix can be made for models fairly easily starting in Edifice.

So, for now, assuming that #992 gets merged, I believe what you can do to fix the buggy behavior in RMF is apply `LinearVelocityCmd`s to links, and then read the `LinearVelocity` or `WorldLinearVelocity` component from the corresponding link. Does this make sense, and does this also appear to be a valid solution to the issue you're currently facing?

azeey, 2021-08-31:
It looks like the current implementation with DART as the physics engine does not compensate for friction or any other forces when applying a `LinearVelocityCmd`. It simply sets the current state of the link and when the next `step` is called, DART computes the next velocity and position starting from the current velocity set via `LinearVelocityCmd`.

This, unfortunately, does not follow the pattern we have for `JointVelocity[Cmd|Reset]` where, `JointVelocityCmd` compensates for joint friction and other effects (within joint force limits) while `JointVelocityReset` sets the state of the joint and does not compensate for joint friction. 

> I guess my final question is whether it is intended that when applying a VelocityCmd component the behavior depends on the friction...

So to answer the question, I don't think it was intended. A better name for the current `LinearVelocityCmd` would have been `LinearVelocityReset`.

adlarkin, 2021-09-01:
Thanks for the explanation! I think that @luca-della-vedova is using TPE though, so if the behavior that was explained in https://github.com/ignitionrobotics/ign-gazebo/issues/966#issuecomment-906233694 is when TPE is used, then I assume that we can fix the TPE behavior since it's a plugin inside of `ign-physics`, right?

luca-della-vedova, 2021-09-01:
> Thanks for the explanation! I think that @luca-della-vedova is using TPE though, so if the behavior that was explained in [#966 (comment)](https://github.com/ignitionrobotics/ign-gazebo/issues/966#issuecomment-906233694) is when TPE is used, then I assume that we can fix the TPE behavior since it's a plugin inside of `ign-physics`, right?

TPE was only used as a fallback after we observed the issues with friction that made controlling robots through `VelocityCmd` components a bit tough on dartsim.
I think with the TPE fix and doing "open loop" control in dartsim without checking the current model velocity we are able to cover all our use cases and the DART behavior is not really a bug, just a feature.
We can close this, thanks all for the help!

closed 2021-09-01 by luca-della-vedova