gazebosim/gz-sim #981: Support spherical coordinates

chapulina, 2021-08-18:
The SDFormat spec provides the [//world/spherical_coordinates](http://sdformat.org/spec?ver=1.7&elem=world#world_spherical_coordinates) element which allows defining latitude and longitude for the origin of a simulated world. We should support that in Ignition Gazebo.

## Desired behavior

It should be possible to query the pose of any entity in the world in spherical coordinates, using as reference the coordinates set at startup. For example, for this world:

```xml
  <world name="flat_earth">

    <spherical_coordinates>
      <surface_model>EARTH_WGS84</surface_model>
      <world_frame_orientation>ENU</world_frame_orientation>
      <latitude_deg>22.9</latitude_deg>
      <longitude_deg>43.2</longitude_deg>
      <elevation>0</elevation>
    </spherical_coordinates>

    <model name="robot">
      <pose>0 0 0.0 0 0 0</pose>
      ...
    </model>

  </world>
```

Developers should be able to query:

```cpp
auto latLonElev = sphericalCoords(_robotEntity, _ecm);
EXPECT_EQ(latLonElev.X(), 22.9);
EXPECT_EQ(latLonElev.Y(), 43.2);
EXPECT_EQ(latLonElev.Z(), 0);
```

## Alternatives considered

We could choose not to include spherical coordinates as part of the core simulator and leave it up to each plugin to figure it out. But since the SDF spec supports it and several plugins can benefit from it, I think it's worth it providing core functionality for it.

## Implementation suggestion

Gazebo Classic supports that through a [gazebo::common::SphericalCoordinates](http://osrf-distributions.s3.amazonaws.com/gazebo/api/11.0.0/classgazebo_1_1common_1_1SphericalCoordinates.html) object belonging to `gazebo::physics::World`, which is populated [here](https://github.com/osrf/gazebo/blob/b502f17729302fa0da8e2a00ffbd69112bb7e6aa/gazebo/physics/World.cc#L343-L355). All other classes that want to make use of spherical coordinates can query that.

That class has been ported to Ignition Math: [ignition::math::SphericalCoordinates](https://ignitionrobotics.org/api/math/6.8/classignition_1_1math_1_1SphericalCoordinates.html), which we can use here.

I'd suggest:

1. Parse `<spherical_coordinates>` into `sdf::World`
1. Create a `component::SphericalCoordinates` that wraps `ignition::math::SphericalCoordinates`
1. Add that component to the `World` entity when parsing the SDF
1. Provide a helper function in `Util.hh` that returns the spherical coordinates for any entity according to its pose and the world's component

## Additional context

We should make sure we don't inherit issue https://github.com/osrf/gazebo/issues/2022 from Gazebo classic. We may also want to address https://github.com/ignitionrobotics/ign-math/issues/51.

chapulina, 2021-09-01:
Other nice-to-have features:

* Updating the world origin coordinates via an `ign-transport` service
* Support for spherical coordinates when inserting new models through the `/create` service
* Support for spherical coordinates when moving models through the `/set_pose` (or equivalent) service
* Displaying spherical coordinates on the component inspector

chapulina, 2021-09-01:
I started working on this issue and I'm trying to understand how the SDF element `<spherical_coordinates><world_frame_orientation>` is meant to be used.

I tracked down some background on that element that I'll summarize here for future reference:

I believe that tag was added to the SDFormat spec in the context of https://github.com/osrf/gazebo/issues/1959:

* [SDFormat PR (274)](https://osrf-migration.github.io/sdformat-gh-pages/#!/osrf/sdformat/pull-requests/274/page/1?slug=add-reference_frame-attribute-for-imu) was closed and reopened as [PR 284](https://osrf-migration.github.io/sdformat-gh-pages/#!/osrf/sdformat/pull-requests/284/page/1), which was merged. 
* That [added](https://github.com/ignitionrobotics/sdformat/compare/5c43578131a4522ec9c07a630dc058c825cb66c0..a0ce387f082f1176d21a7526a6deae6938c3520a) the following to the SDF spec:

    * `<imu><orientation_reference_frame>`
    * `<spherical_coordinates><world_frame_orientation>`

* The merged PR points to branch [issue_1959_gazebo7](https://github.com/osrf/gazebo/compare/gazebo11...issue_1959_gazebo7) on Gazebo classic as the corresponding changes to Gazebo, but it looks like that was never merged.
* Also, searching for `orientation_reference_frame` or `world_frame_orientation` on Gazebo 11 gives no results. So I think it's safe to say that we don't have a working reference implementation for how `world_frame_orientation` should be used.

I'm still looking into it; I may either ignore that element for now (and forever?), or port part of that Gazebo 11 branch to Ignition Math.

scpeters, 2021-09-02:
you're right, it looks like we never started using the `<world_frame_orientation />` tag in gazebo, though it's not only relevant to IMU's; it would also affect the output of gazebo's GPS sensor. To be honest, our SphericalCoordiantes class is very rudimentary. If there was a way to use [proj](https://github.com/OSGeo/PROJ), I think that would be the most powerful approach. Maybe @mjcarroll has an opinion based on his experience with NavSat hardware?

chapulina, 2021-09-03:
Thanks for the input, @scpeters . I looked a bit at proj and it seems to be pretty powerful. I think we definitely should leave the architecture flexible in case we want to integrate it later, but for now, I was thinking of keeping it simple and using the class that we already have. My current idea:

1. Fix the most pressing bug with `SphericalCoordinates`, see https://github.com/ignitionrobotics/ign-math/pull/235
2. Add `ignition::math::SphericalCoordinates sdf::World::SphericalCoordinates`, ignoring `<world_frame_orientation>`
3. Add a `gazebo::component::SphericalCoordinates` component that wraps `math::SphericalCoordinates`
4. Provide the utility function and services described above.

In the future, if we need more precision than what Ignition Math is providing and want to integrate `proj`, we can create a system plugin that reads / writes the reference lat / lon / elevation / heading to/from `gazebo::component::SphericalCoordinates` and offers alternative transport services. In this scenario, `math::SphericalCoordinates` is just used to hold the data but not for computation.

Let me know what you think of this 2-step approach.

PR #1008 (2021-09-22): 🌐 Spherical coordinates
# 🎉 New feature

Closes #981

Requires
* https://github.com/ignitionrobotics/ign-math/pull/235
* https://github.com/ignitionrobotics/sdformat/pull/685
* https://github.com/ignitionrobotics/ign-msgs/pull/177

## Summary

Add support for spherical coordinates (latitude / longitude). Users are able to set the lat / lon for the world origin, and query the lat / lon for other entities with respect to that.

* Set from SDF, see example world
* See on component inspector
* Plot world origin coordinates
* Change world origin coordinates from plugins
* Change world origin coordi [... cut]