gazebosim/gz-sim #999: Add function to the EntityComponentManager :get the entity from the id

0smile, 2021-08-30:
## Desired behavior
Given an id, return the entity which the id belongs to.

chapulina, 2021-08-30:
Hi @0smile , could you elaborate on what you mean by the "entity"? Entities in Ignition Gazebo are just IDs, see the [terminology tutorial](https://ignitionrobotics.org/api/gazebo/5.1/terminology.html).

0smile, 2021-08-31:
> Hi @0smile , could you elaborate on what you mean by the "entity"? Entities in Ignition Gazebo are just IDs, see the [terminology tutorial](https://ignitionrobotics.org/api/gazebo/5.1/terminology.html).

Hi chapulina, 
The ”entity“ I mentioned refers to the ”Entity“ in the definition of Ignition. My requirement to the function is to give a number of type int or long, and the function can return the entity of the corresponding number.

My requirement comes from that I extracted the ID of the collision object from contact sensor message. The type is long (although I can also extract the corresponding entity of the object from the msg, the type is msgs::Entity, so it cannot be operated by EntityComponentManager). Now I need to find the corresponding gazebo::Entity instance with this ID.

chapulina, 2021-08-31:
> the ID of the collision object from contact sensor message

That should be the `Entity` you want. You can do `gazebo::Entity collisionEntity = msg.id();`

0smile, 2021-09-01:
> > the ID of the collision object from contact sensor message
> 
> That should be the `Entity` you want. You can do `gazebo::Entity collisionEntity = msg.id();`

Thank you very much chapulina, you solved my question. I will close the issue.

closed 2021-09-01 by 0smile