gazebosim/gz-sim #1004: ign-gazebo5 Docker build fails; unable to locate package dart6-data

playertr, 2021-08-31:
## Environment
* OS Version:  Ubuntu 18.04
* Source or binary build?
Source: Branch `ign-gazebo5`, commit `3c1a591ce1`

[Omitted rendering-related system information; problem affects Docker build]

## Description
When I run the [Docker build instructions ](https://github.com/ignitionrobotics/ign-gazebo/tree/ign-gazebo5/docker) from the ign-gazebo5 branch, installation fails with: 
```
E: Unable to locate package dart6-data
E: Package 'libdart6-collision-ode-dev' has no installation candidate
E: Package 'libdart6-dev' has no installation candidate
E: Package 'libdart6-utils-urdf-dev' has no installation candidate
The command '/bin/sh -c scripts/install_ign_deps.sh' returned a non-zero code: 100
```
* Expected behavior: I expected the Docker image for ign-gazebo:base to complete building. 

Note 1: I did not first build ign-gazebo locally from source. Am I supposed to do this first?

Note 2: I would prefer to use an official image pulled from Docker Hub, but I was not able to find one.

* Actual behavior: The Docker build command failed.

## Steps to reproduce

1. Clone `ign-gazebo`.
```
git clone https://github.com/ignitionrobotics/ign-gazebo.git

```
2. Attempt to build the ign-gazebo:base image.
```
cd ign-gazebo/
docker build . -f ./docker/Dockerfile.base -t ign-gazebo:base
```

## Output
<details>
<summary> Console Output </summary>

```
tim@tim-UBUNTU:~/Research/ign-gazebo$ docker build . -f ./docker/Dockerfile.base -t ign-gazebo:base
Sending build context to Docker daemon  93.83MB
Step 1/10 : FROM ubuntu:focal
focal: Pulling from library/ubuntu
35807b77a593: Already exists 
Digest: sha256:9d6a8699fb5c9c39cf08a0871bd6219f0400981c570894cd8cbea30d3424a31f
Status: Downloaded newer image for ubuntu:focal
 ---> fb52e22af1b0
Step 2/10 : RUN apt-get update  && apt-get install -y       sudo wget lsb-release gnupg curl  && apt-get clean
 ---> Running in 4cea2d80a490
Get:1 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]
Get:2 http://archive.ubuntu.com/ubuntu focal InRelease [265 kB]
Get:3 http://security.ubuntu.com/ubuntu focal-security/restricted amd64 Packages [488 kB]
Get:4 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages [1036 kB]
Get:5 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [790 kB]
Get:6 http://security.ubuntu.com/ubuntu focal-security/multiverse amd64 Packages [30.1 kB]
Get:7 http://archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]
Get:8 http://archive.ubuntu.com/ubuntu focal-backports InRelease [101 kB]
Get:9 http://archive.ubuntu.com/ubuntu focal/restricted amd64 Packages [33.4 kB]
Get:10 http://archive.ubuntu.com/ubuntu focal/multiverse amd64 Packages [177 kB]
Get:11 http://archive.ubuntu.com/ubuntu focal/main amd64 Packages [1275 kB]
Get:12 http://archive.ubuntu.com/ubuntu focal/universe amd64 Packages [11.3 MB]
Get:13 http://archive.ubuntu.com/ubuntu focal-updates/universe amd64 Packages [1068 kB]
Get:14 http://archive.ubuntu.com/ubuntu focal-updates/multiverse amd64 Packages [33.4 kB]
Get:15 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 Packages [1478 kB]
Get:16 http://archive.ubuntu.com/ubuntu focal-updates/restricted amd64 Packages [535 kB]
Get:17 http://archive.ubuntu.com/ubuntu focal-backports/main amd64 Packages [2668 B]
Get:18 http://archive.ubuntu.com/ubuntu focal-backports/universe amd64 Packages [6324 B]
[... 2911 lines cut ...]
Get:64 http://archive.ubuntu.com/ubuntu focal/main amd64 libvulkan-dev amd64 1.2.131.2-1 [482 kB]
Get:65 http://archive.ubuntu.com/ubuntu focal/main amd64 x11proto-xext-dev all 2019.2-1ubuntu1 [2616 B]
Get:66 http://archive.ubuntu.com/ubuntu focal/main amd64 libxext-dev amd64 2:1.3.4-0ubuntu1 [82.2 kB]
Get:67 http://archive.ubuntu.com/ubuntu focal/universe amd64 qml-module-qt-labs-folderlistmodel amd64 5.12.8-0ubuntu1 [37.0 kB]
Get:68 http://archive.ubuntu.com/ubuntu focal/universe amd64 qml-module-qt-labs-settings amd64 5.12.8-0ubuntu1 [29.2 kB]
Get:69 http://archive.ubuntu.com/ubuntu focal/universe amd64 qml-module-qtquick-templates2 amd64 5.12.8+dfsg-0ubuntu1 [61.2 kB]
Get:70 http://archive.ubuntu.com/ubuntu focal/universe amd64 qml-module-qtquick-controls2 amd64 5.12.8+dfsg-0ubuntu1 [1318 kB]
Get:71 http://archive.ubuntu.com/ubuntu focal/universe amd64 qml-module-qtquick-privatewidgets amd64 5.12.8-0ubuntu2 [42.5 kB]
Get:72 http://archive.ubuntu.com/ubuntu focal/universe amd64 qml-module-qtquick-dialogs amd64 5.12.8-0ubuntu2 [118 kB]
Get:73 http://archive.ubuntu.com/ubuntu focal/universe amd64 qt5-qmake-bin amd64 5.12.8+dfsg-0ubuntu1 [990 kB]
Get:74 http://archive.ubuntu.com/ubuntu focal/universe amd64 qtchooser amd64 66-2build1 [24.7 kB]
Get:75 http://archive.ubuntu.com/ubuntu focal/universe amd64 qt5-qmake amd64 5.12.8+dfsg-0ubuntu1 [189 kB]
Get:76 http://archive.ubuntu.com/ubuntu focal/universe amd64 qt5-qmltooling-plugins amd64 5.12.8-0ubuntu1 [180 kB]
Get:77 http://archive.ubuntu.com/ubuntu focal/universe amd64 qtbase5-dev-tools amd64 5.12.8+dfsg-0ubuntu1 [700 kB]
Get:78 http://archive.ubuntu.com/ubuntu focal/universe amd64 qtbase5-dev amd64 5.12.8+dfsg-0ubuntu1 [991 kB]
Get:79 http://archive.ubuntu.com/ubuntu focal/universe amd64 qtdeclarative5-dev-tools amd64 5.12.8-0ubuntu1 [430 kB]
Get:80 http://archive.ubuntu.com/ubuntu focal/universe amd64 qtdeclarative5-dev amd64 5.12.8-0ubuntu1 [359 kB]
Get:81 http://archive.ubuntu.com/ubuntu focal/universe amd64 qtquickcontrols2-5-dev amd64 5.12.8+dfsg-0ubuntu1 [10.5 kB]
debconf: delaying package configuration, since apt-utils is not installed
Fetched 21.9 MB in 6s (3910 kB/s)
Selecting previously unselected package libdouble-conversion3:amd64.
(Reading database ... 31176 files and directories currently installed.)
Preparing to unpack .../00-libdouble-conversion3_3.1.5-4ubuntu1_amd64.deb ...
Unpacking libdouble-conversion3:amd64 (3.1.5-4ubuntu1) ...
Selecting previously u

chapulina, 2021-08-31:
Thanks for the report. I believe the problem is on these lines:

https://github.com/ignitionrobotics/ign-gazebo/blob/3c1a591ce110a985dbb45274f7f835d800080289/docker/scripts/install_ign_deps.sh#L70-L73

Those packages are available on Bionic, you can compare against this list:

https://github.com/ignitionrobotics/ign-gazebo/blob/3c1a591ce110a985dbb45274f7f835d800080289/.github/ci/packages-bionic.apt#L1-L4

But the image is building on Focal, so it should use these packages:

https://github.com/ignitionrobotics/ign-gazebo/blob/3c1a591ce110a985dbb45274f7f835d800080289/.github/ci/packages-focal.apt#L1-L5

Would you be able to give that change a try and open a pull request if it works for you?

playertr, 2021-08-31:
Those changes got the image to build! Thank you.

I'm now tracking down a different bug that is preventing me from seeing a GUI within the Docker container. I don't think that is related to these changes to the build script, so I opened the PR for these changes.

playertr, 2021-08-31:
The bug that I was experiencing that prevented me from running in the GUI was as follows:

The `base` version of the Docker image, built with `[Dockerfile.base](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo5/docker/Dockerfile.base)`, did not have OpenGL support. Therefore, when I ran `ign gazebo shapes.sdf` from within an image, no GUI would appear. 

The image built by `Dockerfile.ignition` got around this problem by starting from the `nvidia/opengl:1.2-glvnd-devel-ubuntu20.0` image, which comes with OpenGL support. However, that file uses `ign-gazebo` as the Docker entrypoint, which I believe was deprecated in a Ignition Edifice.

My solution was to change `Dockerfile.base` to use the `nvidia/opengl` base image. This required the [workaround](https://github.com/ignitionrobotics/ign-gazebo/blob/3c1a591ce110a985dbb45274f7f835d800080289/docker/Dockerfile.ignition#L36) from `Dockerfile.ignition`. to install tzdata without prompting the user for interaction while building the image.

I'm happy to provide this change in another PR (or the [existing one](https://github.com/ignitionrobotics/ign-gazebo/pull/1005)) . However, is the Docker file still a supported use case? It's not clear to me whether the nightly build Dockerfile is intended to be a headless, isolated server or one with X forwarding capabilities.

Summoning @russkel , @nkoenig , and @mjcarroll , much-appreciated original contributors to the Dockerfile who might have guidance on their intended use.

chapulina, 2021-09-01:
> that file uses ign-gazebo as the Docker entrypoint, which I believe was deprecated in Ignition Edifice.

Ah yes, that also needs updating :grimacing: 

> is the Docker file still a supported use case?

We've been doing a bad job at keeping that updated because we don't run any regular automated testing for it. Our Docker support is also currently split across a few different places, such as:

* https://github.com/ignitionrobotics/ign-gazebo/tree/ign-gazebo5/docker
* https://github.com/adlarkin/ign-rocker
* https://github.com/ignition-tooling/action-ignition-ci
* https://github.com/ignition-tooling/setup-ign-docker
* https://github.com/osrf/docker_images/pull/378

I'd love to have it all organized into a coherent set of Dockerfiles and build / run scripts that we can confidently point users to. We'll probably have different ones according to the use case (development of downstream projects, development of Ignition itself, CI for downstream projects, etc). @adlarkin took a look into this a while back and I believe the result was `ign-rocker`, but I think we never put the time into cross-referencing documentation and cleaning up the various approaches.

adlarkin, 2021-09-01:
I do agree that our Docker resources/documentation is scattered across various locations, which makes it hard for us to maintain and hard for others to use. The most recent development I am aware of is https://github.com/ignition-tooling/gzdev (more specifically, https://github.com/ignition-tooling/gzdev#ign-docker-env) which is meant to provide users with an ignition environment that is GUI/GPU friendly and also supports development of ignition packages fairly easily.

I think that we should consolidate all of our Docker resources into one place. [Dockerhub](https://hub.docker.com/) seems to be a good place to host images, and we can either add documentation for the images on Dockerhub itself, or use Dockerhub to link to documentation elsewhere. It looks like some of this is being addressed in https://github.com/osrf/docker_images/pull/378, but I'm not sure what the status is of that PR. I find https://github.com/ignition-tooling/gzdev#ign-docker-env to be very useful for day-to-day ignition development, but it requires things like https://github.com/osrf/rocker which users may or may not want to use.

If we don't have an issue open already that keeps track of cleaning up our Docker resources/support, it would probably be worth opening one so that we have a way to track what Docker support/infrastructure is lacking. This would also be visible to the community so that others are aware of the current status of Docker support for ignition.

chapulina, 2021-09-01:
> The most recent development I am aware of is https://github.com/ignition-tooling/gzdev

Ah see, I didn't even list that one

> Dockerhub seems to be a good place to host images

+1 to that

> It looks like some of this is being addressed in osrf/docker_images#378, but I'm not sure what the status is of that PR.

I believe that's probably lacking input from the core Ignition team. If one of us steps up to take it to the finish line, we should wrap it up, document it, and announce it. 

> If we don't have an issue open already that keeps track of cleaning up our Docker resources/support, it would probably be worth opening one 

@adlarkin , would you be able to open one in `release-tools`? You don't need to be the one to address it, but I think you have the most context to ticket the issue.

adlarkin, 2021-09-02:
> would you be able to open one in release-tools?

:+1: here's the issue I created: https://github.com/ignition-tooling/release-tools/issues/506

PR #1005 (2021-09-03): Update DART deps to Focal
# 🦟 Bug fix

Fixes #1004

## Summary
Within the Docker `install_ign_deps.sh` script, I updated DART dependencies from packages within the Bionic apt repositories to their Focal counterparts, at @chapulina 's suggestion from #1004 .

NOTE! : After making the changes, building `ign_gazebo:base` and `ign_gazebo:nightly` successfully completed. However, I have not been able to verify that the images function properly. I've been unsuccessfully debugging X forwarding to make `glxgears` or `ign gazebo shapes.sdf` work in the container with a GUI. 

I'm opening this PR becaus [... cut]
closed 2021-09-03 by chapulina