gazebosim/gz-sim #1011: Running `sensors_demo.sdf` causes PC to hang

Blast545, 2021-09-07:
## Environment
* OS Version: Ubuntu 18.04 
* Source build, Ignition dome

## Description
* Expected behavior:  Running `ign gazebo sensors_demo.sdf` runs the simulation until it's closed.
* Actual behavior: After about 3-4 mins my PC hangs.

I've checked the output coming from `ps  -aux` as a quick check to see that the memory percentage used by `ign gui` was growing without bounds. At some point my PC hangs because it runs out of memory and the SO kills the simulation.

## Steps to reproduce

Running `ign gazebo sensors_demo.sdf` with Ignition Dome installed from source.

I asked @gonzodepedro to verify that this error occurred and he had the same issue with his environment. I don't where it happens, but feel free to ping me to provide more information.

chapulina, 2021-09-07:
I just tried binaries and a source build on Bionic and couldn't reproduce this.

[image]

Do you see any messages before it hangs, if you run verbose? Does this only happen with Dome or also other versions?

Blast545, 2021-09-07:
@chapulina I only tried with Dome from source, I'll reinstall the whole environment just in case and let you know if still occurs.

Blast545, 2021-09-07:
I reinstalled Dome from source and still have the same problem x.x

darksylinc, 2021-09-19:
I took a shot out of curiosity (I'm on main branch) and I can see GUI memory kept climbing. There's definitely a leak.

`ign gazebo -s -r sensors_demo.sdf` is fine.

But `ign gazebo -g` keeps climbing.

AllianceDarkSylinc, 2021-09-19:
FOUND IT!

The [bug is in ign-gui](https://github.com/ignitionrobotics/ign-gui/blob/1cad501ad0c7da9f7811efa75ed3af80e5ee6a07/src/plugins/image_display/ImageDisplay.cc#L201)

The image returned by output.Data is never deleted:

```cpp
if (pixelFormat != common::Image::PixelFormatType::RGB_INT8)
{
unsigned int outputSize = 0;
unsigned char *data = nullptr;
output.Data(&data, outputSize);

for (unsigned int j = 0; j < height; ++j)
{
  for (unsigned int i = 0; i < width; ++i)
  {
	unsigned int idx = j*width*3 + i * 3;
	int r = data[idx];
	int g = data[idx + 1];
	int b = data[idx + 2];
	QRgb value = qRgb(r, g, b);
	image.setPixel(i, j, value);
  }
}

delete [] data; // <----------- THIS IS MISSING
}
```

After adding that delete, memory stopped climbing. I don't know if there are other (subtle) leaks, but this one was a massive one.

darksylinc, 2021-09-19:
FOUND IT!

The [bug is in ign-gui](https://github.com/ignitionrobotics/ign-gui/blob/1cad501ad0c7da9f7811efa75ed3af80e5ee6a07/src/plugins/image_display/ImageDisplay.cc#L201)

The image returned by output.Data is never deleted:

```cpp
if (pixelFormat != common::Image::PixelFormatType::RGB_INT8)
{
unsigned int outputSize = 0;
unsigned char *data = nullptr;
output.Data(&data, outputSize);

for (unsigned int j = 0; j < height; ++j)
{
  for (unsigned int i = 0; i < width; ++i)
  {
	unsigned int idx = j*width*3 + i * 3;
	int r = data[idx];
	int g = data[idx + 1];
	int b = data[idx + 2];
	QRgb value = qRgb(r, g, b);
	image.setPixel(i, j, value);
  }
}

delete [] data; // <----------- THIS IS MISSING
}
```

After adding that delete, memory stopped climbing. I don't know if there are other (subtle) leaks, but this one was a massive one.

chapulina, 2021-09-20:
> FOUND IT!

Awesome!