gazebosim/gz-sim #1019: Document the various ways of controlling a joint

chapulina, 2021-09-09:
Ignition Gazebo offers many out-of-box controllers with different features and intended use cases, for example:

* [JointController](https://ignitionrobotics.org/api/gazebo/5.1/classignition_1_1gazebo_1_1systems_1_1JointController.html)
* [JointPositionController](https://ignitionrobotics.org/api/gazebo/5.1/classignition_1_1gazebo_1_1systems_1_1JointPositionController.html)
* [JointTrajectoryController](https://ignitionrobotics.org/api/gazebo/5.1/classignition_1_1gazebo_1_1systems_1_1JointTrajectoryController.html)

It can be difficult for users to choose what to use. So it would be nice to have a high-level joint control tutorial explaining the features and usage of each of them, and why a user would choose one over the other.

## Desired behavior

We should have a new tutorial in https://ignitionrobotics.org/api/gazebo/5.1/tutorials.html which summarizes all ways of controlling a joint.

## Alternatives considered

Instead of documenting just joint controllers, we could document all ways of moving things, including links, models, etc. This reminds me of this great Gazebo classic tutorial: [Setting Velocity on Joints and Links](http://gazebosim.org/tutorials?tut=set_velocity&cat=)

## Implementation suggestion

Add a tutorial here: https://github.com/ignitionrobotics/ign-gazebo/tree/ign-gazebo5/tutorials

ishaanmane, 2021-09-23:
Hey @chapulina Can I please work on this issue

ahcorde, 2021-09-23:
Hi @ishaanmane

Nobody pick this issue yet! it's all yours! Happy to review your PRs.

ishaanmane, 2021-09-27:
Thank you working on it

LeePerry, 2023-09-25:
Hi. Is there any progress on this? I’m a new user to Gazebo looking for a joint control tutorial. (I can’t get any of the related demo worlds in Garden to work.) Thanks

yaswanth1701, 2023-11-19:
Hi  @ahcorde, I think I can documentation the various joint controllers  and also for the movement of link and models. So, should I contribute to this issue ?

ahcorde, 2023-11-20:
Hi @yaswanth1701, 

Feel free to contribute, there are two places that you can contribute with tutorials:
 - https://github.com/gazebosim/gz-sim/tree/gz-sim7/tutorials ( the joint controller docs may live here)
 - https://github.com/gazebosim/docs/tree/master/harmonic ( if yo uare thinking about porting this http://gazebosim.org/tutorials?tut=set_velocity&cat= too, the best place probrably is here)

yaswanth1701, 2023-11-20:
Oh, thanks a lot. I will start working on this.

azeey, 2023-12-05:
@yaswanth1701 thank you so much for taking this on! 

What you have so far is definitely in the right direction. I would say it would be great to build a full example of a robot where the joint is controlled with these modes and to include example commands and expected results (eg. a gif showing the robot move in a certain direction or robot arm going to a certain configuration). We should assume the audience to be beginners learning Gazebo so including just the `<plugin>` snippets might not  be sufficient to get them going.

Regarding the parameters, I think it would be better to link to https://gazebosim.org/api/sim/8/classgz_1_1sim_1_1systems_1_1JointPositionController.html than to list them in the tutorial.

yaswanth1701, 2023-12-05:
@azeey thanks for the review, Sure I will try make it more friendly towards beginners and will let you know on update soon.

strangerwhoisharborofdoom, 2026-03-04:
Following up: I see that PR #2263 was merged on Jan 9, 2024. The joint controller tutorial has been successfully added to gz-sim7 branch.

Since the main request from this issue (documenting joint controllers) has been addressed, this issue might be ready to close. However, if there's still value in documenting JointPositionController and JointTrajectoryController separately, those could be tracked in new issues.

Thanks @yaswanth1701 for completing this tutorial!

PR #2263 (2024-01-08): Added tutorial for Gazebo joint controller plugin

# 🎉 New feature

Closes #1019 

## Summary

Added a tutorial for Gazebo's joint controller plugin.

## Test it

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [x] Added example and/or tutorial
- [x] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://gazebosim.org/docs/all/contr [... cut]