gazebosim/gz-sim #1022: Modifying lights in component inspector not working

jennuine, 2021-09-10:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build?
  
Source, `main` branch, commit 1fb320715525a86361f2710ffa6aadb4c56a58c0

## Description
* Expected behavior: Lights to change color when using the component inspector
* Actual behavior: Most of the time nothing happens (sometimes will change color but only seems to be once). When changing one of the RGBA sliders then changing another, the first slider will reset back. For example, if I change R then B, R seems to reset back.

[image]

## Steps to reproduce

1. Apply fix from https://github.com/ignitionrobotics/ign-gazebo/pull/1018
2. `ign gazebo -v 4 lights.sdf`
3. Attempt to change the color of any of the lights

ahcorde, 2021-09-29:
I'm debugging this issue and something is weird here.

the flow is: The data is generated by the GUI -> `UserCommand` will receive the new light data -> `renderutil` should read this new data and modify the light

 - The data is fine in the two first steps. `UserCommand` receive the data properly. The first iteration when this method run, the `LightCmd` component is created, and `renderutil` is able to read this new data, in the `_ecm.Each<components::LightCmd>`  
 - but when this method run for the second time, the data remains with the data of the first iteration.
 - I'm removing the `LightCmd` component with `_ecm.RemoveComponent<components::LightCmd>`
 - I check that the state has `OneTimeChange` activated 

Why `lightCmd` does not change ?

I tried this with `GzScene3D` and `MinimalScene`, the result is the same in both cases.

ahcorde, 2021-09-29:
This issues is also extensible to `ParticleEmitterCMD`.  I tried `particle_emitter2.sdf` world.

I tried to change the rate with this command: 

```sh
ign topic -t /model/fog_generator/link/fog_link/particle_emitter/emitter/cmd -m ignition.msgs.ParticleEmitter -p 'rate: {data: 100}'
```

but the `particuleEmitterCMD` is setted to the first value when the CMD was created.

PR #1580 (2022-07-20): Fix component updates

# 🦟 Bug fix

Fixes #1022

## Summary

To speed up performance, removed components were cached/stored here: 
https://github.com/gazebosim/gz-sim/blob/1f6c1263fea66cd1736f420741acbe030a3f3089/include/ignition/gazebo/detail/View.hh#L140-L149

Then when the same component was re-added: https://github.com/gazebosim/gz-sim/blob/1f6c1263fea66cd1736f420741acbe030a3f3089/src/EntityComponentManager.cc#L1088-L1089

The component data was moved from `invalidData` to `validData` but the data was never updated.

The fix is to check that component needs updated in the `EntityComponentM [... cut]
closed 2022-07-20 by jennuine