gazebosim/gz-sim #1038: <specular> bug

Kakcalu13, 2021-09-20:
System information:
    - Ign Citadel on Ubuntu 20.04 Focal
    - Ros2 Foxy (Doubt this has to do with anything)

Hi there! I hope this doesn't make you want to choke me :P This time, I hope to bring you a valuable bug! 

I'm using the documentations:
[SDF format](http://sdformat.org/spec?ver=1.8&elem=material#material_pbr)
[Ign migrationsdf](https://ignitionrobotics.org/api/gazebo/5.1/migrationsdf.html) 

I see there's two type to use the texture which is `<specular>` and `<metal>`

I just looked into #328 which used png file on `<metal>`

I tried on `<specular>` and it instantly crash the whole gazebo. Let's use the example code below:
```
            <material>
              <pbr>
                <metal>
                  <albedo_map>floor.png</albedo_map>
                  <normal_map>floor.png</normal_map>
                </metal>
              </pbr>
            </material>
```

Which results as image here:
[image]

I was like Okay, this is probably on my end. So I wanted to try on `specular` and see if it will be any different.

I replaced metal to specular following this one:
[image]

which leads to this:
 ```
            <material>
              <pbr>
                <specular>
                  <albedo_map>floor.png</albedo_map>
                  <normal_map>floor.png</normal_map>
                </specular>
              </pbr>
            </material>
```
which results as this instant crash following with the bug:
[image]

I ran with the command: `ign gazebo -r models/sdf/diff_drive.sdf --verbose`

So I ran to get the full log textfile by run `ign gazebo -r models/sdf/diff_drive.sdf --verbose >> fulllog.txt`

which leads this

[fulllog.txt](https://github.com/ignitionrobotics/ign-gazebo/files/7198181/fulllog.txt)

Odd enough that it doesn't include the post error so I had to copy and paste in a different file from the output on terminal
[post-crash.txt](https://github.com/ignitionrobotics/ign-gazebo/files/7198184/post-crash.txt)

Is it expected or a bug?

P.S. 

_I know you are very extremely busy but If you happen to have some spare time to explain me how to import a png or any requirement png for it to import..Can I know? O:-) 
I'm willing to give you 30 dollars through paypal or cashapp lol_

iche033, 2021-10-08:
ah yes, I think we haven't tested `<specular>` much.. We should at least catch the crash and print out an warning.

Kakcalu13, 2021-10-08:
Gotcha!

I end up using something like this:
```
          <material>
            <ambient>0.8 0.8 0.8 1</ambient>
            <diffuse>0.8 0.8 0.8 1</diffuse>
            <pbr>
                <metal>
                  <albedo_map>floor.png</albedo_map>
                  <normal_map>floor.png</normal_map>
                </metal>
              </pbr>
          </material>
```

I just wanted to see what it looks like with the `<specular>`

Let me know if you wanted me to test something. I've been using Citadel almost all the time now :) Love it! Can't wait to try Fortress soon! 

Unrelated to this subject, do you, by any chance, have a friction doc? :angel:

iche033, 2021-10-08:
https://github.com/ignitionrobotics/ign-gazebo/pull/1099 should prevent the crash and try and render using the `<metal>` workflow.

The model you pasted should look very similar in `<specular>` once it's supported as it does not use any materials that are specific to any one particular workflow, e.g. `<metal>` has `<metalness_map>` and `<roughness_map>` while `<specular`> uses `<glossiness_map>` and `<specular_map>`.

> Unrelated to this subject, do you, by any chance, have a friction doc?

There is a nice write-up here by one of our users:
https://www.allisonthackston.com/articles/ignition-vs-gazebo.html

If you just need some info on what the parameters mean, there is also a doc on physics parameters in gazebo-classic. Not everything has been ported to ignition yet though.
https://gazebosim.org/tutorials?tut=physics_params&cat=physics

Kakcalu13, 2021-10-08:
Thank you so much!
Also I'm closing this since #1099 should fix this issue!

closed 2021-10-08 by Kakcalu13