gazebosim/gz-sim #1049: Default gui elements cause a crash in fortress binary installation

tfoote, 2021-09-24:
I was testing ignition fortress prerelease from debian packages on bionic. 

Full reproduction:
All worlds with the default gui elements settings crash on rendering. For example shapes.sdf

Copying a gui element from one of the other files prevents the crashes. 

Example tracebacks: https://gist.github.com/tfoote/b8237191b0b6d6811b656460c208179f

Working on shapes.sdf we commented everything out and it still failed.
Iterating with @iche033 we got it working by copying the `<gui>` tag from other example sdf files.

```

<?xml version="1.0" ?>
<!--

Try moving a model:

    ign service -s /world/shapes/set_pose --reqtype ignition.msgs.Pose --reptype ignition.msgs.Boolean --timeout 300 --req 'name: "box", position: {z: 5.0}'

-->
<sdf version="1.6">
  <world name="shapes">
    <scene>
      <ambient>1.0 1.0 1.0</ambient>
      <background>0.8 0.8 0.8</background>
    </scene>
    <gui fullscreen="0">

      <!-- 3D scene -->
      <plugin filename="GzScene3D" name="3D View">
        <ignition-gui>
          <title>3D View</title>
          <property type="bool" key="showTitleBar">false</property>
          <property type="string" key="state">docked</property>
        </ignition-gui>

        <engine>ogre2</engine>
        <scene>scene</scene>
        <ambient_light>0.4 0.4 0.4</ambient_light>
        <background_color>0.8 0.8 0.8</background_color>
        <camera_pose>-6 0 6 0 0.5 0</camera_pose>
      </plugin>

      <!-- World control -->
      <plugin filename="WorldControl" name="World control">
        <ignition-gui>
          <title>World control</title>
          <property type="bool" key="showTitleBar">false</property>
          <property type="bool" key="resizable">false</property>
          <property type="double" key="height">72</property>
          <property type="double" key="width">121</property>
          <property type="double" key="z">1</property>

          <property type="string" key="state">floating</property>
          <anchors target="3D View">
            <line own="left" target="left"/>
            <line own="bottom" target="bottom"/>
          </anchors>
        </ignition-gui>

        <play_pause>true</play_pause>
        <step>true</step>
        <start_paused>true</start_paused>
        <service>/world/velocity_control/control</service>
        <stats_topic>/world/velocity_control/stats</stats_topic>

      </plugin>

[... 248 lines cut ...]
            <ambient>1 0 1 1</ambient>
            <diffuse>1 0 1 1</diffuse>
            <specular>1 0 1 1</specular>
          </material>
        </visual>
      </link>
    </model>
-->
    </world>
</sdf>

```

This is running nvidia and validated as running velocity_control.sdf and diff_drive.sdf and sensors_demo.sdf.

## Environment
* OS Version: Ubunti Bionic inside a container
* Source or binary build?
  fortress prerelease debians
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: both ogre and ogre2]
      - [ ] Sensor rendering error. 
      - [x] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode

chapulina, 2021-09-24:
Thanks for the ticket. I just tried a fresh `rocker` Bionic container with the same commands as you and things are working for me.

I think this may be related to the NVIDIA driver on the host. I'm on a Focal host with:

```
NVIDIA-SMI 470.63.01    Driver Version: 470.63.01    CUDA Version: 11.4 
```

dignakov, 2021-09-28:
Please let me know if there are any logs or info I can provide as well as I'm seeing the same GL3.3 error and segfault on an intel GPU.