gazebosim/gz-sim #1056: Warning for some example world's models

matosinho, 2021-09-25:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build?
  * Source
  * main (Fortress)
  * commit: 653b0fb58abd6ea8b3e3937e43d5aee509e125fc
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: default.
      - [ ] Sensor rendering error.
      - [x] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [x] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [ ] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
      0a:00.0 VGA compatible controller [0300]: NVIDIA Corporation TU106 [GeForce RTX 2060 SUPER] [10de:1f06] (rev a1)
      direct rendering: Yes
      OpenGL vendor string: NVIDIA Corporation
      OpenGL renderer string: GeForce RTX 2060 SUPER/PCIe/SSE2
      OpenGL core profile version string: 4.6.0 NVIDIA 460.91.03
      OpenGL core profile shading language version string: 4.60 NVIDIA
      OpenGL core profile context flags: (none)
      OpenGL core profile profile mask: core profile
      OpenGL version string: 4.6.0 NVIDIA 460.91.03
      OpenGL shading language version string: 4.60 NVIDIA
      OpenGL context flags: (none)
      OpenGL profile mask: (none)
      OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 460.91.03
      OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

<details>  

```
19:29:01: Creating resource group General
19:29:01: Creating resource group Internal
19:29:01: Creating resource group Autodetect
19:29:01: SceneManagerFactory for type 'DefaultSceneManager' registered.
19:29:01: Registering ResourceManager for type Material
19:29:01: Registering ResourceManager for type Mesh
19:29:01: Registering ResourceManager for type Skeleton
19:29:01: MovableObjectFactory for type 'ParticleSystem' registered.
19:29:01: ArchiveFactory for archive type FileSystem registered.
19:29:01: ArchiveFactory for archive type Zip registered.
19:29:01: ArchiveFactory for archive type EmbeddedZip registered.
19:29:01: DDS codec registering
19:29:01: FreeImage version: 3.18.0
19:29:01: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
19:29:01: Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,psb,cut,xbm,xpm,gif,hdr,g3,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
19:29:01: PVRTC codec registering
19:29:01: ETC codec registering
19:29:01: Registering ResourceManager for type HighLevelGpuProgram
19:29:01: Registering ResourceManager for type Compositor
19:29:01: MovableObjectFactory for type 'Entity' registered.
19:29:01: MovableObjectFactory for type 'Light' registered.
19:29:01: MovableObjectFactory for type 'BillboardSet' registered.
[... 263 lines cut ...]
```bash
Warning [Model.cc:170] Non-unique names detected in XML children of model with name[lift_drag_demo_model].
Warning [Model.cc:170] Non-unique names detected in XML children of model with name[lift_drag_demo_model].
Warning [Model.cc:170] Non-unique names detected in XML children of model with name[lift_drag_demo_model].
Warning [Model.cc:170] Non-unique names detected in XML children of model with name[lift_drag_demo_model].
```
`multicopter_velocity_control.sdf` has these warning upon loading:
```bash
Warning [Utils.cc:129] [/sdf/model[@name="x3"]/link[@name="X3/base_link"]/inertial/pose[@relative_to=""]:/home/matosinho/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/x3 uav/4/model.sdf:L7]: XML Attribute[relative_to] in element[pose] not defined in SDF.
Warning [Utils.cc:129] [/sdf/model[@name="x3"]/link[@name="X3/rotor_0"]/inertial/pose[@relative_to=""]:/home/matosinho/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/x3 uav/4/model.sdf:L39]: XML Attribute[relative_to] in element[pose] not defined in SDF.
Warning [Utils.cc:129] [/sdf/model[@name="x3"]/link[@name="X3/rotor_1"]/inertial/pose[@relative_to=""]:/home/matosinho/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/x3 uav/4/model.sdf:L104]: XML Attribute[relative_to] in element[pose] not defined in SDF.
Warning [Utils.cc:129] [/sdf/model[@name="x3"]/link[@name="X3/rotor_2"]/inertial/pose[@relative_to=""]:/home/matosinho/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/x3 uav/4/model.sdf:L169]: XML Attribute[relative_to] in element[pose] not defined in SDF.
Warning [Utils.cc:129] [/sdf/model[@name="x3"]/link[@name="X3/rotor_3"]/inertial/pose[@relative_to=""]:/home/matosinho/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/x3 uav/4/model.sdf:L234]: XML Attribute[relative_to] in element[pose] not defined in SDF.
Warning [Utils.cc:129] [/sdf/model[@name="X4"]/link[@name="base_link"]/inertial/pose[@relative_to=""]:/home/matosinho/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/x4 uav config 1/12/model.sdf:L8]: XML Attribute[relative_to] in element[pose] not defined in SDF.
Warning [Utils.cc:129] [/sdf/model[@name="X4"]/link[@name="rotor_0"]/inertial/pose[@relative_to=""]:/home/matosinho/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/x4 uav config 1/12/model.sdf:L336]: XML Attribute[relative_to] in element[pose] not defined in SDF.
Warning [Utils.cc:129] [/sdf/model[@name="X4"]/link[@name="rotor_1"]/inertial/pose[@relative_to=""]:/home/matosinho/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/x4 uav config 1/12/model.sdf:L403]: XML Attribute[relative_to] in element[pose] not defined in SDF.
Warning [Utils.cc:129] [/sdf/model[@name="X4"]/link[@name="rotor_2"]/inertial/pose[@relative_to=""]:/home/matosinho/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/x4 uav config 1/12/model.sdf:L470]: XML Attribute[relative_to] in element[pose] not defined in SDF.
Warning [Utils.cc:129] [/sdf/model[@name="X4"]/link[@name="rotor_3"]/inertial/pose[@relative_to=""]:/home/matosinho/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/x4 uav config 1/12/model.sdf:L537]: XML Attribute[relative_to] in element[pose] not defined in SDF.
Warning [Utils.cc:129] [/sdf/model[@name="X4"]/link[@name="rotor_4"]/inertial/pose[@relative_to=""]:/home/matosinho/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/x4 uav config 1/12/model.sdf:L604]: XML Attribute[relative_to] in element[pose] not defined in SDF.
Warning [Utils.cc:129] [/sdf/model[@name="X4"]/link[@name="rotor_5"]/inertial/pose[@relative_to=""]:/home/matosinho/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/x4 uav config 1/12/model.sdf:L671]: XML Attribute[relative_to] in element[pose] not defined in SDF.
Warning [Model.cc:170] Non-unique names detected in XML children of model with name[x3].
Warning [Model.cc:170] Non-unique names detected in XML children of model with name[X4].
Warning [Model.cc:170] Non-unique names detected in XML children of model with name[X4].
Warning [Model.cc:170] Non-unique names detected in XML children of model with name[x3].
```

scpeters, 2021-09-29:
> `track_drive.sdf` has these warning upon loading:
> 
> ```shell
> Warning [Utils.cc:129] [/sdf/world[@name="track_drive"]/model[@name="vehicle_blue"]/link[@name="base_link"]/inertial/pose[@relative_to=""]:/home/matosinho/dev/ROS2/fortress_party/install/share/ignition/ignition-gazebo6/worlds/track_drive.sdf:L88]: XML Attribute[relative_to] in element[pose] not defined in SDF.
> Warning [Utils.cc:129] [/sdf/world[@name="track_drive"]/model[@name="vehicle_green"]/link[@name="base_link"]/inertial/pose[@relative_to=""]:/home/matosinho/dev/ROS2/fortress_party/install/share/ignition/ignition-gazebo6/worlds/track_drive.sdf:L1082]: XML Attribute[relative_to] in element[pose] not defined in SDF.
> ```

these warnings about `//inertial/pose[@relative_to=""]` should be fixed by https://github.com/ignitionrobotics/sdformat/pull/720

matosinho, 2021-09-30:
I dunno if I did something wrong, but I have changed to the sdf11, pulled your changes, ran `colcon build --merge-install` again and source again the `setup.bash`. Unfortunatelly, the warning is still apearing to me.

scpeters, 2021-09-30:
If you have a source workspace and debian packages installed to `/usr`, it can be tricky to ensure that `ign` commands are using the intended libraries. If you have `ign-tools` in your workspace, I think it should find the right libraries. Otherwise, you can set `IGN_CONFIG_PATH` to the `install/share/ignition` folder of your colcon workspace.

Also, `sdf11` works with `ign-gazebo5`. I think you will need to use the `main` branch of sdformat to work with ign-gazebo6