gazebosim/gz-sim #1057: Make joint state publisher's topic configurable

chapulina, 2021-09-27:
## Desired behavior

The [JointStatePublisher](https://ignitionrobotics.org/api/gazebo/6.0/classignition_1_1gazebo_1_1systems_1_1JointStatePublisher.html#details) plugin always publishes on the `/world/<world_name>/model/<model_name>/state` topic. It would be convenient to let users change it, especially so that models can be more easily reused across worlds of different names.

## Alternatives considered

N/A

## Implementation suggestion

The topic should be configurable from SDF.

The same way that we accept a `joint_name` element here:

https://github.com/ignitionrobotics/ign-gazebo/blob/59ff8532d0699f18cf9d0e34aa1bc0afecb73671/src/systems/joint_state_publisher/JointStatePublisher.cc#L61

We could accept a `topic` element, like how this other plugin does it for example:

https://github.com/ignitionrobotics/ign-gazebo/blob/59ff8532d0699f18cf9d0e34aa1bc0afecb73671/src/systems/detachable_joint/DetachableJoint.cc#L99-L106

jasmeet0915, 2021-10-01:
Hi, I was looking for first time issues and have been able to resolve this one. I would like to make a PR for the same!

chapulina, 2021-10-01:
That's great, thanks @jasmeet0915 !

PR #1076 (2021-11-15): Configurable joint state publisher's topic

# 🎉 New feature

Closes #1057 

## Summary

Makes the topic for JointStatePublisher configurable

#### Expected behavior: 
A user should be able to provide a topic for JointStatePublisher using `<topic>` in the sdf

[image]

## Test it

 * Choose an example world (I tested with [diff_drive.sdf](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo5/examples/worlds/diff_drive.sdf)) and add JointStatePublisher to it while also specifying a topic with `<topic>` tags.
 * Then check if this topic gets listed with `ign topic -l` and you are able t [... cut]
closed 2022-03-02 by chapulina