gazebosim/gz-sim #1066: fortress ign gazebo crashes after launch (OpenGL version?)

dignakov, 2021-09-28:
After installing fortress, when I try to run shapes.sdf the program opens a GUI which doesn't show anything and crashes. There is an error saying that:

```
[GUI] [Err] [Ogre2RenderEngine.cc:894]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /home/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3364)
```

But the OpenGL core profile version is 4.5 . . .

## Environment
* OS Version: Ubuntu 20.04 (clean install)
* Source or binary build?: binary - `Ignition Gazebo, version 6.0.0~pre1`
*  GPU: `Mesa DRI Intel(R) HD Graphics 4600 (HSW GT2)`
* OpenGL: 
```
$ glxinfo | grep version
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
    Max core profile version: 4.5
    Max compat profile version: 3.0
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.1
OpenGL core profile version string: 4.5 (Core Profile) Mesa 21.0.3
OpenGL core profile shading language version string: 4.50
OpenGL version string: 3.0 Mesa 21.0.3
OpenGL shading language version string: 1.30
OpenGL ES profile version string: OpenGL ES 3.1 Mesa 21.0.3
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10
    GL_EXT_shader_implicit_conversions, GL_EXT_shader_integer_mix, 
```

 - Rendering plugin: ogre2
 - [no?] Sensor rendering error. 
 - [yes] GUI rendering error. 
 - EGL headless mode:
 - [no] Running in EGL headless mode 
 - Generally, mention all circumstances that might affect rendering capabilities: 
      - [no] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [no] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [yes] running on real hardware
      - [no] running in virtual machine
      - [no] running in Docker/Singularity
      - [no] running remotely (e.g. via SSH)
      - [no] running in a cloud
      - [no] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [no] GPU is concurrently used for other tasks
        - [whatever is default in 20.04] desktop acceleration
        - [no] video decoding (i.e. a playing Youtube video)
        - [no] video encoding
        - [no] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [no] multiple simulators running at the same time
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
        LANG=C lspci -nn | grep VGA: `00:02.0 VGA compatible controller [0300]: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor Integrated Graphics Controller [8086:0412] (rev 06)`
          echo "$DISPLAY": `:0`
          `LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package`:
```
direct rendering: Yes
    Preferred profile: core (0x1)
    Max core profile version: 4.5
    Max compat profile version: 3.0
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.1
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) HD Graphics 4600 (HSW GT2)
OpenGL core profile version string: 4.5 (Core Profile) Mesa 21.0.3
OpenGL core profile shading language version string: 4.50
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 3.0 Mesa 21.0.3
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL ES profile version string: OpenGL ES 3.1 Mesa 21.0.3
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10

```

ps aux | grep Xorg: 
```
dmitri       819  0.7  0.9 804368 76764 tty2     Sl+  09:06   1:27 /usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1000/gdm/Xauthority -background none -noreset -keeptty -verbose 3
dmitri     19667  0.0  0.0  17672   724 pts/4    S+   12:25   0:00 grep --color=auto Xorg
```
sudo env LANG=C X -version:
```
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
Build Operating System: linux Ubuntu
Current Operating System: Linux toaster 5.11.0-36-generic #40~20.04.1-Ubuntu SMP Sat Sep 18 02:14:19 UTC 2021 x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.11.0-36-generic root=UUID=12d954bc-1690-4505-b836-85dc6e8ab314 ro quiet splash vt.handoff=7
Build Date: 06 July 2021  10:17:51AM
xorg-server 2:1.20.11-1ubuntu1~20.04.2 (For technical support please see http://www.ubuntu.com/support) 
Current version of pixman: 0.38.4
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.

```
          
   
-  Please, attach the ogre.log or ogre2.log file from  `~/.ignition/rendering` : `https://gist.github.com/dignakov/70de2bcb4d13bb5a020a8198baa2a981`
    
- Details
  
```
https://gist.github.com/dignakov/ad42d8c3c563051fba7ea3ea448d4136
```

## Description
* Expected behavior: GUI starts and shows the scene.
* Actual behavior:  GUI starts with almost nothing shown and crashes

## Steps to reproduce

1. Clean install ubuntu 20.04
2. follow installation isntructions
3. run `ign gazebo shapes.sdf -v 4`

## Output

chapulina, 2021-09-28:
Looks like a possible duplicate of #1049. I'll close this, but please reopen if it turns out to be a different issue.

dignakov, 2021-09-28:
@chapulina  I think the crash might be happening at the same place.

```
[GUI] [Err] [Ogre2RenderEngine.cc:830] Failed to create dummy render window.
/usr/lib/ruby/ignition/cmdgazebo6.rb:438: [BUG] Segmentation fault at 0x0000000000000220
```
but I'm not sure it's the same cause . . . since the other console output has

```
libGL error: MESA-LOADER: failed to retrieve device information
libGL error: Version 4 or later of flush extension not found
libGL error: failed to load driver: i915
libGL error: failed to open /dev/dri/card0: No such file or directory
libGL error: failed to load driver: iris
```

but is being run as an nvidia runtime `rocker --pull --user --nvidia --x11 ubuntu:bionic` . . . So both can't create a window (and both get a gl 3.3 error) but the other issue might have a few more things going on so I don't want to take over that discussion, so if it ends up diverging please redirect me back here and I'll stop commenting there.

0smile, 2021-10-10:
I came to the same issue. I'm using the WSLg under Windows11. It works perfectly with citadel, but for fortress, it crashed with the same opengl verision problem.

chapulina, 2021-10-11:
Ok, reopening because it seems to be a separate issue

addy1997, 2022-04-06:
Using this command ` ign gazebo shapes.sdf --render-engine ogre` I was able to launch it without any issues.

[image]

This [link](https://answers.gazebosim.org/question/27597/ignition-crashes-directly-after-start/) might be useful

sk3p71c, 2022-05-13:
I had the same issue after installing Fortress on a Ros2 "foxy" docker image. You can change the engine to ogre more permanently by modifying the config file at ~/.ignition/gazebo/6/gui.config (replace ogre2 with ogre). Unfortunately this config file doesn't appear to exist until you've run gazebo once. Also see https://github.com/conda-forge/libignition-gazebo-feedstock/issues/27

martincerven, 2024-07-30:
This also happens on Harmonic & Jazzy inside VM, possible due to lower OpenGL version, I additionally had to use [qt var](https://github.com/ros2/rviz/issues/1245#issuecomment-2241915391):
```
export QT_QPA_PLATFORM=xcb
gz sim --render-engine ogre
```
Otherwise I'd get segfault.

iameijaz, 2025-03-16:
Also happens with Raspberry Pi 5, Ubuntu 24.04, ROS 2 Jazzy, Gazebo Harmonic, but adding this  `--render-engine ogre` helps.