gazebosim/gz-sim #1084: Fortress GUI hangs on close

nkoenig, 2021-10-05:
## Environment
* OS Version: Ubuntu 18.04
* Source build

## Description

The GUI sometimes hangs when closing.

## Steps to reproduce

1. `ign gazebo -v 4 shapes.sdf`
2. Translate the user camera a bit
3. Close the GUI.
4. Hang.

nkoenig, 2021-10-05:
The last message I see is

```
[GUI] [Dbg] [MinimalScene.cc:570] Destroy scene [scene]
```

nkoenig, 2021-10-06:
Here is a backtrace:

```
0x00007fe4f4176529 in futex_wait (private=<optimized out>, expected=12, 
    futex_word=0x55b2a7943844)
    at ../sysdeps/unix/sysv/linux/futex-internal.h:61
61	../sysdeps/unix/sysv/linux/futex-internal.h: No such file or directory.
(gdb) bt
#0  0x00007fe4f4176529 in futex_wait (private=<optimized out>, expected=12, futex_word=0x55b2a7943844) at ../sysdeps/unix/sysv/linux/futex-internal.h:61
#1  0x00007fe4f4176529 in futex_wait_simple (private=<optimized out>, expected=12, futex_word=0x55b2a7943844) at ../sysdeps/nptl/futex-internal.h:135
#2  0x00007fe4f4176529 in __pthread_cond_destroy (cond=0x55b2a7943820)
    at pthread_cond_destroy.c:54
#3  0x00007fe4b4ce8f32 in ignition::gui::plugins::RenderSync::~RenderSync() (this=0x55b2a79437f8, __in_chrg=<optimized out>)
    at /home/nkoenig/work/ignition/fortress/ign-gui/src/plugins/minimal_scene/MinimalScene.cc:181
#4  0x00007fe4b4ce8f32 in ignition::gui::plugins::RenderWindowItem::Implementation::~Implementation() (this=0x55b2a79437e0, __in_chrg=<optimized out>)
    at /home/nkoenig/work/ignition/fortress/ign-gui/src/plugins/minimal_scene/MinimalScene.cc:181
```

chapulina, 2021-10-06:
Should be fixed in https://github.com/ignitionrobotics/ign-gui/pull/300

closed 2021-10-06 by chapulina