gazebosim/gz-sim #1085: Fortress orbit camera is jumpy

nkoenig, 2021-10-05:
## Environment
* OS Version: Ubuntu 18.04
* Source

## Description

The orbit camera can jump dramatically, which is most noticeable when orbiting, but also occurs when translating.

## Steps to reproduce

1. Load any world.
2. Navigate around using the orbit camera.
3.The camera will occasionally jump instead of orbiting smoothly.

nkoenig, 2021-10-05:
It also seems that orbit camera sometimes doesn't orbit or translate about the location under the mouse. I assume these issues are related.

chapulina, 2021-10-05:
I'm working on it, the fix will go in `ign-gui`. It seems like we're losing some mouse events, so I'm implementing an event queue instead of only publishing the latest event available.

chapulina, 2021-12-13:
I believe this was fixed in https://github.com/ignitionrobotics/ign-gui/pull/295

closed 2021-12-13 by chapulina

iche033, 2022-04-26:
hmm I still experience this issue in Fortress with `MinimalScene` and other GUI plugins from the default `gui.config`. Not sure if it's affecting anyone else.