gazebosim/gz-sim #1114: Minimal Scene crash on Fortress

Kakcalu13, 2021-10-15:
## Environment
* OS Version: Ubuntu 20.04
* Binary build: Fortress 6.0.0 
* Local only (no docker available, or is it?)
* ROS2: Foxy only
* Full SDF file: [freenove_smart_car](https://github.com/feagi/feagi-core/blob/feature-feature-doc-ign-foxy-draft-part3/third_party/gazebo/smart_car/freenove_4wd_car_description/models/sdf/freenove_smart_car.sdf)

## Rendering system info:
 On Linux, provide the outputs of the following commands:
 ```
00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 530 [8086:191b] (rev 06)
bwuk@robots:~/feagi-core/src$           echo "$DISPLAY"
:0
bwuk@robots:~/feagi-core/src$           LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
direct rendering: Yes
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL vendor string: Intel
OpenGL renderer string: Mesa Intel(R) HD Graphics 530 (SKL GT2)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 21.0.3
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.6 (Compatibility Profile) Mesa 21.0.3
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 21.0.3
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
bwuk@robots:~/feagi-core/src$           ps aux | grep Xorg
bwuk         974  0.9  0.7 1047068 61344 tty2    Sl+  Oct11  56:02 /usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1000/gdm/Xauthority -background none -noreset -keeptty -verbose 3
bwuk     3052280  0.0  0.0  18292   664 pts/6    S+   11:44   0:00 grep --color=auto Xorg
```

</details>

## Minimal Scene crash
[image]
```
[GUI] [Err] [RenderEngineManager.cc:470] Failed to load plugin [] : couldn't load library on path [/usr/lib/x86_64-linux-gnu/].
[GUI] [Err] [MinimalScene.cc:518] Engine [] is not supported
[GUI] [Err] [MinimalScene.cc:657] Unable to initialize renderer
Error while loading the library [/usr/lib/x86_64-linux-gnu/]: /usr/lib/x86_64-linux-gnu/: cannot read file data: Is a directory
[GUI] [Err] [RenderEngineManager.cc:470] Failed to load plugin [] : couldn't load library on path [/usr/lib/x86_64-linux-gnu/].
[GUI] [Err] [MinimalScene.cc:518] Engine [] is not supported
[GUI] [Err] [MinimalScene.cc:657] Unable to initialize renderer
```
[image]

chapulina, 2021-11-12:
> couldn't load library on path [/usr/lib/x86_64-linux-gnu/].

This is very strange, it seems that it's looking for the rendering engine in the wrong path. Are you able to load with Ogre 1? You can edit the SDF file, or do `--render-engine ogre`

Kakcalu13, 2022-03-24:
My deeply apology for so late reply. I found this while searching for similar issue (I found one good issue to help me resolve on my end :P )

I just ran this just now.
```
Warning [GenericJoint.hpp:1480] [GenericJoint::setRestPosition] Value of _q0 [0], is out of the limit range [0.1, 3.1] for index [0] of Joint [Joint].
Error while loading the library [/usr/lib/x86_64-linux-gnu/]: /usr/lib/x86_64-linux-gnu/: cannot read file data: Is a directory
[GUI] [Err] [RenderEngineManager.cc:470] Failed to load plugin [] : couldn't load library on path [/usr/lib/x86_64-linux-gnu/].
[GUI] [Err] [MinimalScene.cc:525] Engine [] is not supported
[GUI] [Err] [MinimalScene.cc:666] Unable to initialize renderer
Error while loading the library [/usr/lib/x86_64-linux-gnu/]: /usr/lib/x86_64-linux-gnu/: cannot read file data: Is a directory
[GUI] [Err] [RenderEngineManager.cc:470] Failed to load plugin [] : couldn't load library on path [/usr/lib/x86_64-linux-gnu/].
[GUI] [Err] [MinimalScene.cc:525] Engine [] is not supported
[GUI] [Err] [MinimalScene.cc:666] Unable to initialize renderer

```

This is all it gives me. It simply froze my Fortress like this:
[image]

My version of Fortress is 6.3.0.

But then when I ran `--render-engine ogre`
[image]

This worked.

When I run it on my file, it crashed immediately.

I'm convinced it's all on my end. I just need to load the plugin since there is no plugin for it on my file. I'm closing this issue.

Sorry! I was kinda newbie at that time. :sweat_smile:

closed 2022-03-24 by Kakcalu13