gazebosim/gz-sim #1152: unversioned symlinks to systems are broken in build directory

peci1, 2021-11-01:
## Environment
* OS Version: Ubuntu 18.04
* Source or binary build? Source build ign-gazebo3, ign-gazebo4

## Description

https://github.com/ignitionrobotics/ign-gazebo/blob/1f44a0548d90abec21d3c9e9c59a4a84e12f54dc/src/systems/CMakeLists.txt#L69

I think this line generates broken symlinks for me. What it generates are symlinks in `CMAKE_BINARY_DIR` pointing to the versioned files in the same directory. However, the versioned files are located in `CMAKE_BINARY_DIR/lib`. Therefore, the unversioned symlinks are broken.

This is a problem in tests which need the plugin search mechanism to work, e.g. 

https://github.com/ignitionrobotics/ign-gazebo/blob/1f44a0548d90abec21d3c9e9c59a4a84e12f54dc/test/helpers/EnvTestFixture.hh#L37-L38

I observed this behavior with both "UNIX Makefiles" and Ninja CMake generators.

This is the file structure I have in the build directory:

```
$ ll libignition-gazebo*-physics-system.so*
lrwxrwxrwx 1 peci1 peci1 37 lis  1 14:30 libignition-gazebo-physics-system.so -> libignition-gazebo3-physics-system.so
$ ll lib/libignition-gazebo*-physics-system.so*
lrwxrwxrwx 1 peci1 peci1       39 lis  1 14:59 lib/libignition-gazebo3-physics-system.so -> libignition-gazebo3-physics-system.so.3*
lrwxrwxrwx 1 peci1 peci1       44 lis  1 14:59 lib/libignition-gazebo3-physics-system.so.3 -> libignition-gazebo3-physics-system.so.3.11.0*
-rwxrwxr-x 1 peci1 peci1 43980584 lis  1 14:59 lib/libignition-gazebo3-physics-system.so.3.11.0*
```

Can anyone please confirm this behavior?

---

I also tried to fix this and it seems the Windows approach from ign-gazebo6 works when adapted to use `create_symlink` instead of `copy` and `"$<TARGET_FILE_DIR:${system_target}>/${versioned}"` as the first argument.

https://github.com/ignitionrobotics/ign-gazebo/blob/35ec2a75f6d0522bb840e0b704768864cfeb4ec4/src/systems/CMakeLists.txt#L78-L81

Would that be a viable solution?

chapulina, 2021-11-01:
> Can anyone please confirm this behavior?

Yup, I also see broken symlinks in the build directory. This was never a use case we cared about, we've only been concerned that the symlinks work after installed.

> This is a problem in tests which need the plugin search mechanism to work

We've been solving this on CI by using the installed libraries instead of the ones in the build folder, which is not ideal...

> Would that be a viable solution?

The Windows solution is a workaround because we didn't get symlinks to work there. Ideally we'd keep using symlinks to avoid duplicated files. If we want the build folder to work, I'd prefer we fix the build folder's structure.

peci1, 2021-11-01:
> > Would that be a viable solution?
> 
> The Windows solution is a workaround because we didn't get symlinks to work there. Ideally we'd keep using symlinks to avoid duplicated files. If we want the build folder to work, I'd prefer we fix the build folder's structure.

Maybe I did not explain this well. The fix I suggest is keeping symlinks, but generates them in the custom command as on windows. Proposed solution for non-windows would therefore be:

```CMake
ADD_CUSTOM_COMMAND(TARGET ${system_target} POST_BUILD 
   COMMAND "${CMAKE_COMMAND}" -E create_symlink 
     "$<TARGET_FILE_DIR:${system_target}>/${versioned}"
     "$<TARGET_FILE_DIR:${system_target}>/${unversioned}") 
```

The only problem I noticed is it generates absolute path symlink, which would break relocatibility. But I think the build folder is not relocatable at all, so it's probably not an issue. I'm not sure, though, what happens with the symlinks after installation.

Maybe this would work, too?

```CMake
ADD_CUSTOM_COMMAND(TARGET ${system_target} POST_BUILD 
   COMMAND "${CMAKE_COMMAND}" -E create_symlink 
     "${versioned}"
     "$<TARGET_FILE_DIR:${system_target}>/${unversioned}") 
```

Yet another solution could be to keep the current `EXECUTE_PROCESS` and add `WORKING_DIRECTORY` parameter to force it to put the symlink in the `lib` subdir.

chapulina, 2021-11-01:
> The fix I suggest is keeping symlinks, but generates them in the custom command

Ahh got it

> another solution could be to keep the current EXECUTE_PROCESS and add WORKING_DIRECTORY parameter to force it to put the symlink in the lib subdir

Yeah I think that's promising

peci1, 2021-11-02:
Ok, so the easiest way works (adding `WORKING_DIRECTORY`). Fix in #1160.

chapulina, 2021-11-03:
#1160

PR #1160 (2021-11-03): Fix generation of systems library symlinks in build directory
# 🦟 Bug fix

Fixes #1152

## Summary
As discussed in #1152, the unversioned library symlinks for systems are generated into the root of the build directory, but tests expect them in the `lib` subdirectory of the build directory. This PR fixes the location.

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [x] `codecheck` passed (See [contributing](https://ignitionrobotics.org/docs/all/contributing#contributing-code))
- [x] All tests pa [... cut]
closed 2021-11-03 by chapulina