gazebosim/gz-sim #1154: Increase/Decrease viewport clipping distance

ColeOSRF, 2021-11-01:
## Desired behavior
Right now the viewport clipping distance is locked to 1000 meters with no way to adjust it. We'd like to be able to Increase and decrease the viewport clipping distance in Ignition GUI.

chapulina, 2021-11-01:
+1, this is part of https://github.com/ignitionrobotics/ign-gazebo/issues/421

iche033, 2021-11-02:
we can probably either put this in the [View Angle](https://github.com/ignitionrobotics/ign-gazebo/tree/ign-gazebo6/src/gui/plugins/view_angle) plugin or make a new gui plugin.

If we're to do it in the View Angle plugin:
1. Add a spinbox (`IgnSpnBox`) to configure the camera's far clip distance in [ViewAngle.qml](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo6/src/gui/plugins/view_angle/ViewAngle.qml)
2. In spinbox's `onEditingFinished` handler in QML, invoke a function in the ViewAngle.cc to set a variable containing the camera's far clip distance value.
3. The actual call to set the camera's far clip distance needs to be done in the rendering thread ([OnRender()](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo6/src/gui/plugins/view_angle/ViewAngle.cc#L287)). The View Angle gui plugin already has a pointer to the rendering camera, so it all comes down to the [Camera::SetFarClipPlane](https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering6/include/ignition/rendering/Camera.hh#L125) call. 

similar steps if it needs to be done in a separate gui plugin.

PR #1162 (2021-11-04): Set camera clipping plane distance
# 🎉 New feature

Related to https://github.com/ignitionrobotics/ign-gui/issues/308
Closes #1154 

## Summary
Set the user camera's near/far clipping plane distance through the `View Angle` plugin. 

## Test it
1. `ign gazebo minimal_shapes.sdf`
2. Expand `View Angle` plugin
3. Enter near/far values in spin boxes

[image]

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Added example and/or tutorial
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [x] `codecheck` passed (See [contributing](https://igniti [... cut]
closed 2021-11-04 by chapulina