gazebosim/gz-sim #1156: Generic point cloud visualization

chapulina, 2021-11-01:
The [VisualizeLidar](https://github.com/ignitionrobotics/ign-gazebo/blob/c08fc87e2b965ab388f6dc4a74f8668213ecae8f/src/gui/plugins/visualize_lidar/VisualizeLidar.cc) plugin allows visualizing laser scans as, among other visualizations, point clouds. That plugin is very tied to Lidar sensors though. It would be great to have a generic plugin that displays point clouds on the 3D scene coming from any source, not specifically sensors. The immediate use case we have at the moment is visualizing scientific data.

## Desired behavior

Open a GUI plugin, choose a topic to display, and it shows on the 3D scene. Some other nice to haves:

* Configure color, point size, display shape...
* Only generate markers for points within the user camera's view, for performance. We're dealing with hundreds of thousands of points spread across several kilometers, so it's not necessary to have it all on the scene at the same time.

## Alternatives considered

Extend the lidar visualization plugin to handle any point cloud. I think that's a stretch though, and we'll need to address various assumptions about frame of reference and others.

## Implementation suggestion

Create a GUI plugin that subscribes to point cloud messages and makes marker requests.

## Additional context

We have a prototype on a downstream project that we're planning to upstream.

mabelzhang, 2021-11-03:
Noting to self (or whoever ends up being the lucky camper to implement this):
This ticket includes per-vertex coloring.

That is, there needs to be a way to specify the color for each point in the cloud, separately.
This would improve performance drastically, because currently, to get different colored points, they need to be in different Marker messages, as every point in a MarkerArray would be the same color. For hundreds of markers, that is a drain on memory.

Some design decisions need to be made, as making an array of Materials can add a big overhead to memory. If a lighter implementation can be found, that would be preferred.

chapulina, 2022-03-02:
The first version has been merged in #346. Let's ticket new issues for new features as needed.

closed 2022-03-02 by chapulina