gazebosim/gz-sim #1202: Visualize lidar plugin segfaults

nlamprian, 2021-11-15:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build? Source / Dome / f8942cd9b19cf0b61a1d5a7fb9f47d0c1d558e84
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system.
    - Rendering plugin: ogre2
      - [ ] Sensor rendering error. 
      - [x] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [x] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          ➜  ~ LANG=C lspci -nn | grep VGA  # might require installing pciutils
          09:00.0 VGA compatible controller [0300]: NVIDIA Corporation Device [10de:2206] (rev a1)
          ➜  ~ echo "$DISPLAY"
          :0.0
          ➜  ~ LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
          direct rendering: Yes
          OpenGL vendor string: NVIDIA Corporation
          OpenGL renderer string: NVIDIA GeForce RTX 3080/PCIe/SSE2
          OpenGL core profile version string: 4.6.0 NVIDIA 470.82.00
          OpenGL core profile shading language version string: 4.60 NVIDIA
          OpenGL core profile context flags: (none)
          OpenGL core profile profile mask: core profile
          OpenGL version string: 4.6.0 NVIDIA 470.82.00
          OpenGL shading language version string: 4.60 NVIDIA
          OpenGL context flags: (none)
          OpenGL profile mask: (none)
          OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 470.82.00
          OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
          ➜  ~ ps aux | grep Xorg
          root        1498  1.9  0.9 26213464 608632 tty7  Ssl+ Nov13  54:02 /usr/lib/xorg/Xorg -core :0 -seat seat0 -auth /var/run/lightdm/root/:0 -nolisten tcp vt7 -novtswitch
          nlampri+ 3806274  0.0  0.0  19392  2832 pts/6    S+   08:28   0:00 grep --color=auto Xorg
          ➜  ~ sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          X.Org X Server 1.20.11
          X Protocol Version 11, Revision 0
          Build Operating System: linux Ubuntu
          Current Operating System: Linux nlamprian-ryzen 5.11.0-40-generic #44~20.04.2-Ubuntu SMP Tue Oct 26 18:07:44 UTC 2021 x86_64
          Kernel command line: BOOT_IMAGE=/vmlinuz-5.11.0-40-generic root=/dev/mapper/vgroot-lvroot ro quiet splash
          Build Date: 06 July 2021  10:17:51AM
          xorg-server 2:1.20.11-1ubuntu1~20.04.2 (For technical support please see http://www.ubuntu.com/support) 
          Current version of pixman: 0.38.4
          	Before reporting problems, check http://wiki.x.org
          	to make sure that you have the latest version.
          ```          
    - [ ] Please, attach the ogre.log or ogre2.log file from  `~/.ignition/rendering`
    
<details>
  
```
08:31:24: Creating resource group General
08:31:24: Creating resource group Internal
08:31:24: Creating resource group Autodetect
08:31:24: SceneManagerFactory for type 'DefaultSceneManager' registered.
[... 745 lines cut ...]
#22   Object "/home/nlamprian/.rvm/rubies/ruby-3.0.0/bin/../lib/libruby.so.3.0", at 0x7f454e7985ea, in 
#21   Object "/home/nlamprian/.rvm/rubies/ruby-3.0.0/lib/ruby/3.0.0/x86_64-linux/fiddle.so", at 0x7f4549b0ac43, in 
#20   Object "/home/nlamprian/.rvm/rubies/ruby-3.0.0/bin/../lib/libruby.so.3.0", at 0x7f454e76235a, in rb_nogvl
#19   Object "/home/nlamprian/.rvm/rubies/ruby-3.0.0/lib/ruby/3.0.0/x86_64-linux/fiddle.so", at 0x7f4549b09eeb, in 
#18   Object "/lib/x86_64-linux-gnu/libffi.so.7", at 0x7f4549a9f409, in 
#17   Object "/lib/x86_64-linux-gnu/libffi.so.7", at 0x7f4549a9fff4, in 
#16   Object "/home/nlamprian/ignition_dome_ws/install/lib/libignition-gazebo4-ign.so.4.12.0", at 0x7f4549004e0d, in runGui
#15   Object "/home/nlamprian/ignition_dome_ws/install/lib/libignition-gazebo4-gui.so.4", at 0x7f4548db6a6b, in ignition::gazebo::v4::gui::runGui(int&, char**, char const*)
#14   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f4547b89245, in QCoreApplication::exec()
#13   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f4547b814da, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#12   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f4547bda564, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#11   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f45459bf4a2, in g_main_context_iteration
#10   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f45459bf3ff, in 
#9    Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f45459bf17c, in g_main_context_dispatch
#8    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f4547bdaf66, in 
#7    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f4547b855b7, in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
#6    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f4547b82939, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#5    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f45473460ef, in QApplication::notify(QObject*, QEvent*)
#4    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f454733ca65, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#3    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f4547baed59, in QObject::event(QEvent*)
#2    Object "/home/nlamprian/ignition_dome_ws/install/lib/libignition-gazebo4-gui.so.4", at 0x7f4548dbe26a, in ignition::gazebo::v4::GuiRunner::OnStateQt(ignition::msgs::SerializedStepMap const&)
#1    Object "/home/nlamprian/ignition_dome_ws/install/lib/libignition-gazebo4-gui.so.4", at 0x7f4548dbe14a, in ignition::gazebo::v4::GuiRunner::UpdatePlugins()
#0    Object "/home/nlamprian/ignition_dome_ws/install/lib/ign-gazebo-4/plugins/gui/libVisualizeLidar.so", at 0x7f450e8decf4, in ignition::gazebo::v4::VisualizeLidar::Update(ignition::gazebo::v4::UpdateInfo const&, ignition::gazebo::v4::EntityComponentManager&)
Segmentation fault (Address not mapped to object [0x10])
```

ahcorde, 2021-12-02:
@nlamprian, can you try this [patch](https://github.com/ignitionrobotics/ign-gazebo/pull/1224) ?

nlamprian, 2021-12-02:
Now I get spammed with
`[GUI] [Err] [VisualizeLidar.cc:314] The entity could not be found.Error displaying lidar visual`

I'm guessing there is a mixup somewhere in the component manager. The problem happens when the world and a model have the same name.

peci1, 2021-12-03:
I think this is normal while the GUI is loading. Does the spamming stop when GUI loads?

NB: I don't say the behavior is correct, I just say it has been there for long and has no "practical" implications.

nlamprian, 2021-12-03:
The problem happens after the GUI loads when I refresh the topic list (which autoselects the first topic)

darksylinc, 2022-09-19:
Is this bug still present?

It sounds like a bug that should've been fixed by #1538

peci1, 2022-09-19:
This bug is still there (in Fortress). If the world name is the same as model name, console spams " [GUI] [Err] [VisualizeLidar.cc:314] The entity could not be found.Error displaying lidar visual". The visualization works, though.