gazebosim/gz-sim #1227: Separate linear and angular min/max values in DiffDrive

roni-kreinin, 2021-12-01:
## Desired behavior

Currently the min/max velocity, acceleration, and jerk parameters are applied to both linear and angular vel/accel/jerk. It would be useful to separate linear and angular as you often want different values for these.

## Implementation suggestion
Separate the parameters into:
- min_linear_velocity
- max_linear_velocity
- min_angular_velocity
- max_angular_velocity
- min_linear_acceleration
- max_linear_acceleration
- min_angular_acceleration
- max_angular_acceleration
- min_linear_jerk
- max_linear_jerk
- min_angular_jerk
- max_angular_jerk

roni-kreinin, 2021-12-02:
Added PR. See #1229

chapulina, 2021-12-03:
Thanks for the PR. For completeness, we should apply the same fix to all other controllers. I'll leave this issue open until they're all addressed:

- [ ] `AckerMannSteering`
- [ ] `MecanumDrive`

PR #1229 (2022-01-11): Min and max parameters for velocity, acceleration, and jerk apply to linear and angular separately.

# 🎉 New feature

Closes #1227

## Summary
`min_velocity` is now separated into `min_linear_velocity` and `min_angular_velocity`. Same for other parameters.

## Test it
Use DiffDrive plugin with the new parameter tags. i.e. `<min_linear_velocity>1.0</min_linear_velocity>`

## Checklist
- [ ] Signed all commits for DCO
- [ ] Added tests
- [ ] Added example and/or tutorial
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] `codecheck` passed (See [contributing] [... cut]
closed 2026-09-29 by azeey