gazebosim/gz-sim #1232: Model editor updates

nkoenig, 2021-12-03:
1. Add tests to cover `ModelCommandAPI`. See [here](https://github.com/ignitionrobotics/ign-gazebo/pull/1231#discussion_r762293623)
2. Make sure everything in ModelCommandAPI and on the Component inspector has units. See [here](https://github.com/ignitionrobotics/ign-gazebo/pull/1231#discussion_r762291282)
3. Add View test. See [here](https://github.com/ignitionrobotics/ign-gazebo/pull/1231#discussion_r762286160).

chapulina, 2022-03-01:
The model editor work has been merged as a separate plugin, `ComponentInspectorEditor`, for expediency. We should port all the features from that plugin to `ComponentInspector` and then deprecate it so we have only one inspector.

chapulina, 2022-03-19:
PR porting some of the refactoring that organizes the C++ code into various files:

* https://github.com/ignitionrobotics/ign-gazebo/pull/1400