gazebosim/gz-sim #1237: /server_control stop does nothing

peci1, 2021-12-07:
## Environment
* OS Version: Ubuntu 18.04
* Source or binary build? Dome binary from distribution

## Description
* Expected behavior: Calling `ign service -s /server_control --reqtype ignition.msgs.ServerControl --reptype ignition.msgs.Boolean --timeout 1000 --req 'stop: 1'` should stop the Gazebo server, i.e. exit the server binary.
* Actual behavior: Nothing happens.

## Steps to reproduce

1. `ign gazebo`
2. `ign service -s /server_control --reqtype ignition.msgs.ServerControl --reptype ignition.msgs.Boolean --timeout 1000 --req 'stop: 1'`
3. Nothing happens

## Output
This is written to the console:

```
[GUI] [Dbg] [TmpIface.cc:37] OnServerControl: request
[GUI] [Dbg] [TmpIface.cc:38] stop: true

[GUI] [Dbg] [TmpIface.cc:42] OnServerControl: response
[GUI] [Dbg] [TmpIface.cc:43] data: true
```

I saw #654, so who should be actually handling the ServerControl thing?

The implementation in Dome is actually just a dummy:

https://github.com/ignitionrobotics/ign-gazebo/blob/c473a08ea3a8cc73a9fb722efb08772abcf48dfe/src/gui/TmpIface.cc#L37-L45

However, I'm seeking for a gui-only/service-calling way to exit the server (to allow exiting the simulation in a touch-only kiosk where gui is launched from a separate console than server).

Is this a kind of supported scenario? I see Fortress does not even have TmpIface, so...

I know I could `exec()` a `killall ruby`, but that's not very nice...

chapulina, 2021-12-07:
> The implementation in Dome is actually just a dummy:
> I see Fortress does not even have TmpIface, so...

Yeah `TmpIface` was just there to show how we could hook it up, but it has never been functional. That's why it was removed. The functionality has never been implemented and it would be nice to have.

peci1, 2021-12-07:
Okay. I'll consider a PR. I'll try adding it to `ServerPrivate`.

When sending the PR against Citadel, the service server created by TmpIface would need to be deleted so that the one in `ServerPrivate` does not collide with it. I assume it's okay given the fact that the implementation did nothing. Am I right?

chapulina, 2021-12-07:
> I assume it's okay given the fact that the implementation did nothing. Am I right?

Yup, thanks!

peci1, 2021-12-08:
#1240 fixes this issue.

PR #1240 (2021-12-23): Implement /server_control::stop
# 🎉 New feature

Closes #1237

## Summary
Implemented the `stop` part of `/server_control` service.

## Test it
1. `ign gazebo -v4`
2. `ign service -s /server_control --reqtype ignition.msgs.ServerControl --reptype ignition.msgs.Boolean --timeout 1000 --req 'stop: 1'`
3. The server should stop and exit in a while.

## Checklist
- [x] Signed all commits for DCO
- [x] Added tests
- [ ] Added example and/or tutorial
- [x] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [x] `codecheck` passed (See [contributing](https://ignitionrobotics [... cut]
closed 2022-06-24 by nkoenig