gazebosim/gz-sim #1242: Move general-purpose functions from VisualizationCapabilities to Util

chapulina, 2021-12-10:
## Desired behavior

There are functions inside the visualization capabilities plugin that could be generally useful for other plugins, for example:

https://github.com/ignitionrobotics/ign-gazebo/blob/25df2996cf68649fc0b7cdb898d90f8953d7f005/src/gui/plugins/visualization_capabilities/VisualizationCapabilities.cc#L96-L105

We should move these to the `Util` header, so other plugins can make use of them without duplicating code. When doing so, we should make sure that the functions are actually generic and don't have specific behaviour for that plugin.

## Alternatives considered

Duplicating the functions on other plugins.

We may also want to wait until other plugins actually need these functions to make them generic. It could be the case that no plugins actually need them.

## Implementation suggestion

Make sure that the functions take in the ECM as one of the parameters, and that they don't rely on any internal state (all parameters should be passed to the function).

## Additional context

Motivated by this comment: https://github.com/ignitionrobotics/ign-gazebo/pull/1105#discussion_r766131530

rishitej04, 2024-12-25:
can you assign it to me I would like to work on this

pausofi17, 2026-03-16:
Hi, I'm interested in working on this issue. Is it still available? I'm applying to GSoC 2026 with Open Robotics.