gazebosim/gz-sim #1245: Some sensors system does not work when model is spawned

iche033, 2021-12-11:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build? binary, ignition-gazebo 6.2.0

## Description

spawn a model with air pressure, imu, or magnetometer sensors into the world

* Expected behavior: data are published to ign topics
* Actual behavior: ign-gazebo produces these errors:
  ```
  [Err] [AirPressure.cc:229] Failed to update magnetometer: 42. Entity not found.
  ```

## Steps to reproduce

1. `ign gazebo -v 4 sensors.sdf`
2. Drag and drop this UAV model: https://app.ignitionrobotics.org/OpenRobotics/fuel/models/X3%20UAV%20Config%203
3. Press play to see errors printed in the console

## Possible cause of issue

Looks like it's an issue with the order of which systems are run in. For the sensors.sdf world, the error is gone is you just move the User Command system in the world SDF file before all the different sensor systems:

```xml
diff --git a/examples/worlds/sensors.sdf b/examples/worlds/sensors.sdf
index 85746503..751dc860 100644
--- a/examples/worlds/sensors.sdf
+++ b/examples/worlds/sensors.sdf
@@ -19,6 +19,10 @@
       filename="ignition-gazebo-physics-system"
       name="ignition::gazebo::systems::Physics">
     </plugin>
+    <plugin
+      filename="ignition-gazebo-user-commands-system"
+      name="ignition::gazebo::systems::UserCommands">
+    </plugin>
     <plugin
       filename="ignition-gazebo-air-pressure-system"
       name="ignition::gazebo::systems::AirPressure">
@@ -35,10 +39,6 @@
       filename="ignition-gazebo-magnetometer-system"
       name="ignition::gazebo::systems::Magnetometer">
     </plugin>
-    <plugin
-      filename="ignition-gazebo-user-commands-system"
-      name="ignition::gazebo::systems::UserCommands">
-    </plugin>
     <plugin
       filename="ignition-gazebo-scene-broadcaster-system"
       name="ignition::gazebo::systems::SceneBroadcaster">

```

chapulina, 2021-12-13:
Possibly related to #797

chapulina, 2022-01-25:
I believe this should be fixed since https://github.com/ignitionrobotics/ign-gazebo/pull/1281, please reopen if that's not the case.

closed 2022-01-25 by chapulina