gazebosim/gz-sim #1247: Port U

iche033, 2021-12-14:
## Desired behavior

## Alternatives considered

## Implementation suggestion

## Additional context

chapulina, 2021-12-14:
> usv_gazebo_dynamics_plugin

This looks like [ignition::gazebo::systems::Hydrodynamics](https://ignitionrobotics.org/api/gazebo/6.3/classignition_1_1gazebo_1_1systems_1_1Hydrodynamics.html#details). It even links to the original plugin. 

One thing to note is that the plugin has been added without tests in #749, I've been meaning to add some tests soon.

We also have a [fork of it on LRAUV](https://github.com/osrf/lrauv/blob/main/lrauv_ignition_plugins/src/HydrodynamicsPlugin.hh) because some bugs have been fixed. Those should be upstreamed soon, progress is tracked here: https://github.com/osrf/lrauv/issues/37

> usv_gazebo_thrust_plugin

Maybe [ignition::gazebo::systems::Thruster](https://ignitionrobotics.org/api/gazebo/6.3/classignition_1_1gazebo_1_1systems_1_1Thruster.html) can be extended to support all features on that plugin.

> usv_gazebo_wind_plugin

It looks similar to [ignition::gazebo::systems::WindWffects](https://ignitionrobotics.org/api/gazebo/6.3/classignition_1_1gazebo_1_1systems_1_1WindEffects.html), I think the functionality could be combined.

> buoyancy_gazebo_plugin

I think the core functionality should be achieved with [ignition::gazebo::systems::Buoyancy](https://ignitionrobotics.org/api/gazebo/6.3/classignition_1_1gazebo_1_1systems_1_1Buoyancy.html#details), but note that it's a world plugin instead of a model plugin. It also seems to have some different parameters, not sure if the drag is best kept at `LiftDrag` or `Hydrodynamics`.

iche033, 2021-12-14:
after some discussion with @caguero. The plan is:
* try out the Hydrodynamics and Buoyancy plugin as is 
* look more into the Thruster plugin and see if it supports the actuator capability we need
* port the wave model part of the `usv_gazebo_dynamics_plugin` to Ignition
* convert the WAM-V urdf file from VRX to sdf with a single fixed configuration

caguero, 2021-12-14:
We'll extend the `Thruster` plugin to support the articulation of the thruster.