gazebosim/gz-sim #1252: Quick start dialog

chapulina, 2021-12-15:
A user starting Gazebo for the first time may not know what to do after they run `ign gazebo` for the first time. An empty world shows up... And now what?

It's common for applications to offer an initial dialog with some quickstart options for beginners. These can range from simple examples to complex simulations, ready to be used out of the box.

## Desired behavior

When a user opens `ign gazebo` for the first time, they see a dialog with options for loading example worlds, or even more complex worlds from Ignition Fuel, such as the Edifice or Fortress worlds. There should be a checkbox for the user to choose not to see the box in the future.

## Alternatives considered

Besides letting the user load a world, we could also have some sort of instructions like a self-guided tutorial, so they walk through the interface and know where to poke around.

## Implementation suggestion

We could try to get inspiration from other programs to design a dialog that's easy to use. I think we should have a handful of recommended entry points (3 or 4) more prominently placed, with thumbnails and whatnot, as well as a list of other examples available.

The setting for the user not to see the dialog in the future could be saved in a file under `~/.ignition/gazebo`, similar to Gazebo classic's `~/.gazebo/gui.ini`.

mayman99, 2022-04-18:
I've drafter sth like this, initial menu taken from blender.
[image]

I believe there is a place for one more item on the left side, but I'm not sure which option to add there, if we would like to add any.

chapulina, 2022-04-20:
Thanks for the draft, @mayman99 ! 

One thing to consider is displaying thumbnails of some of the worlds that can be loaded. The ones installed with Gazebo don't have thumbnails, but the ones from Fuel do. We could curate a list of a few worlds that are featured with thumbnails, to start with those can be just the Edifice and Fortress worlds from Fuel. The installed worlds can stay in the dropdown.

One possible inspiration is the quickstart dialog [here](https://magazine.renderosity.com/article/5663/get-informed-and-inspired-every-time-you-start-cinema-4d) I didn't find a better screenshot, but see how there's a thumbnail for each example and a bit of text above it.

mayman99, 2022-04-21:
Sounds good, like the thumbnails idea.

chapulina, 2022-05-04:
The plan sounds good to me. I don't think a plugin would be the best approach here since we want this to show up every time.

chapulina, 2022-08-03:
* https://github.com/gazebosim/gz-sim/pull/1536

PR #1622 (2022-07-30): Alternative quick setup menu
# 🎉 New feature

## Summary
Closes #1252
Alternative to #1536

## Test it
Run `ign gazebo -v 4`

https://user-images.githubusercontent.com/1587438/181843751-fd9a421b-c0e7-42fb-8a2d-edcc79b7faf5.mp4

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Added example and/or tutorial
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests [... cut]
PR #1536 (2022-08-02): Quick start dialog
# 🎉 New feature
depends on [ign-gui418](https://github.com/gazebosim/gz-gui/pull/418)

Closes #1252 
[image]
[image]

- [x] Signed all commits for DCO
- [x] Added tests
- [ ] Added example and/or tutorial
- [x] Updated documentation (as needed)
- [x] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [x] `codecheck` passed (See [contributing](https://ignitionrobotics.org/docs/all/contributing#contributing-code))
- [x] All tests passed (See [test coverage](https://ignitionrobotics.org/docs/all/contributing#test-coverage))
- [... cut]
closed 2022-08-03 by chapulina