gazebosim/gz-sim #1260: Download Fuel models on the background

chapulina, 2021-12-20:
When opening a world with models that have not been downloaded, there can be a long delay before the GUI opens.
It's also not possible to close gazebo while it's doing the initial download of models.

## Desired behavior

While models are loading, the user should be able to interact with the window and close it if wanted.

## Alternatives considered

I guess we could not open the window until the models are loaded, but that may be more confusing for users.

## Implementation suggestion

* Download models in a separate thread
* Display models as they're downloaded
* Cancel loading if a signal is received
* Display status to user of what models are being loaded and what are missing
* Once we have a splash screen (https://github.com/ignitionrobotics/ign-gui/issues/336), we can use it to display this information

ahcorde, 2022-04-08:
I have been working in this issue but I have some blockers. This is the branch of the current implementation https://github.com/ignitionrobotics/ign-gazebo/tree/ahcorde/download_background

- When I call `Root::Load()`

https://github.com/ignitionrobotics/ign-gazebo/blob/0e255a076c9063356c301e9d70118263051b3307/src/Server.cc#L168

I need to "fail" this call when there is a `include` tag because I launch a thread for each one of the `include` tag.

https://github.com/ignitionrobotics/ign-gazebo/blob/0e255a076c9063356c301e9d70118263051b3307/src/ServerPrivate.cc#L558-L561

The callback method will return an empty string making the `Load` method to fail, but when the thread has downloaded the resources are included. This is not problematic at all (we can filter the error in the load method), but when a include fails then I loose the `pose` and `name`, which are inside the `include` tag. But the model is not included because I force to fail to create the thread.

**First question** I don't know how to recover this data, is there any way to get this values ? the only idea that came to my mund is reloading the file again

This gif shows how models are downloaded in the background but without the right position and naming.

[image]

-------- 

The other issue is about SDFs that define a geometry with an URI, this is still blocking the GUI. Like this one

```
 <model name="Radio">
      <pose>3 -1.5 0 0 0 0</pose>
      <static>true</static>
      <link name="link">
        <collision name="collision">
          <geometry>
            <mesh>
              <uri>https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Radio/4/files/meshes/Radio.dae</uri>
            </mesh>
          </geometry>
        </collision>
        <visual name="visual">
          <geometry>
            <mesh>
              <uri>https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Radio/4/files/meshes/Radio.dae</uri>
            </mesh>
          </geometry>
        </visual>
      </link>
    </model>
```
The model is downloaded in the same thread but for some reason this is blocking the runner

**Question**: I'm not really sure who might be blocking the runner. any thoughts?

-----

This feature might include some changes in sdformat or/and fuel, Do you have any thoughts ?
@chapulina @scpeters @azeey

scpeters, 2022-05-03:
> when a include fails then I loose the `pose` and `name`, which are inside the `include` tag. But the model is not included because I force to fail to create the thread.
>
> **First question** I don't [...]

This reminds me of the `PreserveIncludes` option in [PrintConfig.hh](https://github.com/ignitionrobotics/sdformat/blob/sdf12/include/sdf/PrintConfig.hh#L68-L76) (used in various libsdformat APIs for printing to strings) and also the `ToElementUseIncludeTag` option in `OutputConfig.hh` from https://github.com/ignitionrobotics/sdformat/pull/918 that is used to configure the `ToElement()` APIs.

> * When I call `Root::Load()`
> 
> https://github.com/ignitionrobotics/ign-gazebo/blob/0e255a076c9063356c301e9d70118263051b3307/src/Server.cc#L168
> 
> I need to "fail" this call when there is a `include` tag because I launch a thread for each one of the `include` tag.

> The callback method will return an empty string making the Load method to fail, but when the thread has downloaded the resources are included. This is not problematic at all (we can filter the error [...]

I don't really understand all that is happening here

closed 2025-08-26 by azeey