gazebosim/gz-sim #1272: Allow selecting headless rendering device

peci1, 2022-01-04:
## Environment
* OS Version: Ubuntu 18.04
* Source or binary build? binary 6.3.0

## Description
* Expected behavior: Users should be able to specify which rendering device should be used for headless rendering (i.e. pass `/dev/dri/card1` to gazebo somehow to select the corresponding GPU)
* Actual behavior: It's not possible, a default device is always used.

## Analysis

[OGRE EGL headless example](https://github.com/OGRECave/ogre-next/blob/02d1cc5d21a01f693507ad710d0aed10dfe8b404/Samples/2.0/Tutorials/Tutorial_EglHeadless/Tutorial_EglHeadless.cpp#L213) shows how to select a rendering device:

    root->getRenderSystem()->setConfigOption( "Device", itor->second.possibleValues[devIdx] );

Where `possibleValues` comes from `root->getRenderSystem()->getConfigOptions().find("Device")`.

The names of the devices are a bit weird (see the example), but important is that the HW master DRM nodes have their names like `/dev/dri/card0` appended to the end of the weird name:

https://github.com/OGRECave/ogre-next/blob/master/RenderSystems/GL3Plus/src/windowing/EGL/PBuffer/OgreEglPBufferSupport.cpp#L92-L103

I think ign-rendering should pass the device name somewhere around

https://github.com/ignitionrobotics/ign-rendering/blob/8e7b7fbbbff519a6fb567efdd4ec08468e8d8f6a/ogre2/src/Ogre2RenderEngine.cc#L608

I'm not really sure how this would work on Windows/Mac, but I guess EGL is anyways Linux-only?

It seems to me this change should be done by someone who is closer to the rendering part, so I'm not offering a PR for this issue.

matt-the-ogre, 2026-09-11:
Adding a data point that may be relevant to how this gets implemented: on some hosts, **device selection alone would not be sufficient**, because the vendor's EGL registers no device at all.

Setup: Debian 13 host, GTX 1060, driver 550.163.01, Docker 29.5.3 with the `nvidia` runtime, container `ros:jazzy-ros-base` + Gazebo Harmonic (gz-sim 8.11.0), run with `--gpus all` and `NVIDIA_DRIVER_CAPABILITIES=all`. `libEGL_nvidia.so.0` is mounted in the container and `/usr/share/glvnd/egl_vendor.d/10_nvidia.json` is present and valid.

Calling `eglQueryDevicesEXT` directly (the same enumeration `EglPBufferSupport` uses) gives:

```
EGL devices found: 2

#0  DRM node : /dev/dri/card0
    exts     : EGL_EXT_device_drm EGL_EXT_device_drm_render_node
               EGL_EXT_device_query_name EGL_EXT_device_persistent_id
    eglInitialize: FAILED

#1  DRM node : (none)
    exts     : EGL_MESA_device_software
    eglInitialize: OK  vendor=Mesa Project  version=1.5
```

Both entries are Mesa's — a DRM node it cannot drive (the card is NVIDIA proprietary), and its software rasterizer. There is no NVIDIA EGL device in the list.

What `gz sim -s -r --headless-rendering --render-engine-server ogre2` does with this is take index `#0` and abort:

```
libEGL warning: pci id for fd 52: 10de:1c03, driver (null)
libEGL warning: egl: failed to create dri2 screen
[Err] [Ogre2RenderEngine.cc:1304] Unable to create the rendering window:
  OGRE EXCEPTION(3:RenderingAPIException): eglInitialize failed for device
  EGL_EXT_device_drm ... #0 /dev/dri/card0 in EGLSupport::getGLDisplay
[Err] OpenGL 3.3 is not supported. Please update your graphics card drivers.
```

Two things that might be worth considering for whoever picks this up:

1. **Skipping devices where `eglInitialize` fails** would be a cheap, independent improvement — it turns a hard abort into a usable fallback even without user-facing selection. In the trace above, `#0` fails and `#1` would have worked (as software, but it would run).
2. **A selection API alone may not rescue hosts like this one.** There is no correct device to select here. Anyone hitting this specific shape needs to fix EGL vendor registration first, so it may be worth having the error say *which* devices were enumerated rather than only the one that failed — that alone would have saved a lot of guessing on my end.

Happy to run diagnostics on this host if it is useful. The enumeration script is [here](https://github.com/matt-the-ogre/ros2-gazebo-infra/blob/main/scripts/egl_devices.py) (plain `ctypes`, no build needed).

For reference, the workaround is `LIBGL_ALWAYS_SOFTWARE=1` with no GPU passed to the container at all — Mesa then falls back cleanly to llvmpipe. Counterintuitively, *attaching* the GPU makes it worse, since that is what puts the undrivable DRM node at index `#0`.

matt-the-ogre, 2026-09-11:
**Correcting my comment above** — I found the actual cause on my host, and it was not Gazebo's fault at all. Posting the resolution in case anyone else lands here with the same symptom.

The reason no NVIDIA EGL device appeared: **`libEGL_nvidia.so.0` was being mounted into the container, but its dependencies were not.**

```
$ ldd /usr/lib/x86_64-linux-gnu/nvidia/current/libEGL_nvidia.so.0 | grep "not found"
        libnvidia-glsi.so.550.163.01 => not found
```

`libnvidia-eglcore` and `libnvidia-glcore` were missing too. On Debian the driver's libraries are split across `/usr/lib/x86_64-linux-gnu/` and `/usr/lib/x86_64-linux-gnu/nvidia/current/`, and nvidia-container-toolkit mounted only the latter. libglvnd **silently skips a vendor whose library fails to `dlopen`** — no warning, no error — so NVIDIA registered nothing and only Mesa's devices remained.

Staging the host's full `libnvidia-*` set and mounting it with `LD_LIBRARY_PATH` fixes it completely:

```
EGL devices found: 3          (was 2)

#0  EGL_NV_device_cuda EGL_EXT_device_drm ...
    eglInitialize: OK  vendor=NVIDIA  version=1.5      <- was FAILED
#1  EGL_EXT_device_drm ...          eglInitialize: FAILED
#2  EGL_MESA_device_software        eglInitialize: OK  vendor=Mesa Project
```

Since NVIDIA's device lands at index `#0`, `gz sim -s -r --headless-rendering --render-engine-server ogre2` now works unmodified — no device selection needed. Real-time factor on a GTX 1060, TurtleBot 4 in a warehouse, same host:

| lidar rate | Mesa llvmpipe | NVIDIA EGL |
|---|---|---|
| 62 Hz | 0.026 | **0.381** |
| 10 Hz | 0.251 | **0.772** |

**So my "a selection API alone may not rescue hosts like this one" is wrong for this host** — device selection was never the blocker; a broken library mount was. Apologies for the noise.

The other suggestion still seems worth considering on its own merits, though: **skipping devices where `eglInitialize` fails** rather than aborting on `#0`. In my broken state, `#2` (Mesa software) would have initialized fine and given a slow-but-working sim instead of a segfault. And listing the enumerated devices in the error message would have pointed me at the real problem immediately — the current message names only the device that failed, which is what sent me looking at device *selection* rather than at whether the vendor library had loaded.