gazebosim/gz-sim #1278: Invalid package dependencies in ign_gazebo5

harshmahesheka, 2022-01-11:
I was trying to build ign_gazebo based on steps on https://github.com/ignitionrobotics/ign-gazebo/tree/ign-gazebo5#readme, but in the dependencies, it contains ignition-physics6 which is not available on the ign_physics site. You can check those dependencies on https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo5/CMakeLists.txt  .

chapulina, 2022-01-11:
Ahh I think the problem is this line:

`git clone https://github.com/ignitionrobotics/ign-gazebo -b main`

It should be 

`git clone https://github.com/ignitionrobotics/ign-gazebo -b ign-gazebo5`

Can you give that a try?

harshmahesheka, 2022-01-12:
Oh, yes it worked. Thanks for the help .Also 
`sudo apt-add-repository ppa:dartsim/ppa `
`sudo apt-get update`
this must be done before 
`sudo apt -y install   $(sort -u $(find . -iname 'packages-'$SYSTEM_VERSION'.apt' -o -iname 'packages.apt') | tr '\n' ' ')
`
Should I add a Pull request?

harshmahesheka, 2022-01-12:
Yeah, it worked. Thanks for the help. Also, I want to work on this issue could you please assign it to me.

chapulina, 2022-01-12:
Thanks for the fix!

closed 2022-01-12 by chapulina